espressif/arduino-esp32 #7869: esp32-s3 is not connected to wifi

hyu7000, 2023-02-20:
### Board

esp32s3 

### Device Description

it's esp32 s3 dev module

### Hardware Configuration

i just use esp32s3 dev kit

### Version

v2.0.6

### IDE Name

arduino IDE

### Operating System

window 10

### Flash frequency

QIOI 80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

if i use esp-idf, my board can connect to wifi. but, arduino is not.

when i use esp-idf, terminal show me msg: 
"I (2406) wifi station: got ip:192.168.0.19
I (2406) wifi station: connected to ap SSID:myWifi password:myPass"

when i use arduino IDE and some example code, it's cannot connect to wifi.

i don't know why that is.

### Sketch

```cpp
#include <WiFi.h>

const char* ssid = "myWifi";
const char* password = "myPass";

void setup(){
    Serial.begin(115200);
    delay(1000);

    WiFi.useStaticBuffers(true);
    WiFi.mode(WIFI_STA); //Optional
    WiFi.begin(ssid, password);
    Serial.println("\nConnecting");

    while(WiFi.status() != WL_CONNECTED){
        Serial.print(".");
        delay(100);
    }

    Serial.println("\nConnected to the WiFi network");
    Serial.print("Local IP: ");
    Serial.println(WiFi.localIP());
}

void loop(){}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8

Connecting
................................................................................

+ dot was generated until terminal exit.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-02-20:
Does it work without `WiFi.useStaticBuffers(true);`?

SuGlider, 2023-02-20:
@hyu7000 - Is your AP using WPA3?

SuGlider, 2023-02-20:
If your networks uses WEP or WPA, you may need to set the minimum security level to it, using `Wifi.setMinSecurity()`
By default, Arduino sets the minimum to WPA2 PSK.

mrengineer7777, 2023-03-30:
@hyu7000 Any updates on this?

lbernstone, 2023-04-19:
> I have the similar issue with Arduino IDE 2.0.4 The WiFi.setMinSecurity(WIFI_AUTH_WEP); statement does not help. I turn on Core Debug to Verbose and I have the following results. I think I am connected to the WiFi router but at the WiFi.status(), I always have 0 as the return codes. My SSID is lubernet . [WiFi] Connecting to lubernet [ 185][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY [ 219][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started [ 221][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0 [ 221][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START [ 277][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: lubernet, BSSID: 44:d9:e7:45:d7:65, Channel: 1, Auth: WPA2_PSK [ 278][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED

You are giving it invalid static IP info.  Not the same issue.

VojtechBartoska, 2023-08-18:
Closing this ticket as expired, if needed you can reopen. Thanks

closed 2023-08-18 by VojtechBartoska

vincentj87, 2023-09-30:
i think i had similar issue as @DMNFArdunio here, when i set debug core level to verbose i got something like this
[image]

Matley174, 2025-02-16:
Good Morning, i have same problem, i try to load some example sketch, if i use WiFi scan i read a real wifi network presente
`Scan start
Scan done, elapsed time: 2989 ms
7 networks found
Nr | SSID                             | RSSI | CH | Encryption
 1 |                                  |  -38 |  1 | WPA2+WPA3  ** This is a guest network 2.4Ghz
 2 |                                  |  -38 |  1 | WPA2  ** This is a guest network 5Ghz
 3 | Miami-Dave                       |  -39 |  1 | WPA2+WPA3
 4 | TIM-52399316                     |  -87 |  2 | WPA2
 5 | TP-Link_6C16                     |  -87 | 10 | WPA+WPA2
 6 | TP-Link_FA98                     |  -89 | 11 | WPA+WPA2
 7 | NETGEAR_EXT                      |  -92 | 11 | open`
`
But if i load simply sketch for connect to wi-fi network, ESP32S3 not conntect, it's stay on state 6
`#include <WiFi.h>

// Replace with your network credentials (STATION)
const char* ssid = "Miami-Dave";
const char* password = "Matley174";

IPAddress local_IP(192, 168, 74, 210); // Set your Static IP address
IPAddress gateway(192, 168, 74, 1); // Set your Gateway IP address
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8);   //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional

void initWiFi() { WiFi.mode(WIFI_STA);
                  WiFi.setMinSecurity(WIFI_AUTH_WPA2_WPA3_PSK);   // WIFI_AUTH_OPEN is the minimum
                  WiFi.begin(ssid, password);
                  Serial.print("Connecting to ");
                  Serial.print(ssid);
                  Serial.println(password);
                  delay(5000);
                  Serial.print("WiFi status : ");
                  Serial.println(WiFi.status());
                  Serial.print("local IP : ");
                  Serial.println(WiFi.localIP());
                  // Print ESP MAC Address
                  Serial.print("MAC address: ");
                  Serial.println(WiFi.macAddress());
  while (WiFi.status() != WL_CONNECTED) { Serial.print('.');  delay(1000);  }
  
}

void setup() {  Serial.begin(115200);
                // Configures static IP address
                if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {  Serial.println("STA Failed to configure");  }
                initWiFi();
                Serial.print("RRSI: ");
                Serial.println(WiFi.RSSI());
              }

void loop() {
  // put your main code here, to run repeatedly:
}`
Feedback
`Connecting to Miami-Dave; PASSWORD
WiFi status : 6
local IP : 192.168.74.210
MAC address: A0:85:E3:E7:63:C0`

This is a data of my ESP
'SP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac'

xinkiknix, 2025-02-28:
I had a similar problem, application working fine at one location, failing to connect at other location. The reason as I found out is the security setting on the router. WEP, WPA level

> Scan start
Scan done
3 networks found
Nr | SSID                             | RSSI | CH | Encryption
 1 | RT-N66U-24                       |  -50 |  6 | WPA
 2 | Livebox-7314                     |  -76 |  1 | WPA2
 3 | SFR_CDCF                         |  -89 |  1 | WPA+WPA2

Setting the connection security minimum level to:
 WiFi.mode(WIFI_STA);
  WiFi.setMinSecurity(**_WIFI_AUTH_WEP_**);
  WiFi.begin(ssid, password);

solved the problem for connection to SSID RT-N66U-24. Arduino standard minimum leVel is  **_WPA2 PSK_**

So this is NOT a ESP32x problem but a difference in network security level at the router.

Matley174, 2025-02-28:
> Good Morning, i have same problem, i try to load some example sketch, if i use WiFi scan i read a real wifi network present
> 
> `Scan start Scan done, elapsed time: 2989 ms 7 networks found Nr | SSID | RSSI | CH | Encryption 1 | | -38 | 1 | WPA2+WPA3 ** This is a guest network 2.4Ghz 2 | | -38 | 1 | WPA2 ** This is a guest network 5Ghz 3 | Miami-Dave | -39 | 1 | WPA2+WPA3 4 | TIM-52399316 | -87 | 2 | WPA2 5 | TP-Link_6C16 | -87 | 10 | WPA+WPA2 6 | TP-Link_FA98 | -89 | 11 | WPA+WPA2 7 | NETGEAR_EXT | -92 | 11 | open ` But if i load simply sketch for connect to wi-fi network, ESP32S3 not conntect, it's stay on state 6
> 
> `#include <WiFi.h>
> 
> const char* ssid = "Miami-Dave"; const char* password = "*******";
> 
> IPAddress local_IP(192, 168, 74, 210); // Set your Static IP address IPAddress gateway(192, 168, 74, 1); // Set your Gateway IP address IPAddress subnet(255, 255, 255, 0); IPAddress primaryDNS(8, 8, 8, 8); //optional IPAddress secondaryDNS(8, 8, 4, 4); //optional
> 
> void initWiFi() { WiFi.mode(WIFI_STA); WiFi.setMinSecurity(WIFI_AUTH_WPA2_WPA3_PSK); // WIFI_AUTH_OPEN is the minimum WiFi.begin(ssid, password); Serial.print("Connecting to "); Serial.print(ssid); Serial.println(password); delay(5000); Serial.print("WiFi status : "); Serial.println(WiFi.status()); Serial.print("local IP : "); Serial.println(WiFi.localIP()); // Print ESP MAC Address Serial.print("MAC address: "); Serial.println(WiFi.macAddress()); while (WiFi.status() != WL_CONNECTED) { Serial.print('.'); delay(1000); } } void setup() { Serial.begin(115200); // Configures static IP address if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) { Serial.println("STA Failed to configure"); } initWiFi(); Serial.print("RRSI: "); Serial.println(WiFi.RSSI()); } void loop() { // put your main code here, to run repeatedly: }`
> 
> Feedback
> 
> `Connecting to Miami-Dave ******* WiFi status : 6 local IP : 192.168.74.210 MAC address: A0:85:E3:E7:63:C0`
> 
> This is a data of my ESP32-S3 WOORM-1
> 
> 'SP-ROM:esp32s3-20210327 Build:Mar 27 2021 rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT) SPIWP:0xee mode:DIO, clock div:1 load:0x3fce2820,len:0x1188 load:0x403c8700,len:0x4 load:0x403c8704,len:0xbf0 load:0x403cb700,len:0x30e4 entry 0x403c88ac'

UPDATE!! i have buy a new ESP32-S3 WOORM-1 and without any change of software work fine with same network. i think fist module have some problem about Wi-Fi communication ...