espressif/arduino-esp32 #7870: impossible to install 2.0.7 in my Arduino 1.8.19 Portable Windows 10

mikekgr, 2023-02-20:
### Board

ESP32=S3

### Device Description

ESP32-S3 DevKit

### Hardware Configuration

ESP32-S3

### Version

other

### IDE Name

Arduino 1.8.19

### Operating System

Windows 10

### Flash frequency

irrelevant 

### PSRAM enabled

yes

### Upload speed

irrelevant 

### Description

Dear Sirs,
having installing and working version 2.0.6 , I tried to update to 2.0.7 but it is not possible, I got the below error:

Tool esptool_py is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool esptool_py is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Tool esptool_py is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:84)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)

Please tell me what to do?

Thanks and Best Regards,
Mike Kranidis

### Sketch

```cpp
irrelevant
```

### Debug Message

```plain
Tool esptool_py is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool esptool_py is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Tool esptool_py is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:84)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-02-20:
I've updated the json. Could you restart your IDE and try again?

mikekgr, 2023-02-20:
> I've updated the json. Could you restart your IDE and try again?

Yes it is working fine now, many thanks Sir.

closed 2023-02-20 by me-no-dev