espressif/arduino-esp32 #7873: FunctionalInterrupt.ino example crashes on start on ESP32-S3

jetpax, 2023-02-20:
### Board

esp32-s3-devkitC-1

### Device Description

a button on GPIO18

### Hardware Configuration

GPIO18 is connected to ground through a button

### Version

v2.0.6

### IDE Name

Platformio

### Operating System

macOS

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Example compiles, but crashes on start with 'GPIO isr service already installed' error and enters a bootloop 

This is due to a problem with attaching the interrupt in the pin constructor, as pointed out by ESP_Sprite [in this discussion](https://esp32.com/viewtopic.php?t=26062)

Here is a fixed version that uses the onboard Boot button and one external button connected to pin 18.

```C++
#include <Arduino.h>
#include <FunctionalInterrupt.h>

#define BUTTON1 0
#define BUTTON2 18

class Button
{
public:
	Button(uint8_t reqPin) : PIN(reqPin){
		pinMode(PIN, INPUT_PULLUP);
	};

        void begin(){
            attachInterrupt(PIN, std::bind(&Button::isr,this), FALLING);
        }

	~Button() {
		detachInterrupt(PIN);
	}

	void ARDUINO_ISR_ATTR isr() {
		numberKeyPresses += 1;
		pressed = true;
	}

	void checkPressed() {
		if (pressed) {
			Serial.printf("Button on pin %u has been pressed %u times\n", PIN, numberKeyPresses);
			pressed = false;
		}
	}

private:
    const uint8_t PIN;
    volatile uint32_t numberKeyPresses;
    volatile bool pressed;
};

Button button1(BUTTON1);
Button button2(BUTTON2);

void setup() {
    Serial.begin(115200);
    button1.begin();
    button2.begin();
}

void loop() {
	button1.checkPressed();
	button2.checkPressed();
}

```

### Sketch

```cpp
This is the original erroring sketch:

++
#include <Arduino.h>
#include <FunctionalInterrupt.h>

#define BUTTON1 16
#define BUTTON2 17

class Button
{
public:
	Button(uint8_t reqPin) : PIN(reqPin){
		pinMode(PIN, INPUT_PULLUP);
		attachInterrupt(PIN, std::bind(&Button::isr,this), FALLING);
	};
	~Button() {
		detachInterrupt(PIN);
	}

	void ARDUINO_ISR_ATTR isr() {
		numberKeyPresses += 1;
		pressed = true;
	}

	void checkPressed() {
		if (pressed) {
			Serial.printf("Button on pin %u has been pressed %u times\n", PIN, numberKeyPresses);
			pressed = false;
		}
	}

private:
	const uint8_t PIN;
    volatile uint32_t numberKeyPresses;
    volatile bool pressed;
};

Button button1(BUTTON1);
Button button2(BUTTON2);

void setup() {
    Serial.begin(115200);
}

void loop() {
	button1.checkPressed();
	button2.checkPressed();
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376c80
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
E (85) gpio: esp_ipc_call_blocking failed (0x103)
E (85) gpio: gpio_install_isr_service(449): GPIO isr service already installed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4202180b  PS      : 0x00060733  A0      : 0x820034b0  A1      : 0x3fceb1f0  
A2      : 0x00000000  A3      : 0x3fc954fc  A4      : 0x3fc954fc  A5      : 0x00060723  
A6      : 0x00060720  A7      : 0x00000001  A8      : 0x82003125  A9      : 0x3fceb1c0  
A10     : 0x3fc91224  A11     : 0x00000011  A12     : 0x00000011  A13     : 0x3fceb1f0  
A14     : 0x02c91210  A15     : 0x00ffffff  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffff7  

Backtrace: 0x42021808:0x3fceb1f0 0x420034ad:0x3fceb210 0x420020fd:0x3fceb240 0x420017ca:0x3fceb270 0x420016ad:0x3fceb290 0x42001776:0x3fceb2d0 0x42009b72:0x3fceb2f0 0x403769e1:0x3fceb320 0x403cd6db:0x3fceb350 0x403cd99a:0x3fceb380 0x403c992d:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

ELF file SHA256: 3540ea344dce2122

E (171) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Leaveyou, 2023-02-20:
I have the same issue:
2 pins, 2 interrupts initialized in a class constructor.
I narrowed the version down. 
I found it used to work in version 2.0.2 but fails ever since 2.0.3.

jetpax, 2023-02-20:
Yes, the fix is in the Description...

SuGlider, 2023-02-21:
@P-R-O-C-H-Y , PTAL. Thanks!

Leaveyou, 2023-02-21:
@jetpax, You're right in that I missed the point that you're providing a workaround. I still think this should be fixed though.

me-no-dev, 2023-02-21:
I agree with this fix. @PilnyTomas didn't you test this sketch?

PilnyTomas, 2023-06-15:
The PR is merged, so I'm closing this issue.

closed 2023-06-15 by PilnyTomas

munkerench, 2024-06-24:
Will this fix be available in the 2.x versions of the platform?  I cannot install 3.x yet.