espressif/arduino-esp32 #7881: Two SPI can not Work

savagerex, 2023-02-23:
### Board

ESP32 S3

### Device Description

ADXL 355 - SPI
external Flash - SPI

### Hardware Configuration

NONE

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I use sample code and use two SPI. it can work.

but i use my project code and use two SPI. it can not work.

it looks like SPI2 fail....

### Sketch

```cpp
#include "SPI.h"
#include <SerialFlash.h>
#include "SPIFFS.h"
#include "BoardConfig.h"
#include "Sensor.h"
#include <WiFi.h>
//#include <PubSubClient.h>
#include <MQTT.h>

const char* ssid = "vivo Y21";     //SW1-WIFI6   ASUS-38   vivo Y21 
const char* pass = "1234567890";   //wireless    12345678   1234567890

const char* mqtt_server = "broker.emqx.io";   //broker.emqx.io  192.168.50.128
const char *topic = "DataTopic";
const char *mqtt_username = "emqx";
const char *mqtt_password = "public";

//WiFiClient espClient;
//PubSubClient client(espClient);

WiFiClient net;
MQTTClient client(72100);
[... 355 lines cut ...]
//}

/*
   Read multiple registries
*/
void readMultipleData(int *addresses, int dataSize, int *readedData) {
  digitalWrite(GPIO_OUTPUT_CS, LOW);
  for (int i = 0; i < dataSize; i = i + 1) {
    byte dataToSend = (addresses[i] << 1) | READ_BYTE;
    SPI1.transfer(dataToSend);
    readedData[i] = SPI1.transfer(0x00);
  }
  digitalWrite(GPIO_OUTPUT_CS, HIGH);
}
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

savagerex, 2023-02-23:
Ruuning the code to here.

led1GPin  con not  flashing.

while (!SerialFlash.ready()) {
        delay(500);
        digitalWrite(led1GPin, HIGH);
        delay(500);
        digitalWrite(led1GPin, LOW);
      }

SuGlider, 2023-02-23:
I think that `SerialFlash` library controls the `CSPIN` by its own.
Therefore, `SPI2` shall be initialized by `SPI2.begin(SCK, MISO, MOSI);` and **leave `CSPIN` not defined**.

By other hand `SPI1` with the `ADXL 355 sensor` is **correct**, because it uses `SPI Transactions` controlled by the `SPI1` peripheral.
Therefore, it must initialize `SPI1` including `GPIO_OUTPUT_CS` in `SPI1.begin()`

SuGlider, 2023-02-23:
@savagerex - 
Has `SerialFlash` library ever worked before? 
I see from the Library Code that it has no support to ESP32. Only to ESP8266.

savagerex, 2023-02-23:
@SuGlider   serialFlash library can work.

savagerex, 2023-02-23:
@SuGlider   

i have modified the code. SPI2.begin(SCK, MISO, MOSI);
it still can not work. 
led1GPin con not flashing.

savagerex, 2023-02-23:
i write the sample code.
it can work.
but when including SPI2, it can not work.

#include <SerialFlash.h>
#include <SPI.h>

//Buffer sizes
#define USB_BUFFER_SIZE      128
#define FLASH_BUFFER_SIZE    4096

//Max filename length (8.3 plus a null char terminator)
#define FILENAME_STRING_SIZE      13

//State machine
#define STATE_START      0
#define STATE_SIZE      1
#define STATE_CONTENT    2

//Special bytes in the communication protocol
#define BYTE_START      0x7e
#define BYTE_ESCAPE      0x7d
#define BYTE_SEPARATOR    0x7c

#define SensorPowerPin 17
#define led1GPin  41

//SPI Pins (these are the values on the Audio board; change them if you have different ones)
#define MOSI              35
#define MISO              37
#define SCK               36
#define CSPIN             34
//#define CSPIN           21  // Arduino 101 built-in SPI Flash

#define SPI_FREQUENCY           10 * 1000000

#if CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3
#define VSPI FSPI
#endif

//extern SPIClass SPIPORT(VSPI);

//SPIClass * vspi = NULL;
//SPIClass * hspi = NULL;

SPIClass SPI1(HSPI);

void setup(){
  Serial.begin(115200);  //Teensy serial is always at full USB speed and buffered... the baud rate here is required but ignored
  pinMode(led1GPin, OUTPUT);
  pinMode(SensorPowerPin, OUTPUT);
  digitalWrite(SensorPowerPin, HIGH);
  delay(15);

//  vspi = new SPIClass(VSPI);
//  hspi = new SPIClass(HSPI);

//  SPI.begin(SCK, MISO, MOSI, CSPIN);
//  SPI.setDataMode(SPI_MODE0);
//  SPI.setBitOrder(MSBFIRST);
//  SPI.setFrequency(SPI_FREQUENCY);

  SPI1.begin(SCK, MISO, MOSI, CSPIN);
  SPI1.setDataMode(SPI_MODE0);
  SPI1.setBitOrder(MSBFIRST);
  SPI1.setFrequency(SPI_FREQUENCY);
  
  
  //Set up SPI
  //SPI.setMOSI(MOSI);  // uncomment these if using the alternate pins
  //SPI.setMISO(MISO);
  //SPI.setSCK(SCK);
  //if (!SerialFlash.begin(CSPIN)) {
  if (!SerialFlash.begin(SPI1 , CSPIN)) {
    while (1) {
      Serial.println(F("Unable to access SPI Flash chip"));
      delay(1000);
    }
  }

  //We start by formatting the flash...
  uint8_t id[5];
  SerialFlash.readID(id);
  SerialFlash.eraseAll();

  delay(1000);

  while (!SerialFlash.ready()) {
    delay(500);
    digitalWrite(led1GPin, HIGH);
    delay(500);
    digitalWrite(led1GPin, LOW);
  }

  SerialFlashFile flashFile;

  //SerialFlash.create(filename, size);
  SerialFlash.create("123.txt", 2560000);
  flashFile = SerialFlash.open("123.txt");

  if (!flashFile) {
    Serial.println("There was an error opening the file for writing");
    return;
  }

  uint8_t str[256];

  for(int i = 0; i < 256; i++)
  {
            str[i] = i;   
  }

  for(int j = 0; j < 10000; j++)
  {  
     if(j == 9999)
     {
         str[255] = 44;
     }
     flashFile.write(str, 256);
     
  }
  flashFile.close();

  delay(1000);

  flashFile = SerialFlash.open("123.txt");

      if (!flashFile) {
        Serial.println("There was an error opening the file for writing");
        return;
      }

  uint8_t Readbuffer[256];
  for(int j = 0; j < 10000; j++)
  {  
    flashFile.read(Readbuffer, 256);
    if (j == 9999)
    {
      for(int k = 0; k < 256; k++)
      {
         Serial.println(Readbuffer[k]);
      }
    }
  }    
  flashFile.close();

  
  
}

void loop() {
  // put your main code here, to run repeatedly:

}

me-no-dev, 2023-02-23:
You should not name the second port `SPI2`, because that points to the peripheral struct. Try something like `SPI_2` or `spi2` instead

me-no-dev, 2023-02-23:
Same goes for `SPI1`

SuGlider, 2023-02-23:
@me-no-dev 

`SPI1` and `SPI2` symbols are problem only for ESP32 - as this symbol is defined in `esp32.peripherals.ld`

This project uses ESP32-S3 and from `esp32s3.peripherals.ld` I can see that S3 uses other symbols (`GPSPI2`, `GPSPI3` and `GPSPI4`).

SuGlider, 2023-02-23:
@savagerex 

What happens if you swap `SPI1` and `SPI2` in the original Sketch, like this:

``` cpp
SPIClass SPI1(HSPI);     // it was VSPI before
SPIClass SPI2(VSPI);     // it was HSPI before
```

savagerex, 2023-02-24:
if i swap SPI1 and SPI2.
result is same. it can not work.

SuGlider, 2023-02-24:
Ok. Let's get back to the original sketch posted in this issue.

So if the SPI1 and the Sensor is not executed (commented out), SPI2 works fine with SerialFlash using exactly the same sketch that you has posted in this issue?

savagerex, 2023-02-24:
SPI1(ADXL) can work

SPI2(External Flash) cann not work

savagerex, 2023-02-24:
if it is normal.

LED will flash.

while (!SerialFlash.ready()) {
delay(500);
digitalWrite(led1GPin, HIGH);
delay(500);
digitalWrite(led1GPin, LOW);
}

savagerex, 2023-02-24:
@SuGlider 

So if the SPI1 and the Sensor is not executed (commented out), SPI2 works fine with SerialFlash using exactly the same sketch that you has posted in this issue?

Ans : Right

SuGlider, 2023-02-24:
I see. 
I don't have the same hardware as used in the sketch posted in this issue.

I'll try to reproduce it by driving 2 SPI LCDs with an ESP32-S3.
Please be patient because it may take a couple days to investigate the issue.

I'll report back here as soon as I finish the testing or when I have some more questions.

Thanks.

savagerex, 2023-02-24:
OK!!!

SuGlider, 2023-02-24:
In the meantime, if possible, do you mind testing your sketch using an ESP32 (not the S3)?

savagerex, 2023-02-24:
if you need hardware, i can send to you

SuGlider, 2023-02-24:
> if you need hardware, i can send to you

Thanks. I'll try first with 2 LCDs. If I can't reproduce the issue, I may need the hardware in order to investigate it.

savagerex, 2023-02-24:
ok

SuGlider, 2023-02-24:
@savagerex - Could you please post the `BoardConfig.h` file to check what pins are being used?

savagerex, 2023-02-24:
#pragma once

#ifndef BOARD_CONFIG_H_
#define BOARD_CONFIG_H_

//#if CONFIG_FREERTOS_UNICORE
//#define ARDUINO_RUNNING_CORE 0
//#else
//#define ARDUINO_RUNNING_CORE 1
//#endif
//
//#define DISPLAY_RUNNING_CORE 1

#define SensorPowerPin 17

#define GPIO_OUTPUT_CS    10
#if 1
#define GPIO_INPUT_MISO   13     //2   
#define GPIO_OUTPUT_SCLK  12    //6
#define GPIO_OUTPUT_MOSI  11    //7
#endif

#if 0
#define GPIO_INPUT_MISO   2   
#define GPIO_OUTPUT_SCLK  6
#define GPIO_OUTPUT_MOSI  7
#endif

//#define GPIO_LED           3
//#define GPIO_ADC           0

#define DRY   8         //demo board 5

static const int spiClk = 20000000;

//Buffer sizes
#define USB_BUFFER_SIZE      128
#define FLASH_BUFFER_SIZE    4096

//Max filename length (8.3 plus a null char terminator)
#define FILENAME_STRING_SIZE      13

//State machine
#define STATE_START      0
#define STATE_SIZE      1
#define STATE_CONTENT    2

//Special bytes in the communication protocol
#define BYTE_START      0x7e
#define BYTE_ESCAPE      0x7d
#define BYTE_SEPARATOR    0x7c

#define led1GPin  41

//SPI Pins (these are the values on the Audio board; change them if you have different ones)
#define MOSI              35
#define MISO              37
#define SCK               36
#define CSPIN             34
//#define CSPIN           21  // Arduino 101 built-in SPI Flash

#define SPI_FREQUENCY           10 * 1000000

//#ifdef DEBUG_ESP_HTTP_CLIENT
//#ifdef DEBUG_ESP_PORT
//#define DEBUG_HTTPCLIENT(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ )
//#endif
//#endif
//
//#ifndef DEBUG_HTTPCLIENT
//#define DEBUG_HTTPCLIENT(...)
//#endif

#endif /* BOARD_CONFIG_H_ */

SuGlider, 2023-02-24:
@savagerex - Could you also compile your sketch using `Core Debug Level:"Verbose"` (from the IDE Tools menu option), run it and then post the Serial Monitor output for analysis?

Thanks!

savagerex, 2023-02-24:
.ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
[   203][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   236][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[   237][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   237][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
Connecting..[   413][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: vivo Y21, BSSID: ea:47:d5:13:ff:f1, Channel: 6, Auth: WPA2_PSK
[   414][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   456][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.179.244
[   457][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   460][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.179.244, MASK: 255.255.255.0, GW: 192.168.179.18
WiFi connected - IP address: 192.168.179.244

connecting...
connected!
DeviceId = AD
Flash Ready
[ 47099][D][WiFiClient.cpp:546] connected(): Disconnected: RES: 0, ERR: 128
test8
**************wifi Transfer Time*************19
test9
test8
**************wifi Transfer Time*************19
test9
test8
**************wifi Transfer Time*************20
test9
test8
**************wifi Transfer Time*************20
test9
test8
**************wifi Transfer Time*************19
test9
test8
**************wifi Transfer Time*************19
test9

SuGlider, 2023-03-03:
@savagerex - What is the part number of the ESP32-S3 chip that this project uses (ESP32-S3 | ESP32-S3FN8 | ESP32-S3R2 | ESP32-S3R8 | ESP32-S3R8V)?

savagerex, 2023-03-03:
@SuGlider 

 ESP32-S3FN8

SuGlider, 2023-03-03:
@savagerex - We are still investigating this issue. The initial testing I did trying to use the ESP32 S3 to drive 2 TFT displays using FSPI and HSPI has failed. I couldn't make it work using HSPI. Another colleague from Espressif will also test it next week.

In the meantime, I'd like to suggest another SPI Flash Library:
https://github.com/adafruit/Adafruit_SPIFlash
It supports ESP32 using SPI. If you could try some of the examples using the correct GPIOs and each S3 SPI (FSPI and HSPI) and report back to us, it would help to come up with a diagnostic for the issue.

Thanks!

SuGlider, 2023-03-04:
@savagerex @me-no-dev - some news. I had a bad display in the first testing. So after replacing it, I got both SPI from S3 (HSPI and FSPI) to work together using Arduino 2.0.7. 

Therefore there is no problem with ESP32-S3 Arduino 2.0..7 and Two SPI working in the same sketch.
I used the same ESP32-S3-Mini-1 module.

[image]

-------

[image]

SuGlider, 2023-03-04:
This is the very simple and basic Arduino skecth that I used to drive those 2 displays using HSPI and FSPI:

``` cpp
//This application does not rely on any libraries and it is for ST7735S
//This program is a demo of clearing screen to display black,white,red,green,blue.
#include <SPI.h>

SPIClass LCD_SPI1(FSPI);  // FSPI (S3) - HSPI (Both) - VSPI (ESP32)
SPIClass LCD_SPI2(HSPI);  // FSPI (S3) - HSPI (Both) - VSPI (ESP32)

#if CONFIG_IDF_TARGET_ESP32
// Pinos do ESP32 WROVER-E
#define CS   25
#define RS   27
#define RESET  26
#define SDA  12
#define SCK  14
#elif CONFIG_IDF_TARGET_ESP32S2
#elif CONFIG_IDF_TARGET_ESP32S3
#define RESET1  1
#define CS1     2
#define RS1     3
#define SDA1    4
#define SCK1    5

#define RESET2  10
#define CS2     11
#define RS2     12
#define SDA2    13
#define SCK2    14
#elif CONFIG_IDF_TARGET_ESP32C3
#endif

#define LED  -1   //if you don't need to control the LED pin,you should set it to -1 and set it to 3.3V

void Lcd_Writ_Bus(unsigned char d)
{
   LCD_SPI1.transfer(d);
   LCD_SPI2.transfer(d);
}

void Lcd_Write_Com(unsigned char VH)  
{   
  digitalWrite(RS1,LOW);   // LOW D/C is REGISTER
  digitalWrite(RS2,LOW);   // LOW D/C is REGISTER
  Lcd_Writ_Bus(VH);
}

void Lcd_Write_Data(unsigned char VH)
{
  digitalWrite(RS1,HIGH);  // HIGH D/C is DATA
  digitalWrite(RS2,HIGH);  // HIGH D/C is DATA
  Lcd_Writ_Bus(VH);
}

void Lcd_Write_Com_Data(unsigned char com,unsigned char dat)
{
  Lcd_Write_Com(com);
  Lcd_Write_Data(dat);
}

void Address_set(unsigned int x1,unsigned int y1,unsigned int x2,unsigned int y2)
{
  Lcd_Write_Com(0x2a);
  Lcd_Write_Data(x1>>8);
  Lcd_Write_Data(x1);
  Lcd_Write_Data(x2>>8);
  Lcd_Write_Data(x2);
  Lcd_Write_Com(0x2b);
  Lcd_Write_Data(y1>>8);
[... 221 lines cut ...]
  digitalWrite(SCK2, HIGH);
  
  SPI_Init();
  Lcd_Init();
  Serial.begin(115200);
  delay(500);
  Serial.println("\n\nDisplay 128x160 1.8 inches Hardware SPI\n\n");
  
}

void loop()
{  
   LCD_Clear(0xf800);
   LCD_Clear(0x07E0);
   LCD_Clear(0x001F);
   LCD_Clear(0x0); 
  for(int i=0;i<300;i++)
  {
    Rect(random(127),random(159),random(127),random(159),random(65535)); // rectangle at x, y, with, hight, color
  }
  
//  LCD_Clear(0xf800);
}

```

savagerex, 2023-03-06:
@SuGlider 

after i update to 2.0.7. 

it stiil can not work.

LED does not flash.

while (!SerialFlash.ready()) {
delay(500);
digitalWrite(led1GPin, HIGH);
delay(500);
digitalWrite(led1GPin, LOW);
}

SuGlider, 2023-03-06:
@savagerex - 
Can you try this other SPI Flash Library:
https://github.com/adafruit/Adafruit_SPIFlash

There is an example in the Library that will try to identify the Flash size and Chip:
https://github.com/adafruit/Adafruit_SPIFlash/blob/master/examples/flash_info/flash_info.ino

savagerex, 2023-03-06:
@SuGlider 

Do you have sample code about Read / Write Flash in (https://github.com/adafruit/Adafruit_SPIFlash) ???

savagerex, 2023-03-06:
@SuGlider  I write the sample code, as below.

it still can not work

****** ino file

#include <SPI.h>
#include <SdFat.h>
#include <Adafruit_SPIFlash.h>

// for flashTransport definition
#include "flash_config.h"

Adafruit_SPIFlash flash(&flashTransport);

FatVolume fatfs;
File32 myFile;

#define SensorPowerPin 17

#define MOSI              35
#define MISO              37
#define SCK               36
#define CSPIN             34

#define SPI_FREQUENCY           10 * 1000000

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(SensorPowerPin, OUTPUT);
  digitalWrite(SensorPowerPin, HIGH);
  delay(15);

    SPIOne.begin(SCK, MISO, MOSI, CSPIN);
    //SPI2.begin(SCK, MISO, MOSI);
    SPIOne.setDataMode(SPI_MODE0);
    SPIOne.setBitOrder(MSBFIRST);
    SPIOne.setFrequency(SPI_FREQUENCY);

    delay(1000);

    Serial.println("Adafruit Serial Flash Info example");
    flash.begin();
    Serial.println("Test0");
    // Open file system on the flash
    if ( !fatfs.begin(&flash) ) {
      Serial.println("Error: filesystem is not existed. Please try SdFat_format example to make one.");
      while(1)
      {
        yield();
        delay(1);
      }
    }
  
    Serial.println("initialization done.");
  
    // open the file. note that only one file can be open at a time,
    // so you have to close this one before opening another.
    myFile = fatfs.open("test.txt", FILE_WRITE);
  
    // if the file opened okay, write to it:
    if (myFile) {
      Serial.print("Writing to test.txt...");
      myFile.println("testing 1, 2, 3.");
      // close the file:
      myFile.close();
      Serial.println("done.");
    } else {
      // if the file didn't open, print an error:
      Serial.println("error opening test.txt");
    }
  
    // re-open the file for reading:
    myFile = fatfs.open("test.txt");
    if (myFile) {
      Serial.println("test.txt:");
  
      // read from the file until there's nothing else in it:
      while (myFile.available()) {
        Serial.write(myFile.read());
      }
      // close the file:
      myFile.close();
    } else {
      // if the file didn't open, print an error:
      Serial.println("error opening test.txt");
    }

}

void loop() {
  // put your main code here, to run repeatedly:

}

****** flash_config.h

#ifndef FLASH_CONFIG_H_
#define FLASH_CONFIG_H_
#include <SPI.h>
SPIClass SPIOne(HSPI);

// Un-comment to run example with custom SPI and SS e.g with FRAM breakout
//#define CUSTOM_CS   34      //A5
//#define CUSTOM_SPI  SPIOne  //SPI

#if defined(CUSTOM_CS) && defined(CUSTOM_SPI)
Adafruit_FlashTransport_SPI flashTransport(CUSTOM_CS, CUSTOM_SPI);

#elif defined(ARDUINO_ARCH_ESP32)

// ESP32 use same flash device that store code for file system.
// SPIFlash will parse partition.cvs to detect FATFS partition to use
Adafruit_FlashTransport_ESP32 flashTransport;

#elif defined(ARDUINO_ARCH_RP2040)

// RP2040 use same flash device that store code for file system. Therefore we
// only need to specify start address and size (no need SPI or SS)
// By default (start=0, size=0), values that match file system setting in
// 'Tools->Flash Size' menu selection will be used.
Adafruit_FlashTransport_RP2040 flashTransport;

// To be compatible with CircuitPython partition scheme (start_address = 1 MB,
// size = total flash - 1 MB) use const value (CPY_START_ADDR, CPY_SIZE) or
// subclass Adafruit_FlashTransport_RP2040_CPY. Un-comment either of the
// following line:
//  Adafruit_FlashTransport_RP2040
//    flashTransport(Adafruit_FlashTransport_RP2040::CPY_START_ADDR,
//                   Adafruit_FlashTransport_RP2040::CPY_SIZE);
//  Adafruit_FlashTransport_RP2040_CPY flashTransport;

#else

// On-board external flash (QSPI or SPI) macros should already
// defined in your board variant if supported
// - EXTERNAL_FLASH_USE_QSPI
// - EXTERNAL_FLASH_USE_CS/EXTERNAL_FLASH_USE_SPI

#define EXTERNAL_FLASH_USE_CS  34
#define EXTERNAL_FLASH_USE_SPI SPIOne

#if defined(EXTERNAL_FLASH_USE_QSPI)

Adafruit_FlashTransport_QSPI flashTransport;

#elif defined(EXTERNAL_FLASH_USE_SPI)

Adafruit_FlashTransport_SPI flashTransport(EXTERNAL_FLASH_USE_CS,
                                           EXTERNAL_FLASH_USE_SPI);

#elif defined(__AVR__)

// Use stand SPI/SS for avr port.
// Note cache will be disable due to lack of memory.

Adafruit_FlashTransport_SPI flashTransport(SS, SPI);

#else
#error No (Q)SPI flash are defined for your board !
#endif

#endif

#endif

****** \Documents\Arduino\libraries\Adafruit_SPIFlash-master\src\Adafruit_SPIFlashBase
****** modified part. add(W25Q256JV)

static const SPIFlash_Device_t possible_devices[] = {
    // Main devices used in current Adafruit products
    GD25Q16C, GD25Q32C, GD25Q64C, S25FL116K, S25FL216K,

    // Only a handful of production run
    W25Q16FW, W25Q64JV_IQ,

    // Fujitsu FRAM
    MB85RS64V, MB85RS1MT, MB85RS2MTA, MB85RS4MT,

    // Other common flash devices
    W25Q16JV_IQ, W25Q32JV_IQ, AT25SF041, AT25DF081A, W25Q256JV};    <-----******

SuGlider, 2023-03-07:
I see that this is the `SdFat_ReadWrite` example.
Let's try first using just the default Arduino `SPI` instead of HSPI just to check if it works fine.
Let me know if it works.

``` cpp
****** flash_config.h

#ifndef FLASH_CONFIG_H_
#define FLASH_CONFIG_H_
#include <SPI.h>
SPIClass SPIOne(HSPI);

// Un-comment to run example with custom SPI and SS e.g with FRAM breakout
#define CUSTOM_CS 34 //A5
#define CUSTOM_SPI SPI

#if defined(CUSTOM_CS) && defined(CUSTOM_SPI)
Adafruit_FlashTransport_SPI flashTransport(CUSTOM_CS, CUSTOM_SPI);

```

SuGlider, 2023-03-07:
There is also this other Library: https://github.com/Marzogh/SPIMemory

Based on the example from https://github.com/Marzogh/SPIMemory/blob/master/examples/readWriteString/readWriteString.ino
It may work with the code below (I have no external SPI flash module to make sure it runs --- sorry.).

``` cpp
/*
#include<SPIMemory.h>

uint32_t strAddr;

#define BAUD_RATE 115200
#define RANDPIN A0

#define MOSI 35
#define MISO 37
#define SCK 36
#define CSPIN 34

SPIFlash flash;   // it will use the default Arduino SPI port

void setup() {
  Serial.begin(BAUD_RATE);

  SPI.begin(SCK, MISO, MOSI, CSPIN);   // set SPI pins

  flash.begin();

  randomSeed(analogRead(RANDPIN));

  strAddr = random(0, flash.getCapacity());
  String inputString = "This is a test String";
  flash.writeStr(strAddr, inputString);

  Serial.print(F("Written string: "));
  Serial.println(inputString);
  Serial.print(F("To address: "));
  Serial.println(strAddr);

  String outputString = "";
  if (flash.readStr(strAddr, outputString)) {
    Serial.print(F("Read string: "));
    Serial.println(outputString);
    Serial.print(F("From address: "));
    Serial.println(strAddr);
  }

  while (!flash.eraseSector(strAddr));
}

void loop() {
}
```

savagerex, 2023-03-07:
After i modified the code
it can not work

#define CUSTOM_CS 34 //A5
#define CUSTOM_SPI SPI

#if defined(CUSTOM_CS) && defined(CUSTOM_SPI)
Adafruit_FlashTransport_SPI flashTransport(CUSTOM_CS, CUSTOM_SPI);

SuGlider, 2023-03-07:
> After i modified the code it can not work
> 
> #define CUSTOM_CS 34 //A5 #define CUSTOM_SPI SPI
> 
> #if defined(CUSTOM_CS) && defined(CUSTOM_SPI) Adafruit_FlashTransport_SPI flashTransport(CUSTOM_CS, CUSTOM_SPI);

OK. 

As a last attempt, please try the other proposed library above (https://github.com/espressif/arduino-esp32/issues/7881#issuecomment-1457514402). Thanks.

SuGlider, 2023-03-09:
@savagerex - please let me know if the Marzogh (https://github.com/espressif/arduino-esp32/issues/7881#issuecomment-1457514402) has helped your project.

Thanks.

savagerex, 2023-03-13:
@SuGlider 

it's work. THX

SuGlider, 2023-03-13:
@savarex Thanks for the return. Closing as completed.

closed 2023-03-13 by SuGlider