espressif/arduino-esp32 #7882: Arduino ESP32 v2.0.7 can not upload sketches on Adafruit Feather ESP32-S3

idea--list, 2023-02-23:
### Board

Adafruit Feather ESP32-S3 No PSRAM

### Device Description

Adafruit Feather ESP32-S3 No PSRAM

### Hardware Configuration

Default config of the board

### Version

v2.0.7

### IDE Name

Arduino IDE 2.0.3

### Operating System

Windows 10 x64 Pro

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Just upgraded to Arduino ESP32 v2.0.7 and noticed that none of the sketches can be uploaded to my Adafruit Feather ESP32-S3 board. Code compiles just like before the upgrade, but the upload process fails with error status 2. Tried to burn the bootloader via the menu option but that did not make any difference.

Downgrading to Arduino ESP32 v2.0.6 solves the issue.

### Sketch

```cpp
Any sketch
```

### Debug Message

```plain
Sketch uses 282073 bytes (13%) of program storage space. Maximum is 2097152 bytes.
Global variables use 27636 bytes (8%) of dynamic memory, leaving 300044 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5
Serial port COM16
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 60:55:f9:f5:6b:b8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00054fff...
Flash will be erased from 0x00410000 to 0x00438fff...
Compressed 22624 bytes to 14245...
Writing at 0x00000000... (100 %)

A fatal error occurred: Packet content transfer stopped (received 1 bytes)
Failed uploading: uploading error: exit status 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-02-23:
Tried a lower baud rate? Try 115200

me-no-dev, 2023-02-23:
@Jason2866 the board uses native USB to flash and debug, so baudrate does not matter really.

@ladyada have you noticed such an issue? We have updated esptool, which was supposed to improve flashing over the native USB port.

Jason2866, 2023-02-23:
@me-no-dev  Ahh yes, maybe the latest update to lower flash clock from 240Mhz to 160Mhz which is done AFTER the v4.5 release does solve this issue.

svdrummer, 2023-02-23:
Lillygo TdisplayS3 Same issue.
All config set to default settings.
I have tried the esp32s3 dev module and the new TTgo displayS3
Issues uploading. Downgraded to the last version and all is okay.

me-no-dev, 2023-02-23:
What operating system are you guys on? I am able to upload fine to S3 over the integrated USB (not UART) with the same settings as @svdrummer

Jason2866, 2023-02-23:
@me-no-dev  The posts with this issue i have seen where always Win10/11.

me-no-dev, 2023-02-23:
I wonder what to do to fix this... since a new version of esptool is not released, maybe just roll back to 4.2.1 and ask you guys to reinstall 2.0.7?

Jason2866, 2023-02-23:
Or ask @radimkarnis to release 4.5.1?

dobairoland, 2023-02-24:
> Or ask @radimkarnis to release 4.5.1?

Alright, we will soon.

radimkarnis, 2023-02-24:
@svdrummer @idea--list could I ask you guys to verify if the [latest esptool](https://github.com/espressif/esptool) works? We have implemented a potential fix. You can download the patched binary with the latest fix [here](https://github.com/espressif/esptool/actions/runs/4194891723) (or in the link provided by @Jason2866).

idea--list, 2023-02-25:
@radimkarnis 
Updated to Arduino ESP32 v2.0.7 again and then replaced esptool.exe with that found in your link above.

It began to upload the sketch but aborted at 70% with status code 1 this time. I also noticed Arduino IDE had a hard time determining the COM port of the USB connection. While windows 10 recognized the board easily, though with the label FTHRS3BOOT.

After holding down Boot button then Reset button and releasing those in reverse order uploads did not even start and i got error status 2. This made me wonder as this combination worked always ever since i have this board.

Finally i cut all power then hold Boot button down while applying power to the board. After doing so the updated esptool successfully uploaded the sketch. Since that both Arduino IDE recognizes the COM port easily and also the combination of Boot and Reset buttons work again just like i was used to it.

So the latest esptool solves the issue even though the board seems to behave a bit strange on the first upload cycle with the new esptool.

Jason2866, 2023-02-25:
@idea--list  Imho it is more Windows behaves strange. Every other OS does not have this.

idea--list, 2023-02-25:
@Jason2866 
IMHO it is completely indifferent if it is because of windows or not. At least as far ESP-IDF can be used on windows.

Jason2866, 2023-02-25:
Which esptool version is used in IDF where do you have no issues with Windows?

idea--list, 2023-02-25:
@Jason2866 
I understand for some reason windows is not your favorite OS. It is up to you why. You might be even right with that.

However until Arduino IDE, esp32 arduino core and esp-idf can be used on the windows OS it not the fault of windows if any tool has not been tested well enough on all supported OSes thoroughly enough. In software one can handle almost any shortcomings of any platforms. There are only few things that really can not be mitigated in a specific OS. Uploading a binary file over USB is definitely not something impossible to implement. It has been working before and also the latest version of esptool seems to get the job done as i reported. Just wrote some feedback for those folks who maintain esptool what i experienced as a user having a specific board and trying to program it with arduino-esp32 core on a windows based computer.

I did not have any intention to start a debate about which OS behaves different from other Oses ;-)

Jason2866, 2023-02-25:
Just mentioned that the behaviour you see is probably not caused from esptool.
Used Windows since v3.11 and stopped using when Windows 11 arrived. The reason for change i was unhappy of the mass of updates Windows is doing and i had to fix stuff not working anymore after updates... COM ports handling IS a mess in Windows.

radimkarnis, 2023-02-25:
@idea--list ok so let me explain what happened:

> Arduino IDE had a hard time determining the COM port of the USB connection. While windows 10 recognized the board easily, though with the label FTHRS3BOOT.

`FTHRS3BOOT` - This means there was a pre-flashed binary, that remaps the USB to work as a storage device to easily upload CircuitPython, [see here](https://learn.adafruit.com/adafruit-esp32-s3-feather/circuitpython). Because of this, flashing with esptool wasn't working. In this case, you need to switch to the download mode manually and overwrite the binary, which you attempted -

> After holding down Boot button then Reset button and releasing those in reverse 

You need to keep holding the BOOT button, while you press the RST button (meaning you have to release the BOOT button the last). Anyway, you managed to so the manual reset by power-cycling the board while holding the BOOT btn -

>  i cut all power then hold Boot button down while applying power to the board. After doing so the updated esptool successfully uploaded the sketch.

Yes, at this point you successfully entered the download mode, meaning the CircuitPython binary stopped running and the pins got remapped back to normal. Now esptool can finally start flashing. You say it did without any issue, that is a confirmation the patched esptool version I provided works. Thanks!

> Since that both Arduino IDE recognizes the COM port easily and also the combination of Boot and Reset buttons work again just like i was used to it.

You've overwritten the CircuitPython binary, so the pins do not get remapped anymore. Great, now all works as intended.

So to sum up - issue is fixed. But there was some confusion caused by the pre-loaded app, which switched USB to a mass-storage device `FTHRS3BOOT`.

idea--list, 2023-02-25:
@radimkarnis 
I know these boards ship with CircuitPython. Tried it once but am not a fan of that. I thought i have overwritten circuitPython binary long ago. Anyway thanks for the effort and yes the issue is fixed as far as i am concerned.

svdrummer, 2023-02-26:
> @svdrummer @idea--list could I ask you guys to verify if the [latest esptool](https://github.com/espressif/esptool) works? We have implemented a potential fix. You can download the patched binary [...]

not sure how to upgrade esptool. I have just reverted back until the fix is available in the boards upgrade

me-no-dev, 2023-03-01:
Hi guys! Could you please uninstall and re-install 2.0.7? It should hopefully pull esptool 4.5.1

OmerFarukOruc, 2023-03-01:
I was trying to deal with my esp32-s3 lolin s3 pro and just saw this successfully solved same problems for me. Thanks for the solution. @idea--list

idea--list, 2023-03-01:
@me-no-dev 
Here is what i tried just now:
1. Launched Arduino IDE v2.0.3 and uninstalled the esp32 core.
2. Closed Arduino IDE v2.0.3 and manually deleted esp32 inside Arduino15/packages folder
3. Launched Arduino IDE v2.0.3 again, installed esp32 core v2.0.7

After that now all sketches upload again as expected and i see now 4.5.1 version of the esptool has been pulled

me-no-dev, 2023-03-01:
Thanks for confirming @idea--list

closed 2023-03-01 by me-no-dev

CarLoOSX, 2024-01-12:
happened the same to me in Mac with Platform.io and had to do this button press combination.

PR #7838 (2023-02-14): Sets correct code for UART baud rate detection
## Description of Change
Excludes ESP32S3 and ESP32C3 in the UART HAL code that detects baud rate.
This feature only works for ESP32 and ESP32S2

## Tests scenarios
ESP32, ESP32S2 and ESP32S3.

``` cpp
// This sketch works correctly with ESP32 and ESP32S2 only, so far.
// ESP32C3 and ESP32S3 seems to fail with baud rate detection
void setup() {
  Serial.begin(115200);
  Serial.println("\nTesting baud rate detection\n");
}

void loop() {
  Serial.println("Started detection for next 20 seconds.");
  Serial.println("Use the Serial Monitor. Change the [... cut]