espressif/arduino-esp32 #7884: SSL - Allocation Memory Failed

Hoang-Nam2711, 2023-02-24:
### Board

ESP32 

### Device Description

DevKitC, plain module on module breadboard. 
https://akizukidenshi.com/catalog/g/gM-15674/

### Hardware Configuration

Nothing attached to the development board

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22.04.1 LTS

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi,
I am using BLE and HTTPS at the same time in ESP32.
After receiving data from the BLE device, I tried BLE::deinit following this suggestion https://github.com/espressif/arduino-esp32/issues/3449#issuecomment-649657370 to reduce heap for HTTPS connection.
After a few successful connections (3 times), the heap is insufficient for an HTTPS connection for the 4th time. 

In the sketch, the following step is:
  1. BLE Scan and subscribe for service and characteristic UUID
  2. Send command and wait for notification 
  3. BLE::deinit(false) for free heap 
  4. Connect to the server by HTTPS
  5. Disconnect to the server and BLE rescan

### Sketch

```cpp
`
      #include "BLEDevice.h"
      #include <WiFiClientSecure.h>
      #include <HTTPClient.h>
      
      void setup() {
        Serial.begin(115200);
      
        // WiFi.mode(WIFI_STA);
        WiFi.begin(WIFI_SSID,WIFI_PWD);
        while(WiFi.status() != WL_CONNECTED){
          delay(500);
          Serial.print(".");
        }
        Serial.println("");
        Serial.println("WiFi Connected");
      
        Serial.println("Starting Arduino BLE Client application...");
        Serial.println("==========================================================---");
        // BLEDevice::init("SmartValve");
        // while (!client.connect(serverURL, 56789))
        //   Serial.println("Connection failed!");
      }
      
      void loop() {
        if(initBLE){
          BLEDevice::init("SmartValve");
          BLEScan* pBLEScan = BLEDevice::getScan();
          pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
          pBLEScan->setInterval(1349);
          pBLEScan->setWindow(449);
          pBLEScan->setActiveScan(true);
          pBLEScan->start(5, false);
        }
      
        if(bleDoConnect){
          initBLE = false;
          Serial.print("Connect To Server : ");
          Serial.println(myDevice->getAddress().toString().c_str());
          if(connectToServer()){
            Serial.println("We are now connected to the BLE Server");      
          }else{
            initBLE=true;
            Serial.println("We have failed to connect to the server; there is nothing more we will do");
          }
          bleDoConnect = false;
        }
      
        if(bleConnected){
          byte newValue[20] = {0xXX,0xXX,0xXX};
          byte writeDataF[20] = {0xXX,0xXX,0xXX};
          byte writeDataS[20] = {0xXX,0xXX,0xXX};
          byte end[20] ={0xXX,0xXX,0xXX};
          pRemoteCommandCharacteristic->writeValue(newValue,sizeof(newValue),true);
          pRemoteWriteDataCharacteristic->writeValue(writeDataF,sizeof(writeDataF),false);
          pRemoteWriteDataCharacteristic->writeValue(writeDataS,sizeof(writeDataS),false);
          pRemoteEndCharacteristic->writeValue(end,sizeof(end),false);
          serverConnect = true;
          bleConnected=false;
        }
      
        //delay 2s for waiting received all reply of valve
        delay(2000);
        if(serverConnect){
          Serial.println("HEREEE");
          Serial.printf("\n****Before BLEDevice::deinit ESP.getFreeHeap() %u\n",ESP.getFreeHeap());
          esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
          BLEDevice::deinit(false);
          Serial.printf("\n****After BLEDevice::deinit ESP.getFreeHeap() %u\n",ESP.getFreeHeap());
          //send reply to server
          WiFiClientSecure *client = new WiFiClientSecure;
          if(client) {
            client -> setCACert(caCert);
            client -> setCertificate(clientCert);
            client -> setPrivateKey(clientPrivateKey);
      
            {
              // Add a scoping block for HTTPClient https to make sure it is destroyed before WiFiClientSecure *client is 
              HTTPClient https;
      
              //Setting Payload
              char payload[500];
              snprintf(payload,500,"{\"place\":\"%s\" , \"user\":\"%s\" , \"device\":\"%s\" , \"data\" : \"%s\"}",place,user,device,valveReply);
              Serial.println(payload);
            
              Serial.print("[HTTPS] begin...\n");
              if (https.begin(*client, serverURL)) {  // HTTPS
                https.setUserAgent("ESP32SendCommand");
                https.addHeader("Host","https://XXX");
                https.addHeader("Connection","close");
                https.addHeader("Content-Type","application/json; charset=utf-8");
                https.addHeader("Accept","application/json");
                Serial.print("[HTTPS] POST...\n");
                // start connection and send HTTP header
                int httpCode = https.POST(payload);
                // int httpCode = https.GET();
          
                // httpCode will be negative on error
                if (httpCode > 0) {
                  // HTTP header has been send and Server response header has been handled
                  Serial.printf("[HTTPS] POST... code: %d\n", httpCode);
          
                  // file found at server
                  if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
                    String payload = https.getString();
                    Serial.println(payload);
                  }
                } else {
                  Serial.printf("[HTTPS] POST... failed, error: %s\n", https.errorToString(httpCode).c_str());
                }
          
                https.end();
              } else {
                Serial.printf("[HTTPS] Unable to connect\n");
              }
      
              // End extra scoping block
            }
          
            delete client;
          } else {
            Serial.println("Unable to create client");
          }
          initBLE = true;
          valveReply = "";
          serverConnect = false;
        }
      }

`
```

### Debug Message

```plain
[ 78237][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-25:
@Hoang-Nam2711 

Based on the BLE source code, it seem that it should be `BLEDevice::deinit(true);`, instead of `false`.

https://github.com/espressif/arduino-esp32/blob/bb8c8559b62a0d2e888d51bf12bd50cc070f0b4b/libraries/BLE/src/BLEDevice.cpp#L649-L663

Hoang-Nam2711, 2023-02-27:
@SuGlider 
Thanks for your reply,
I tried setting it to True, but the second time connection, it returns this error. 
 `[ 20614][E][BLEScan.cpp:402] start(): esp_ble_gap_set_scan_params: err: 259, text: Unknown ESP_ERR error`
Even it looped through the BLE setting parameter program
`
          pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());

          pBLEScan->setInterval(1349);

          pBLEScan->setWindow(449);

          pBLEScan->setActiveScan(true);

          pBLEScan->start(5, false);
`

SuGlider, 2023-03-02:
@Hoang-Nam2711 - The sketch is incomplete, but I'll try to suggest something.

I think that you may try to change the strategy of the sketch.
Instead of initializing BLE and WiFi in the `loop()` several times, you may initialize it only once in `setup()`.
Use the callbacks to get the information from both (which I guess may be already in use).

Start BLE advertising only when necessary, but keep it initialized once for all the time.
Just use the HTTP client to send the information whenever necessary, but also initialize it just once.
Do not release any (BLE/WiFi). Just use the APIs to do whatever you may need.

The way it is coded, it fragments the heap too much and then it ends up with lack of memory.

Hoang-Nam2711, 2023-03-03:
@SuGlider 
Sorry for the incomplete sketch. It just some subscribe BLE Service, Characteristic function, Connect To BLE Device function and receive notification from the device and I don't think there is any problem in here. In case you need it, I will post it under this comment.

I tried to init BLE and Wifi the first time I build this sketch. But it can not send HTTPS even the first time (the heap before HTTPS sending is only about 4kB). The debug is message is : 

> [ 78237][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed

So I tried this recommendation [](https://github.com/espressif/arduino-esp32/issues/3449#issuecomment-649657370). Init BLE -> run BLE -> deinit BLE for getting more heap -> start HTTPS -> send -> stop HTTPS. 

```cpp
static void notifyCallback(BLERemoteCharacteristic* pBLERemoteCharacteristic, uint8_t* pData, size_t length, bool isNotify){
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
  Serial.print("data: ");
  Serial.print((char*)pData);
  Serial.println("");
  
  if(pBLERemoteCharacteristic->getUUID().toString() == resDataUUID.toString()){
    for(int i=0; i<length; i++){
      valveReply+=(char)pData[i];
    }
    Serial.print("Valve Reply:");
    Serial.println(valveReply);
  }

}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    bleConnected = false;
    Serial.println("onDisconnect");
  }

};

bool connectToServer(){
  Serial.println("HEREEEEEEEEEEEE");
  Serial.printf("\n****Before createClient ESP.getFreeHeap() %u\n",ESP.getFreeHeap());
  BLEClient* pClient = BLEDevice::createClient();
  Serial.printf("\n****Before createClient ESP.getFreeHeap() %u\n",ESP.getFreeHeap());
  Serial.println(" - Created client");

  // ???
  pClient->setClientCallbacks(new MyClientCallback());

  // Connect to the BLE Server.
  pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
  Serial.println(" - Connected to server");
  //Set Maximum Transmission Unit 
  pClient->setMTU(517); //set client to request maximum MTU from server (default is 23 otherwise)

  // Obtain a reference to the service we are after in the remote BLE server.
  BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    Serial.println(serviceUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our service");

  // Obtain a reference to the characteristic in the service of the remote BLE server.
      pRemoteCommandCharacteristic = pRemoteService->getCharacteristic(commandUUID);
      if (pRemoteCommandCharacteristic == nullptr) {
        Serial.print("Failed to find command characteristic UUID: ");
[... 63 lines cut ...]
      }
  
  bleConnected = true;
  return true;
}

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print(".");
    

    if (advertisedDevice.getAddress().toString() == deviceAdr) {
      BLEDevice::getScan()->stop();
      myDevice = new BLEAdvertisedDevice(advertisedDevice);
      // Serial.print("Forming a connection to ");
      // Serial.println(myDevice->getAddress().toString().c_str());
      bleDoConnect = true;
      bleDoScan = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

Hoang-Nam2711, 2023-03-16:
I changed to NimBLEDevice library and everything worked

closed 2023-03-16 by Hoang-Nam2711