espressif/arduino-esp32 #7885: ESP32 C3 Loop runs slow with Serial not connected

setrin, 2023-02-24:
### Board

Wemos C3 Pico

### Device Description

Wemos C3 Pico with WS2812B LEDs powered through MOSFET and keypad

### Hardware Configuration

GPIO21 is connected to MOSFET which drives power to LEDs
GPIO7 is connected to DIN on LEDs
GPIO 5, 4, 0 and 1 is connected to Keypad Matrix rows
GPIO 8,10 and 20 is connected to Keypad Matrix columns

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Loop function runs "slow" if Serial is not connected to Serial monitor on PC.

ESP after bootup reacts to key presses on Keypad and lights up corresponding LED for indication of keypress.
For power saving i'm using inactivity timer for 30 seconds after which ESP goes to deep sleep. I have attached external interrupts to keypad so i can wake it up. After wake up Loop fn runs "slow". Before deep sleep everything is snappy. I can press how many key i want and there is basically no delay. After deep sleep I can press keys normally, but after 8. key press there starts to be delay on key presses. I can see it with LED indicator.

If i disable Serial.begin and remove all prints, everything works as expected.

I have updated Arduino-ESP32 board library recently. I can't remember this issue before the update to 2.0.7

### Sketch

```cpp
#define DEBUG false  //set to true for debug output, false for no debug output
#define DEBUG_SERIAL if(DEBUG)Serial

#include <Adafruit_NeoPixel.h>
#include <EEPROM.h>
#include <Keypad.h>

unsigned long inactivityMillis = millis();  // will store last time activity was recorded
const long inactivityTime = 30000;  // time at which should ESP to go deepsleep (milliseconds)

const uint8_t ledBrightness = 64;
#define LEDS_POWER_PIN 21
#define LEDS_PIN 7
#define NUM_OF_LEDS 5                                                                                                                                                   
Adafruit_NeoPixel pixels(NUM_OF_LEDS, LEDS_PIN, NEO_RGB + NEO_KHZ800);

const byte ROWS = 4; //four rows
const byte COLS = 3; //three columns
const int NUM_OF_BUTTONS = ROWS * COLS;
char keys[ROWS][COLS] = {
    {'1','2','3'},
    {'4','5','6'},
    {'7','8','9'},
    {'*','0','#'}
  };
byte rowPins[ROWS] = {5, 4, 0, 1}; // {6, 8, 10, 20}; //connect to the row pinouts of the kpd
byte colPins[COLS] = {8, 10, 20}; // {5, 4, 0}; //connect to the column pinouts of the kpd

Keypad kpd = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );

void setup() {
  #if DEBUG == true
    Serial.begin(115200);
  #endif

  InitializeIO();

  pinMode(LEDS_POWER_PIN, OUTPUT);
  digitalWrite(LEDS_POWER_PIN, HIGH);
  pixels.begin();

  // Reset all LEDs
    pixels.clear();
    pixels.show();

  inactivityMillis = millis();
}

void loop() {
    if (kpd.getKeys()){
      for (int i = 0; i < LIST_MAX; i++){
        if ( kpd.key[i].stateChanged ) {
          inactivityMillis = millis();
          int keyIdx = getKeyIndex(kpd.key[i].kchar);
          int keyLed = keyIdx > 4 ? 4 : keyIdx;
          if (kpd.key[i].kstate == PRESSED){
            DEBUG_SERIAL.print("Key ");
            DEBUG_SERIAL.print(keyIdx);
            DEBUG_SERIAL.println(" PRESSED.");

            pixels.setPixelColor(keyLed, pixels.Color(0, 0, 128));
          } else if (kpd.key[i].kstate == RELEASED) {
            DEBUG_SERIAL.print("Key ");
            // DEBUG_SERIAL.print(kpd.key[i].kchar);
            DEBUG_SERIAL.print(keyIdx);
            DEBUG_SERIAL.println(" RELEASED.");

            pixels.setPixelColor(keyLed, pixels.Color(0, 0, 0));
          }
        }
      }
    }
 

  // FastLED.show();
  pixels.show();

  HandleInactivitySleep();
}                               

// Initialize IO at start
  void InitializeIO() {
    for (int i = 0; i < ROWS; i++) {
      gpio_pad_unhold(rowPins[i]);
      gpio_reset_pin((gpio_num_t) rowPins[i]);
      pinMode(rowPins[i], INPUT_PULLUP);
    }

    gpio_deep_sleep_hold_dis();
    for (int i = 0; i < COLS; i++) {
      
      gpio_reset_pin((gpio_num_t) colPins[i]);
      pinMode(colPins[i], OUTPUT);
      digitalWrite(colPins[i], LOW);
      gpio_hold_dis((gpio_num_t) colPins[i]);
    }
  }

// Handle ESP Inactivity Sleep
  void HandleInactivitySleep() {
    unsigned long currentMillis = millis();
    if (currentMillis - inactivityMillis >= inactivityTime) {

      pixels.setPixelColor(4, pixels.Color(0, 128, 0));
      pixels.show();
      delay(3000);
      pixels.setPixelColor(4, pixels.Color(0, 0, 0));
      pixels.show();
      digitalWrite(LEDS_POWER_PIN, LOW);

      for (int i = 0; i < ROWS; i++) {
        esp_deep_sleep_enable_gpio_wakeup(1 << rowPins[i], ESP_GPIO_WAKEUP_GPIO_HIGH);
      } 

      gpio_deep_sleep_hold_en();
      for (int i = 0; i < COLS; i++) {
        pinMode(colPins[i], OUTPUT);
        digitalWrite(colPins[i], HIGH);
        gpio_hold_en((gpio_num_t) colPins[i]);
      }

      DEBUG_SERIAL.println("ESP - Going to sleep in 1 sec");
      esp_deep_sleep_start();
      DEBUG_SERIAL.println("ESP - This will never be printed");
    }
  }

// Get Key index
  int getKeyIndex(char value) {
    int idx = -1;

    for (int r=0; r<ROWS; r++) {
      for (int c=0; c<COLS; c++) {
        if (value == keys[r][c]) {
          idx = r*ROWS + c;
          break;
        }
      }
    }
    return idx;
  }
```

### Debug Message

```plain
I don't have error in Serial as it happens only if Serial is not connected to Serial Monitor on PC
```

### Other Steps to Reproduce

If i disable Serial in code everything works correctly.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-02:
Similar to https://github.com/espressif/arduino-esp32/issues/6983

SuGlider, 2023-03-02:
This issue was fixed by https://github.com/espressif/arduino-esp32/pull/7583
I just tested it and it works correctly with and without USB CDC plugged.

I used the sketch example from the #6983 with Arduino Core 2.0.7

SuGlider, 2023-03-02:
I also tested the sketch from #6983 using the LILO C3 Mini board to make sure that it has the right configuration.
It worked correctly again with the USB plugged and unplugged.

SuGlider, 2023-03-02:
@setrin - Please make sure that the Arduino Core you are using is the 2.0.7. 
Please try the sketch from #6983 - Change the LED to GPIO7 (Lolin C3 Mini) and see if it slows down or it keeps cycling every 2.5 seconds (expected time - versus 4 seconds before fixing it).

VojtechBartoska, 2023-03-03:
thanks for testing this @SuGlider, I'm adding it to 2.0.8 milestone to not forget to close it after the test of author of this issue.

setrin, 2023-03-04:
I have tested mentioned workaround and it works with my sketch perfectly. Thank you :)