espressif/arduino-esp32 #7886: It cannot stop advertisement after that advert.start(4, 0);.

talashganvir, 2023-02-24:
### Related area

bluetooth start adv

### Hardware specification

ESP32C3 DEV MODULE

### Is your feature request related to a problem?

It cannot stop advertisement after that advert.start(4, 0);.

### Describe the solution you'd like

I used stop function but it reset the whole code

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-02-24:
@talashganvir 

There is  `start()` and `stop()`in Advertising API:

https://github.com/espressif/arduino-esp32/blob/7856de7a57420e494176c16c5138174fe2c1dad0/libraries/BLE/src/BLEAdvertising.h#L47-L53

SuGlider, 2023-03-02:
The same for `BLEMultiAdvertising` ... There is `stop(...)`

https://github.com/espressif/arduino-esp32/blob/7856de7a57420e494176c16c5138174fe2c1dad0/libraries/BLE/src/BLEAdvertising.h#L99-L102

en129, 2023-04-07:
I'm also facing the same issue.
I understand there is a stop function. However, I don't know how to use the second argument ext_adv_inst.

talashganvir, 2023-08-18:
> is this still valid @talashganvir <https://github.com/talashganvir>

No

closed 2023-08-21 by VojtechBartoska

DavideLentini, 2023-11-16:
@en129 
@talashganvir 
@SuGlider 
Hi all ! Some problem here . 
How do I use this function for stop the advertising? 
Which kind of data is ext_adv_inst? 
Thanks all