espressif/arduino-esp32 #7887: a chance for having  fully esp8266-compatible WiFi and Web Server/Client libs  to easily upgrade esp8266-code to esp32?

dsyleixa, 2023-02-24:
### Related area

wifi, webserver/client

### Hardware specification

ESP32

### Is your feature request related to a problem?

hi,
being just a hoby programmer I have written a program package for a Smart Home application that is quite extensive and complex for my skills. My esp8266 server builds a website feat. controls and chart tables, connects to a couple of local esp8266 remote clients, and shows local and remotes value both on the local display and on the website.
Unfortunately all my attempts failed trying to upgrade the esp8266 server code to my newer and  more powerful esp32 (which I needed to attach a webcam and a live video canvas on the website): esp8266 wifi- and web-libs are simple incompatible.
So my question and request is:
is there a chance for once having  fully esp8266-compatible WiFi and Web Server/Client libs  to easily upgrade esp8266-code to esp32?

just to show what it's about, this is the current state of development of my esp8288 smart home project:

### Describe the solution you'd like

eg,
when currently using the libs
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

I would like to have fully compatible ESP32 libs which are feat. the same keywords, syntax, and functionality as for the esp8266:
#include <ESP32WiFi.h>
#include <ESP32WebServer.h>

i.e., just exchange either lib, load choose ESP32 instead of esp8266 board core, press ctrl+u:
and it runs!

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2023-02-24:
Consider switching from Webserver to https://github.com/me-no-dev/ESPAsyncWebServer.  It supports both platforms.

dsyleixa, 2023-02-24:
@mrengineer7777

dsyleixa, 2023-02-24:
no, sorry, unfortunately that is no option. The point is:
I know only the code syntax I am currently using, it took me decades to make it run, and I have no idea how to port it to any other API lib. 
So I need ESP32 libs which are 100% compliant to 
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

dsyleixa, 2023-02-24:
no, sorry, unfortunately that is no option. The point is:
I know only the code syntax I am currently using, it took me almost a decade to make it run, I admittedly c+p'ed lot of code from different examples found in the web without having understood all and everything, and I have no idea how to port it to any other API lib. 
So I need ESP32 libs which are 100% compliant to 
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

dsyleixa, 2023-02-24:
no, sorry, unfortunately that is no option. The point is:
I know only the code syntax I am currently using, it took me almost a decade to make it run, I c+p'ed lot of code from different examples found in the web (admittedly without having understood really all and everything), and I have no idea how to port it to any other API lib.
So I need ESP32 libs which are 100% compliant to
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

mrengineer7777, 2023-02-24:
I compared the Webservers between ESP32 and ESP8266. There are massive structural differences between them.  Your request is unreasonable. Sorry I can't help.
[image]

dsyleixa, 2023-02-24:
I think it's just a matter of function wrappers.
The API function names and passed arguments must be the same for both, but the ESP32 beyond and beneath has to be genuine ESP32 code of course.

dsyleixa, 2023-02-25:
PS, just to be precise:
the main problem is the port to integrate the web server code to query and control remote ESP8266,
whilst replacing ESP8266Wifi.h with Wifi.h tends to be the lesser problem (as it currently turns out).

everslick, 2023-09-24:
[WebServer.zip](https://github.com/espressif/arduino-esp32/files/12708341/WebServer.zip)

This is my fork of the esp32 webserver from before esp32 and esp8266 began to divert significantly. It compiles and works with both platforms for me. Maybe it helps you.

dsyleixa, 2023-09-24:
Thank you very much!
What I actually have is the esp8266-WiFi-server version which one can find in millions of ion examples in the web where I also have c+p'ed everything together, , and now I want to port it to esp32.
Is it that what your [WebServer.zip](https://github.com/espressif/arduino-esp32/files/12708341/WebServer.zip) version is providing?
Here my esp8288 Wifi- and webserver code again:
https://github.com/dsyleixa/Arduino/blob/master/ESP8266-IoT/ESP8266_ServerHomeWebMail_089a/ESP8266_ServerHomeWebMail_089a.ino

(I have installed esp8266 community version 2.7.4 (nodeMCU 1.0)

dsyleixa, 2023-09-24:
Thank you very much!
What I actually have is the esp8266-WiFi-/Webserver version which one can find in many ino  examples in the web where I also have c+p'ed everything together, , and now I want to port it to esp32.
Is it that what your [WebServer.zip](https://github.com/espressif/arduino-esp32/files/12708341/WebServer.zip) version is providing?
Here my esp8288 Wifi- and webserver code again:
https://github.com/dsyleixa/Arduino/blob/master/ESP8266-IoT/ESP8266_ServerHomeWebMail_089a/ESP8266_ServerHomeWebMail_089a.ino

(I have installed esp8266 community version 2.7.4 (nodeMCU 1.0)

everslick, 2023-09-27:
> Is it that what your [WebServer.zip](https://github.com/espressif/arduino-esp32/files/12708341/WebServer.zip) version is providing?

yes, it is the same code (basically) but has some minor fixes (I kept it in sync with fixes to either original repository).

Just add the files to your code and 

#include <ESPWebServer.h>

instead to your usual include. It **should* work.

dsyleixa, 2023-09-30:
unfortunately not :(
I copied your folder into my library folder (strangely it is not named ESPWebServer but just WebServer)
the outcommented
//#include  ESP8266WebServer webserver(8081);  
instead
 #include <ESPWebServer.h>
and then exchanged
// ESP8266WebServer webserver(8081);    //for  #include <ESP8266WebServer.h>
by
ESPWebServer webserver(8081);        // for #include <ESPWebServer.h>
for instantiation.

error:
exit status 1
'ESPWebServer' does not name a type

What am I missing?

everslick, 2023-10-02:
The class name declared in ESPWebServer.h is simply `WebServer` not `ESPWebServer` so you have to write `WebServer webserver(8081);`.

I hope this is the only thing I forgot to mention. Sorry for that. ;-)

dsyleixa, 2023-10-02:
hmmm... unfortunately still this error:
```
exit status 1
'webServer' does not name a type
```

dsyleixa, 2023-10-02:
thank you, now it compiles, but it neither builds the Wifi Server html website nor does it connect to my esp8266WebServer  clients.
With the original esp8266 version all is fine though.

dsyleixa, 2023-10-02:
thank you, now it compiles, but it neither builds the html website nor does it connect to my additional remote esp8266 clients.
With the original esp8266 version all is fine though.

everslick, 2023-10-04:
Hard for me to tell what exactly is the issue here. Two things I can think of:

Do you call `webserver.begin(8081);` in setup() and `webserver.handleClient();` in loop()?

If not, try it.

dsyleixa, 2023-10-04:
thank you very much!
Indeed I only had webserver.begin() just like for esp8288, only the instantiation had the 8081.
But now having added it it's receiving all the remote client values!
That's a big step actually.

The html website still does not build, but that's actually a wifi server(client issue.
Anayway, ping does work correctly though:
```

Ping wird ausgeführt für 192.168.2.200 mit 32 Bytes Daten:
Antwort von 192.168.2.200: Bytes=32 Zeit=17ms TTL=255
Antwort von 192.168.2.200: Bytes=32 Zeit=28ms TTL=255
Antwort von 192.168.2.200: Bytes=32 Zeit=45ms TTL=255
Antwort von 192.168.2.200: Bytes=32 Zeit=2ms TTL=255

Ping-Statistik für 192.168.2.200:
    Pakete: Gesendet = 4, Empfangen = 4, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 2ms, Maximum = 45ms, Mittelwert = 23ms
```

Absolutely unnecessary, these 8266/32 libs incompatibilities, tbh.

Now this is the current ported version:
https://github.com/dsyleixa/Arduino/tree/master/ESP32_IoT/ESP32_ServerHomeWebMail_089a

Thanks a lot for your efforts so far!