espressif/arduino-esp32 #7891: esp32-Cam example: adding code in loop() possible?

dsyleixa, 2023-02-25:
### Related area

esp32-cam

### Hardware specification

esp32-cam

### Is your feature request related to a problem?

esp32-Cam: 
i it possible  to add extra code in the loop() function?
I'm not sure because in Arduino IDE example it says:
```
void loop() {
  // put your main code here, to run repeatedly:
  delay(10000);
}
```
whilst in the repo example it is written:
```
void loop() {
  // Do nothing. Everything is done in another task by the web server
  delay(10000);
}
```

s

### Describe the solution you'd like

e.g. I would like additional code for reading button states or sensor values in between if possible

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-02-25:
@dsyleixa - Yes, you can add code to the `loop()` do execute something else that may be needed.
For example, it is possible to blink an LED or any other thing.

The Cam example doesn't need to run anything in the loop. That is the reason for this commentary.
It is a Web Server Application that runs on Network callbacks, thus nothing **else** is necessary in order to run it.

But there is an `Arduino Task` always running that executes the code in `loop()`, which just "delays 10s" and nothing else.

dsyleixa, 2023-02-25:
ok, that`s fine, thank you!
Then just 2 questions left about this esp32-cam example:
1) can I add std::thread threads to do things simultaneously by low priority, not to block anything of the cam funtions?
2) Is it possible to expand the built-in website functionality to add some widgets (e.g., buttons, tables)?
Or must/ can  I build a 2nd website form by the wifi server/client API functions inside loop() using common html code for my widgets?

SuGlider, 2023-02-25:
1. Yes, just include `#include <thread>` in your sketch.  `std::thread` is usable (and actually built on `xTaskCreate`)
You can also use FreeRTOS calls directly. I'm not sure that it is possible to set priority and stack size using `std::thread`, but you can certainly achieve it using FreeRTOSCall `xTaskCreate()`.
Arduino Core 2.0.x is based on IDF 4.4 and it has `std:thread` implemented over FreeRTOS. Please see the example (applicable to Arduino sketches as well) documentation for more details: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/pthread.html
Example: https://github.com/espressif/esp-idf/tree/release/v4.4/examples/cxx/pthread
2. Yes, but I'm not sure how easy that would be. HTML code is in https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/camera_index.h
It is GZ compressed and in this case, it would be necessary to build back the gz binary file, uncompress it, modify whatever necessary, compress it back again, convert it into C `uint8_t` array and then finally modify the `camera_index.h` file.
I think that it would need to build some tools. Beside that, it would require that the new URIs are treated by the code fom https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp 
Therefore, it may be a lot of work... I hope you can make it work for you.

dsyleixa, 2023-02-25:
as to std::.thread I'm already fine with it.
but as to the website-addons....:
hmmmh..., that sounds if it was lightyears from what I could do by my skills...
I thought it might be possible to just add some html code for some widgets to place them beneath the existing video canvas without recompiling and recompressing all and everything...

SuGlider, 2023-02-26:
> would it be easier to let the program just create a 2nd website form feat. all the widgets, to be exectuted in loop() as usual? and BTW, in void loop(): is the delay(10000); essential?

Yes, it is possible. The Cam Example alredy creates a Web Server for the Camera control. 
In the code of  `app_httpd.cpp` this Web Server is executed. You may need to add a new `Routing Request` to the code in
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp#L819-L933

There many Web Server examples in https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples
I think that it may be easier to try to build an application strucuture like https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples/FSBrowser

You can try to get the HTML/CSS/JS of the Camera Example directly from the Web Browser, instead of trying to decompress the content of  https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/camera_index.h

Anyway, this is some work...

dsyleixa, 2023-02-26:
thank you, but I have no clue what a roting request - probably I have to wait for an example code how to do that, e.g. by adding a html button and text String of a sensor value...

dsyleixa, 2023-02-26:
thank you, but I have not even a clue what a routing request is - probably I have to wait for an example code how to do that, e.g. by adding a html button and text String of a sensor value...

dsyleixa, 2023-02-26:
thank you, I actually know how to write html code for a button and a text String for a proprietary website - but I have nor clue what a routing request is ... 
probably I have to wait for an example code how to do that for the video website, e.g. by adding a html button and text String of a sensor value...

dsyleixa, 2023-02-26:
thank you, I actually know how to write html code for a button and a text String for a proprietary website (and I am doing that by wifiserver, not webserver) - but I have no even clue what a routing request is ... 
probably I have to wait for an example code how to do that for the video website, e.g. by adding a html button and text String of a sensor value...

SuGlider, 2023-02-28:
This is actually a change to the example code.
We use https://github.com/espressif/arduino-esp32/labels/Type%3A%20Feature%20request when it has to do with an Arduino Core new functionality.
What you are asking for is actually an Application  code change for an existing example.

Routing Request is when the Web Server receives a Request, like POST, GET, etc, and process it depending on the Requesting Parameters, by executing the related code for it.

There is something similar in many Web Frameworks, like in Node.JS, for instance:
https://expressjs.com/en/guide/routing.html

dsyleixa, 2023-02-28:
ok, I don't understand how to connect my proprietary code for buttons etc. with the vidoe thing.
So just to show what I want to do is sth like this function, to be called from void loop(), to show up it's widgets beneath the video canvas, and for that purpose I need an programming interface to the existing built-in website:

```
void handleWebsite() {
   char timestr[80];
   getLocalTime(timestr);

   String script="";
   String website_title = "myESP32WifiServer";

   WiFiClient client = wifiserver.available();   // listen for incoming client
   if (client) {                               // if you get a client,
      Serial.println("New Client.");           // print a message out the serial port
      String currentLine = "";                 // make a String to hold incoming data from the client
      while (client.connected()) {             // loop while the client's connected
         if (client.available()) {             // if there's bytes to read from the client,
            char c = client.read();            // read a byte, then
            Serial.write(c);                   // print it out the serial monitor
            if (c == '\n') {                   // if the byte is a newline character

               if (currentLine.length() == 0) {
                  script += "HTTP/1.1 200 OK \n";
                  script += "Content-type:text/html \n";
                  script += "\n";

                  script += "<head> \n";
                  script += "<meta http-equiv=\"refresh\" content=\"10\" > \n" ;

                  // utf-8 für "°" Zeichen
                  script += "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
                  script += "<title>";
                  script += website_title;
                  script += "</title> ";
                  script += "</head> \n";

                  script += "<h1>My First WiFiServer with ESP32 - Station Mode &#128522;</h1> \n";

                  script += "<h2>";
                  script += (String)timestr +"<br>\n";
                  script += "</h2> \n";

                  //---------------------------------------
                  // text font Courier, color black
                  script += "<p> <font face=\"courier\">"; // <<<<<<<<<<<<<<<< Courier
                  script += "<font style=\"color:rgb(0,0,0);\" > </p>";
                  script += "<br> \n";

                  script += "<h3> ";

                  script += "Output 1 ist: ";
                  if (OUT1 == 1)
                  {
                     script += ("EIN &nbsp; <wbr> <wbr> <br>\n");
                  }
                  else
                  {
                     script += ("AUS &nbsp; <wbr> <wbr> <br>\n");
                  }

                  script += "<a href=\" /OUT1H\"\"> <button style=\"height:70px;width:140px\" > EIN </button></a>  ";
                  script += "<a href=\" /OUT1L\"\"> <button style=\"height:70px;width:140px\" > AUS </button></a>  ";
                  script += "<br> \n\n";

                  script += "</h3> ";
                  script += "\n";
                  script += "\n";

                  client.print(script);
                  delay(1);

                  // break out of the while loop:
                  break;

               }
               else {    
                  currentLine = "";
               }
            }
            else if (c != '\r') {  // if you got anything else but a carriage return character,
               currentLine += c;      // add it to the end of the currentLine
            }
            delay(1);

            if (currentLine.endsWith("GET /OUT1H")) {
               OUT1=HIGH;              // GET /OUT1H turns the LED on
            }
            if (currentLine.endsWith("GET /OUT1L")) {
               OUT1=LOW;               // GET /OUT1L turns the LED off
            }
         }
      }
      // close the connection:
      client.stop();
      delay(1);
      Serial.println("Client Disconnected.");
   }
   delay(1);
}

```

VojtechBartoska, 2023-08-18:
Closing this ticket as expired, if needed you can reopen. Thanks

closed 2023-08-18 by VojtechBartoska