espressif/arduino-esp32 #7893: Lots of errors with RainMaker

FernandoGarcia, 2023-02-26:
### Board

ESP32-Wrover-IE

### Device Description

[image]

### Hardware Configuration

No.

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Ubuntu 18.04

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi!

I have installed the example RMakerSwitch after select RainMaker partition scheme and erase all flash.

But I can't make the code work, I'm always getting the error below.

How I can solve this problem?

Best regards.

### Sketch

```cpp
//This example demonstrates the ESP RainMaker with a standard Switch device.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

#define DEFAULT_POWER_MODE true
const char *service_name = "PROV_1234";
const char *pop = "abcd1234";

//GPIO for push button
#if CONFIG_IDF_TARGET_ESP32C3
static int gpio_0 = 9;
static int gpio_switch = 7;
#else
//GPIO for virtual device
static int gpio_0 = 0;
static int gpio_switch = 16;
#endif

/* Variable for reading pin status*/
bool switch_state = true;

//The framework provides some standard device types like switch, lightbulb, fan, temperaturesensor.
static Switch my_switch;

void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
        case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32S2
            Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
            printQR(service_name, pop, "softap");
#else
            Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
            printQR(service_name, pop, "ble");
#endif
            break;
        case ARDUINO_EVENT_PROV_INIT:
            wifi_prov_mgr_disable_auto_stop(10000);
            break;
        case ARDUINO_EVENT_PROV_CRED_SUCCESS:
            wifi_prov_mgr_stop_provisioning();
            break;
        default:;
    }
}

void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx)
{
    const char *device_name = device->getDeviceName();
    const char *param_name = param->getParamName();

    if(strcmp(param_name, "Power") == 0) {
        Serial.printf("Received value = %s for %s - %s\n", val.val.b? "true" : "false", device_name, param_name);
        switch_state = val.val.b;
        (switch_state == false) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
        param->updateAndReport(val);
    }
}

void setup()
{
    Serial.begin(115200);
    pinMode(gpio_0, INPUT);
    pinMode(gpio_switch, OUTPUT);
    digitalWrite(gpio_switch, DEFAULT_POWER_MODE);

    Node my_node;
    my_node = RMaker.initNode("ESP RainMaker Node");

    //Initialize switch device
    my_switch = Switch("Switch", &gpio_switch);

    //Standard switch device
    my_switch.addCb(write_callback);

    //Add switch device to the node
    my_node.addDevice(my_switch);

    //This is optional
 //   RMaker.enableOTA(OTA_USING_TOPICS);
    //If you want to enable scheduling, set time zone for your region using setTimeZone().
    //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
    // RMaker.setTimeZone("Asia/Shanghai");
    // Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
    RMaker.enableTZService();

    RMaker.enableSchedule();

    RMaker.enableScenes();

    RMaker.start();

    WiFi.onEvent(sysProvEvent);
#if CONFIG_IDF_TARGET_ESP32S2
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
#endif
}

void loop()
{
    if(digitalRead(gpio_0) == LOW) { //Push button pressed

        // Key debounce handling
        delay(100);
        int startTime = millis();
        while(digitalRead(gpio_0) == LOW) delay(50);
        int endTime = millis();

        if ((endTime - startTime) > 10000) {
          // If key pressed for more than 10secs, reset all
          Serial.printf("Reset to factory.\n");
          RMakerFactoryReset(2);
        } else if ((endTime - startTime) > 3000) {
          Serial.printf("Reset Wi-Fi.\n");
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        } else {
          // Toggle device state
          switch_state = !switch_state;
          Serial.printf("Toggle State to %s.\n", switch_state ? "true" : "false");
          my_switch.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, switch_state);
          (switch_state == false) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
      }
    }
    delay(100);
}
```

### Debug Message

```plain
E (575) esp_rmaker_device: Failed to allocate me����y for Device Switch
[     1][E][RMakerDevice.h:100] standardSwitchDevice(): Switch device not created
E (19) esp_rmaker_fctry: NVS Flash init failed
E (19) esp_rmaker_core: Failed to initialise storage
[    24][E][RMaker.cpp:75] initNode(): Node init failed
E (21) esp_rmaker_device: Failed to allocate memory for Device Switch
[    33][E][RMakerDevice.h:100] standardSwitchDevice(): Switch device not created
E (35) esp_rmaker_device: Device handle cannot be NULL
[    45][E][RMakerDevice.cpp:48] addCb(): Callback register error
E (45) esp_rmaker_node: Node or Device/Service handle cannot be NULL.
[    57][E][RMakerNode.cpp:10] addDevice(): Device was not added to the Node
E (58) esp_rmaker_device: Failed to allocate memory for Service Time
E (65) esp_rmaker_time_service: Failed to create Time Service
[    75][E][RMaker.cpp:131] enableTZService(): Timezone service enable failed
E (77) esp_rmaker_schedule: Couldn't allocate schedule_priv_data
[    88][E][RMaker.cpp:122] enableSchedule(): Schedule enable failed
E (89) esp_rmaker_scenes: Couldn't allocate scenes_priv_data
[    99][E][RMaker.cpp:154] enableScenes(): Scenes enable failed
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-28:
@sanketwadekar / @me-no-dev 

Issue Confirmed, It works fine with 2.0.6 and fails only with 2.0.7

me-no-dev, 2023-02-28:
@sanketwadekar will your PRs fix this issue?

sanketwadekar, 2023-02-28:
These memory allocation issues are seen because the latest esp rainmaker c library is using `heap_caps_calloc()`, `heap_caps_malloc()` APIs for memory allocation. These APIs work fine in Rainmaker IDF examples. I'm not sure why it is breaking in Arduino.

FernandoGarcia, 2023-02-28:
Hi!

Thanks for the attention.

Why the RainMaker partition scheme is available only under ESP32 Dev Module? I can't see it in ESP32 Wrover Module or Wrover Kit.

My board has 8 MB of FLASH it the RainMaker partition scheme suitable for this size?

Best regards.

sanketwadekar, 2023-03-03:
@FernandoGarcia 
Thanks for reporting this issue.
As I said earlier, the rainmaker c libraries are pre-built using config `CONFIG_SPIRAM=y` for esp32 target. Now, not all esp32 modules have PSRAM. So without even checking whether PS RAM is available, rainmaker is allocating memory on ext. ram. This causes memory allocation failures on chips which do not have PS RAM. Even if the module has PSRAM, and if it is not enabled in Arduino IDE "Tools" menu, the same errors will be seen. 

Moreover, if you refer to [this table](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html#main-i-o-connector-jp1), pin 16(in esp32 wrover kit) is used for accessing PSRAM but in the example `gpio_switch` is set to 16 which will cause memory allocation failures since PSRAM would become inaccessible.

Also, this memory allocation issue has exposed a few bugs in the Rainmaker Arduino libraries, which I will fix and will be available in the next release. Until then, you can use Arduino-ESP32 2.0.6 to build your Rainmaker projects. 🙂

FernandoGarcia, 2023-03-03:
Hi!

> Moreover, if you refer to [this table](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html#main-i-o-connector-jp1), pin 16(in esp32 wrover kit) [...]

This type of conflict should be checked  during the compilation for all features, no? Like a sanity check.

Regarding the partition scheme, here's missing the rainmaker option.

[image]

[image]

FernandoGarcia, 2023-03-03:
For sanity check feature I'm not talking only to this specific sketch.

To make it work to all boards you can use a bunch of `#ifdef` like [Marlin firmware](https://github.com/MarlinFirmware/Marlin/blob/579833e2a3d8ab8716fcab85dd26904a718fcec8/Marlin/src/inc/SanityCheck.h) does. :)

FernandoGarcia, 2023-03-03:
Well, I'm talking about something like this pseudocode:

```
int gpio_get_level(gpio_num_t gpio_num);
{
  if(gpio_num == 16)
  {
    return true;
  }
  return false;
}

#if defined(ESP_32) && defined(USE_PSRAM)
  if(gpio_get_level(16) == true)
    {
      #error "GPIO 16 can't be used when PSRAM is enabled"
    }
#endif
```

sanketwadekar, 2023-03-03:
> 

I'll consider this while fixing the bugs and updating the examples. Thanks for sharing!

FernandoGarcia, 2023-03-03:
There's some issues here with erros like ` E (355) gpio: gpio_set_level(226): GPIO output gpio_num error`. This message doesn't help so much beginners.

A  sanity check function with clear message could reduce it significantly.

sanketwadekar, 2023-03-04:
@FernandoGarcia 
I am not completely sure but I think sanity checks should be placed inside the gpio drivers, since the PSRAM gpio issue that we see in this example could happen to any other (non Rainmaker) example. Can you create a new issue addressing the "sanity checks" so that others can also mention their views on this? Let's keep this issue limited to Rainmaker related issues(as the title describes :)) and talk about the gpio sanity checks in a fresh issue.

VojtechBartoska, 2023-04-24:
Hello, this is fixed with 2.0.8 . If you still face some issues, feel free to reopen.

closed 2023-04-24 by VojtechBartoska