espressif/arduino-esp32 #7894: BLE Name Length Too Long Cause Crash

GodayMax, 2023-02-27:
### Board

ESP32 Dev Module

### Device Description

ESP32-WROOM-32

### Hardware Configuration

n/a

### Version

v1.0.6

### IDE Name

PlatformIO

### Operating System

MacOS 12

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Using BLEDevice to init ble device name with a name longer than 30 character
Module will crash when init 

BLEDevice::init(String("MZ_MZL-FC-E-XX-999999-C8F09EE7C0CC").c_str());

### Sketch

```cpp
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

uint8_t txValue = 0;
BLEServer *pServer = NULL;                   //BLEServer指针 pServer
BLECharacteristic *pTxCharacteristic = NULL; //BLECharacteristic指针 pTxCharacteristic
bool deviceBleConnected = false;             //本次连接状态
bool oldDeviceConnected = false;             //上次连接状态

// See the following for generating UUIDs: https://www.uuidgenerator.net/
#define SERVICE_UUID "83677baa-3eb8-4866-b6b6-96e5ed5cc48d" // UART service UUID
#define CHARACTERISTIC_UUID_RX "f5d2b3fe-e6b5-49b5-aa5f-a00bb4156d1d"
#define CHARACTERISTIC_UUID_TX "6f588463-f8f1-44f8-bdae-a1272a1b0f6e"

class MyServerCallbacks : public BLEServerCallbacks
{
    void onConnect(BLEServer *pServer) {};
    void onDisconnect(BLEServer *pServer) {}
};

//ble相关
class MyBLECallbacks : public BLECharacteristicCallbacks {
  void onWrite(BLECharacteristic *pCharacteristic) {}
};

void setup() {
  Serial.begin(115200);
  Serial.println("START >>>>>>>>>>>>>>>>>>>> ");

  BLEDevice::init(String("MZ_MZL-FC-E-XX-999999-C8F09EE7C0CC").c_str());

  // 创建一个 BLE 服务
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks()); //设置回调
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // 创建一个 BLE 特征
  pTxCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
  pTxCharacteristic->addDescriptor(new BLE2902());
  BLECharacteristic *pRxCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_RX, BLECharacteristic::PROPERTY_WRITE);
  pRxCharacteristic->setCallbacks(new MyBLECallbacks()); //设置回调

  BLEAdvertisementData d = BLEAdvertisementData();
  d.setManufacturerData("123");

  pService->start();                  // 开始服务
  pServer->getAdvertising()->setAdvertisementData(d);

  pServer->getAdvertising()->start(); // 开始广播
  Serial.println("ble inited, waiting for connection.....");
}

void loop() {
}
```

### Debug Message

```plain
START >>>>>>>>>>>>>>>>>>>> 
[I][BLEDevice.cpp:569] getAdvertising(): create advertising
ble inited, waiting for connection.....

Stack smashing protect failure!

abort() was called at PC 0x4016007b on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ecac:0x3ffd0910 0x4008ef29:0x3ffd0930 0x4016007b:0x3ffd0950 0x400fef8d:0x3ffd0970 0x40123c25:0x3ffd09d0 0x40124251:0x3ffd09f0 0x4012e75e:0x3ffd0a10 0x40103546:0x3ffd0a30 0x4008ff9e:0x3ffd0a60

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-27:
@GodayMax -
For BLE adverts, because the maximum advert length is 31 bytes, it is recommended to keep the name at 10 bytes or lower (so that you have space for other advert elements), however, you can theoretically use a maximum of 29 bytes for the short name (leaving a byte for the advert type and a byte for the advert length).

SuGlider, 2023-02-27:
@GodayMax - 
Running the sketch posted to this issue using the Arduino Core 2.0.6 and the ArduinoIDE with the menu option `Core Debug Level: "Error"`, I get an error message on the Serial Monitor output:

```
START >>>>>>>>>>>>>>>>>>>> 
[   253][E][BLEDevice.cpp:416] init(): esp_ble_gap_set_device_name: rc=258 Unknown ESP_ERR error
ble inited, waiting for connection.....
```

This also indicates that there was a problem setting a too long name for the BLE device.

SuGlider, 2023-02-27:
### Findings about the issue:

#### The device name must have the maximum length of 29 characters.

If the device name has between 30 and 32 characters long, an error of `Stack smashing protect failure! + ABORT & RESET` will occur

If the device name has a length of 33 or more, it won't reset the board, but the last valid device name will be used.
Also, an error message will be displayed in the Serial Console.
`[BLEDevice.cpp:416] init(): esp_ble_gap_set_device_name: rc=258 Unknown ESP_ERR error `

### Resolution:

A PR to check the length of the name in the `init()` function, which in case of wrong length, shall set the name as `bad length name max 29  bytes`. This may help in future cases like that.

SuGlider, 2023-02-28:
@GodayMax - PR #7901 shall prevent the issue of reseting the board.

GodayMax, 2023-02-28:
Hi, a little earlier I talked with some staff from espressif(Shanghai), the result shows like following images;
[image]
[image]

it is possible to name a device like  --->  MZ_MZL-FC-E-XX-9999-C8F09EE7C0CC without any problem.
So as I know what this PR doing is to make some pre-checks with the setting name's length, instead of supporting longer name,  it just prevents the module from crashing

So I wonder if you guys can make this func to behavior the same way as idf does ?

SuGlider, 2023-03-01:
@GodayMax - I see what you want to achieve. 
It is clear now. 
In this case the PR is not necessary.
I'll remove it.

### Just run this code based on your example - it does exactly what the image above shows.

``` cpp
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

uint8_t txValue = 0;
BLEServer *pServer = NULL;                   //BLEServer指针 pServer
BLECharacteristic *pTxCharacteristic = NULL; //BLECharacteristic指针 pTxCharacteristic
bool deviceBleConnected = false;             //本次连接状态
bool oldDeviceConnected = false;             //上次连接状态

// See the following for generating UUIDs: https://www.uuidgenerator.net/
#define SERVICE_UUID "83677baa-3eb8-4866-b6b6-96e5ed5cc48d" // UART service UUID
#define CHARACTERISTIC_UUID_RX "f5d2b3fe-e6b5-49b5-aa5f-a00bb4156d1d"
#define CHARACTERISTIC_UUID_TX "6f588463-f8f1-44f8-bdae-a1272a1b0f6e"

class MyServerCallbacks : public BLEServerCallbacks
{
    void onConnect(BLEServer *pServer) {};
    void onDisconnect(BLEServer *pServer) {}
};

//ble相关
class MyBLECallbacks : public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {}
};

void setup() {
  Serial.begin(115200);
  Serial.println("START >>>>>>>>>>>>>>>>>>>> ");

  // Set General Access Service (0x1800) with Device Name (0x2A00) 
  BLEDevice::init("MZ_MZL-FC-E-XX-9999-C8F09EE7C0CC");

  // 创建一个 BLE 服务
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks()); //设置回调
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // 创建一个 BLE 特征
  pTxCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
  pTxCharacteristic->addDescriptor(new BLE2902());
  BLECharacteristic *pRxCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_RX, BLECharacteristic::PROPERTY_WRITE);
  pRxCharacteristic->setCallbacks(new MyBLECallbacks()); //设置回调

  pService->start();                  // 开始服务
  
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();

  // sets BLE advertsing information
  BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
  BLEAdvertisementData oScanResponseData = BLEAdvertisementData();

  oScanResponseData.setFlags(0x06); // GENERAL_DISC_MODE 0x02 | BR_EDR_NOT_SUPPORTED 0x04
  oAdvertisementData.setName("ESP_GATTS_DEMO");
  oAdvertisementData.setManufacturerData("123");
  pAdvertising->setAdvertisementData(oAdvertisementData);
  pAdvertising->setScanResponseData(oScanResponseData);

  pAdvertising->start();

  Serial.println("ble inited, waiting for connection.....");
}

void loop() {
}

```

SuGlider, 2023-03-01:
@GodayMax - Let me know if we can close this issue.

SuGlider, 2023-03-02:
Solved.

closed 2023-03-02 by SuGlider