espressif/arduino-esp32 #7896: BT stack doesn't accept connections once it is disabled and re-enabled

garageeks, 2023-02-27:
### Board

ESP32 Dev Module

### Device Description

ESP32-Wroom module

### Hardware Configuration

default

### Version

v2.0.6

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

default

### Description

I'm looking for a method to disable the BT stack, free up some memory to send data via Wi-Fi reliably, then re-enable the BT stack. I'm using this A2DP library to open a I2S sink. `https://github.com/pschatzmann/ESP32-A2DP`
The library initializes the BT stack by itself anytime a sink is started, but leaves it on once the sink is stopped.

Reading the documentation, I understand the BT stack is unitialized with these functions:
```
esp_bluedroid_disable();
esp_bluedroid_deinit();
esp_bt_controller_disable();
esp_bt_controller_deinit();
```

When I run these functions, no BT connections are accepted anymore, as expected.
But when I re-enable the sink, and therefore the BT stack, connections are still not accepted.
Maybe there is a processing time to be observed between these function calls?
I also tried to disable just the Bluedroid system and leave the BT controller untouched, but still no connections.

I understood that in the past once the BT stack is uninitialized, it cannot be used anymore without a reboot.
Is this still the case?

Below I provide a sketch that every 30 seconds shuts down the BT sink and stack, and after 30 seconds re-enable it.

### Sketch

```cpp
/**
 * @file basic-a2dp-audioi2s.ino
 * @brief A2DP Sink with output to I2SStream. This example is of small value
 * since my Bluetooth Library already provides I2S output out of the box.
 *
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

long myTimer;

bool isEnabled = false;

extern void startSpeaker();
extern void endSpeaker();

void setup() {
  Serial.begin(115200);
  Serial.println("hello!");

  startSpeaker();

}

void loop() { 

   if(millis() - myTimer > 30000) {
      myTimer = millis();
      if(isEnabled) {
        endSpeaker();
      } else {
        startSpeaker();
      }
   }

}

void endSpeaker() {
  a2dp_sink.end(false);
  esp_bluedroid_disable();
  esp_bluedroid_deinit();
  esp_bt_controller_disable();
  esp_bt_controller_deinit();
  isEnabled = false;
}

void startSpeaker() {
   i2s_pin_config_t my_pin_config = {
				.bck_io_num = 4,
				.ws_io_num = 23,
				.data_out_num = 12,
				.data_in_num = I2S_PIN_NO_CHANGE
		};
    a2dp_sink.set_pin_config(my_pin_config);
    		a2dp_sink.set_on_connection_state_changed(connection_state_changed);
  		a2dp_sink.set_on_audio_state_changed(audio_state_changed);
  // Start Bluetooth Audio Receiver
  a2dp_sink.set_auto_reconnect(false);
  a2dp_sink.start("test");
  isEnabled = true;
}
```

### Debug Message

```plain
no errors reported
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-02:
@garageeks - 
It seems to be an IDF question instead. This issue also is not related to the Arduino Core code.
Although it uses the Arduino Framework to build it, the sketch uses an external IDF based library.

Maybe someone from the community may help with this issue.

1technophile, 2023-05-04:
> free up some memory to send data via Wi-Fi reliably,

You could use https://github.com/h2zero/NimBLE-Arduino it requires less memory than BlueDroid

VojtechBartoska, 2023-10-04:
Closing this issue as expired, if needed you can reopen.

closed 2023-10-04 by VojtechBartoska