espressif/arduino-esp32 #7899: tinycbor header files missing

zorc1985, 2023-02-27:
### Board

ESP32 S3 Dev Module and others

### Device Description

this is about a software problem

### Hardware Configuration

this is about a software problem

### Version

v2.0.7

### IDE Name

arduino-cli 0.31.0

### Operating System

Ubuntu

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/cbor/port/include/cbor.h#L1

The tinycbor header files referenced here appear to be missing. Same for the equivalent `cbor.h` in other subdirectories of `sdk` like `esp32s3`.

### Sketch

```cpp
#include <cbor.h>
```

### Debug Message

```plain
this is a compiler error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-02-27:
Hi @zorc1985 nice find! I guess this is a special case that we need to handle in the future, because tinycbor source folder is not in the include list, but referenced by relative path.

VojtechBartoska, 2024-02-20:
@me-no-dev is this still relevant? If yes, will we postpone this to 3.1.0 milestone?

me-no-dev, 2024-02-20:
not relevant for 3.0.0

closed 2024-02-20 by me-no-dev