espressif/arduino-esp32 #7903: XIAO ESP32-C3 Rainmaker example crash

MingyaoLiu, 2023-02-28:
### Board

Seeed XIAO ESP32-C3

### Device Description

XIAO ESP32

### Hardware Configuration

Nothing connected.

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

windows 11

### Flash frequency

80 or 40 tried both

### PSRAM enabled

yes

### Upload speed

921600

### Description

Loaded a rainmaker example from arduino IDE on XIAO esp32c3, build successfully, crashes on XIAO. I tried 2 examples one is RMakerCustom, one is RMakerSwitch, both crashes.

Tried wifi provision and some BLE examples, all fine.

### Sketch

```cpp
//This example demonstrates the ESP RainMaker with a standard Switch device.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

#define DEFAULT_POWER_MODE true
const char *service_name = "PROV_1234";
const char *pop = "abcd1234";

//GPIO for push button
static int gpio_0 = 9;
static int gpio_switch = 7;

/* Variable for reading pin status*/
bool switch_state = true;

//The framework provides some standard device types like switch, lightbulb, fan, temperaturesensor.
static Switch my_switch;

void sysProvEvent(arduino_event_t *sys_event) {
  switch (sys_event->event_id) {
    case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32S2
      Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
      printQR(service_name, pop, "softap");
[... 152 lines cut ...]
22:58:57.262 -> 3fca3be0: 0x00000000 0x3fca3be8 0xffffffff 0x3fca3be8 0x3fca3be8 0x00000001 0x00000001 0x00000000
22:58:57.262 -> 3fca3c00: 0x0000ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678 0x00000030 0xabba1234 0x00000024
22:58:57.262 -> 3fca3c20: 0x00002008 0x00000000 0x3fca3c2c 0xffffffff 0x3fca3c2c 0x3fca3c2c 0x00000000 0xb33fffff
22:58:57.262 -> 3fca3c40: 0x00000000 0xbaad5678 0x000000e0 0xabba1234 0x000000d4 0x3fca3ca8 0x3fca3cb4 0x3fca3d28
22:58:57.262 -> 3fca3c60: 0x3fca3cb0 0x00000000 0x3fca3c6c 0xffffffff 0x3fca3c6c 0x3fca3c6c 0x00000001 0x3fca3c80
22:58:57.262 -> 3fca3c80: 0xffffffff 0x3fca9794 0x3fca9794 0x00000000 0x00000020 0x00000004 0x0000ffff 0x00000000
22:58:57.262 -> 3fca3ca0: 0xb33fffff 0x00000000 0x3fcaff30 0x3fcb43d0 0x3fcb70e4 0x4005890e 0x4038b68e 0xffffffff
22:58:57.262 -> 3fca3cc0: 0x00000000 0x00000000 0x00000001 0x00000001 0x00000000 0x00000001 0x600c0000 0x3fc9b000
22:58:57.262 -> 3fca3ce0: 0x3fcdfaac 0x807fffff 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
22:58:57.262 -> 3fca3d00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00020000 0xffffffff 0xffffffff 0xffffffff
22:58:57.262 -> 3fca3d20: 0x00000000 0x00000000 0xbaad5678 0x00000024 0xabba1234 0x00000018 0x3c146b78 0x3fca7994
22:58:57.423 -> 
22:58:57.423 -> 
22:58:57.423 -> 
22:58:57.423 -> ELF file SHA256: 4faa426969ea8030
22:58:57.423 ->
```

### Other Steps to Reproduce

Any rainmaker example

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

MingyaoLiu, 2023-02-28:
the code i pasted has some line different from the example, but loading the example and compile without modification also has the same crashing issue. 
Also tried 'erase all flash before uploading', same result.

MingyaoLiu, 2023-02-28:
I tried 2.0.6, and the examples are loading and able to do provisioning.
I saw another thread that shows there are a bunch of error in rainmaker in 2.0.7: https://github.com/espressif/arduino-esp32/issues/7893
I guess they are the same issue.

sanketwadekar, 2023-02-28:
@SuGlider This crash is seen only when PSRAM is enabled. However, BLE memory allocation errors aren't resolved by disabling PSRAM. I've tried the WiFi Prov example and can confirm that the same errors are seen there too(errors are seen only if BT transport is used.)

VojtechBartoska, 2023-03-01:
Thanks for your comments @sanketwadekar. Any suggestions how to fix this?

Jason2866, 2023-03-07:
C3 has no PSRAM so this can't be the reason.

sanketwadekar, 2023-05-22:
> Hello, this is fixed with 2.0.8 . If you still face some issues, feel free to reopen.

The Bluetooth issue is not resolved yet. Please reopen this issue.

closed 2023-05-31 by me-no-dev