espressif/arduino-esp32 #7905: ESP-32-CAM module: Detected camera not supported.

dsyleixa, 2023-02-28:
### Related area

esp32-cam

### Hardware specification

esp32-cam

### Is your feature request related to a problem?

my ESP-32-CAM module is this one:
https://www.ebay.de/itm/275652261602?var=576979368452
`https://www.ebay.de/itm/275652261602?var=576979368452`
error msg in serial console after connecting:
load:0x40078000,len:10944 load:0x40080400,len:6388 entry 0x400806b4 [E][camera.c:1113] camera_probe(): Detected camera not supported. [E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004 

### Describe the solution you'd like

I am using the example of my Arduino IDE (board core 1.0.6).
ESP32-board: Wrover module,
Partition Scheme: Huge APP (3MB No OTA/1MB SPIFFS)

### Describe alternatives you've considered

Is there perhaps a newer lib with more supported cam typeswhich I can use instead?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-02-28:
You board is not supported in that old version. You need to either update the core, or add the proper pin config for ESP32-CAM from the updated example

closed 2023-02-28 by me-no-dev

dsyleixa, 2023-02-28:
thanks, but 2.0.7 doesn't work either.
After light flashes shortly after upload, serial output states:
```
Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0

E (486) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
Camera init failed with error 0x105
```

Looks as if it's not functional or uncompatible

dsyleixa, 2023-02-28:
thanks, but 2.0.7 doesn't work either.
After light flashes shortly after upload of the esp32-cam example code, serial output states:
```
Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0

E (486) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
Camera init failed with error 0x105
```

What does that error mean?

dsyleixa, 2023-03-01:
tbh, I have no idea which camera pins are correct and which ones have to be uncommented (where?)

dsyleixa, 2023-03-01:
Power is good.
But tbh, I have no idea which camera type and which pins are correct and what has e to be outcommented or uncommented

me-no-dev, 2023-03-01:
I think it's [this one](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino#L24) but I am not sure. You can search the issues here and look online for the proper pin assignment. Quite a few folks use that same module

dsyleixa, 2023-03-01:
thanks a lot - but hell, what a mess!
I don't already see through that confusing pile of letters...

dsyleixa, 2023-03-02:
thanks a lot!
Update, meanwhile the website shows up with controls left-hand (I was missing that I have to uncomment a board type additionally. Now I have this setup below, nonetheless the canvas is completely  black, nothing to see at all.
What the heck..... :(

```
// Select camera model
// ===================
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM

#define CAMERA_MODEL_AI_THINKER // Has PSRAM   // <<<<<<<<<<<<<<<   

//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
// ** Espressif Internal Boards **

#define CAMERA_MODEL_ESP32_CAM_BOARD   // <<<<<<<<<<<<<<< 

//#define CAMERA_MODEL_ESP32S2_CAM_BOARD
//#define CAMERA_MODEL_ESP32S3_CAM_LCD
```

me-no-dev, 2023-03-02:
you should have only one of them uncommented. Try with `CAMERA_MODEL_AI_THINKER ` uncommented only and comment out `CAMERA_MODEL_ESP32S2_CAM_BOARD `

dsyleixa, 2023-03-02:
ok, thank you, did it!
But same result:
website opens, cotrols visible, but completely black canvas

me-no-dev, 2023-03-02:
What happens when you click "Get Still" or "Start Stream"?

dsyleixa, 2023-03-02:
atfter some clicking here and there:
a video appears!
:-D
Thank you a lot! 8-)