espressif/arduino-esp32 #7907: Adafruit_TinyUSB_Arduino not working anymore in firmware 2.0.7

Robinyoh, 2023-02-28:
### Board

ESP32 S2 

### Device Description

LOLIN ESP32 S2 MINI and ESP32 S2 DEV board

### Hardware Configuration

nothing connected to test.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

OSX12.6.3

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Adafruit_TinyUSB_Arduino was working nicely and all of the sudden not anymore. after uploading example script no serial port anymore and no MIDI port.

After going back to ESP32 2.0.6 everything is working again.

The library is looking at esp32-hal-tinyusb.c so maybe there are made some changes?

### Sketch

```cpp
/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This sketch is enumerated as USB MIDI device. 
 * Following library is required
 * - MIDI Library by Forty Seven Effects
 *   https://github.com/FortySevenEffects/arduino_midi_library
 */

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

// Variable that holds the current position in the sequence.
uint32_t position = 0;

// Store example melody as an array of note values
byte note_sequence[] = {
  74,78,81,86,90,93,98,102,57,61,66,69,73,78,81,85,88,92,97,100,97,92,88,85,81,78,
  74,69,66,62,57,62,66,69,74,78,81,86,90,93,97,102,97,93,90,85,81,78,73,68,64,61,
  56,61,64,68,74,78,81,86,90,93,98,102
};

void setup()
{
#if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
  // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
  TinyUSB_Device_Init(0);
#endif

  pinMode(LED_BUILTIN, OUTPUT);
  
  //usb_midi.setStringDescriptor("TinyUSB MIDI");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);

  // Attach the handleNoteOn function to the MIDI Library. It will
  // be called whenever the Bluefruit receives MIDI Note On messages.
  MIDI.setHandleNoteOn(handleNoteOn);

  // Do the same for MIDI Note Off messages.
  MIDI.setHandleNoteOff(handleNoteOff);

  Serial.begin(115200);

  // wait until device mounted
  while( !TinyUSBDevice.mounted() ) delay(1);
}

void loop()
{
  static uint32_t start_ms = 0;
  if ( millis() - start_ms > 266 )
  {
    start_ms += 266;
    
    // Setup variables for the current and previous
    // positions in the note sequence.
    int previous = position - 1;
  
    // If we currently are at position 0, set the
    // previous position to the last note in the sequence.
    if (previous < 0) {
      previous = sizeof(note_sequence) - 1;
    }
  
    // Send Note On for current position at full velocity (127) on channel 1.
    MIDI.sendNoteOn(note_sequence[position], 127, 1);
  
    // Send Note Off for previous note.
    MIDI.sendNoteOff(note_sequence[previous], 0, 1);
  
    // Increment position
    position++;
  
    // If we are at the end of the sequence, start over.
    if (position >= sizeof(note_sequence)) {
      position = 0;
    }
  }

  // read any new MIDI messages
  MIDI.read();  
}

void handleNoteOn(byte channel, byte pitch, byte velocity)
{
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleNoteOff(byte channel, byte pitch, byte velocity)
{
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}
```

### Debug Message

```plain
{
	"name":"Arduino on ESP32-S2",
	"toolchainPrefix":"xtensa-esp32s2-elf",
	"svdFile":"esp32s2.svd",
	"request":"attach",
	"postAttachCommands":[
		"set remote hardware-watchpoint-limit 2",
		"monitor reset halt",
		"monitor gdb_sync",
		"thb setup",
		"c"
	],
	"overrideRestartCommands":[
		"monitor reset halt",
		"monitor gdb_sync",
		"thb setup",
		"c"
	]
}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-02-28:
@SuGlider

hathach, 2023-03-02:
@me-no-dev I will pull new update and check to see if there is any obvious issue.

hathach, 2023-03-02:
Update: I have tested with S2 devmodule and our Adafruit Feather esp32s2 it seems to work well with the stock example with both arduino-esp32 and tinyusb arduino at master. Make sure you have  
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/MIDI/midi_test .

Could you take the picture of your IDE menu setting, the CDC on boot must be enabled

[image]

me-no-dev, 2023-03-02:
Thanks @hathach for looking into this!

Robinyoh, 2023-03-02:
Can not test the esp32 S2 Dev module at the moment, but the main board I use is the Lolin S2 MINI (Deneyap Mini) for a DIY MIDI remote project.
[image]

Verbose output ESP 2.0.6 Serial and MIDI port after upload:
[Lolin S2 Mini MIDI_test 2.0.6 verbose output.zip](https://github.com/espressif/arduino-esp32/files/10871324/Lolin.S2.Mini.MIDI_test.2.0.6.verbose.output.zip)

Verbose output ESP 2.0.7 no Serial port and no MIDI port after upload:
[Lolin S2 Mini MIDI_test 2.0.7 verbose output.zip](https://github.com/espressif/arduino-esp32/files/10871328/Lolin.S2.Mini.MIDI_test.2.0.7.verbose.output.zip)

hoop this helps.

hathach, 2023-03-03:
I tested with master branch of this repo, which should include the 2.0.7 release ?

Gershy13, 2023-03-04:
Also having this issue with 2.0.7 and the ESP32-S2-DEVKIT-C-1. 
2.0.6 works fine for me. Also having an issue suddenly where my board is detected as "Deneyap Mini" on 2.0.6. 
Didn't have this issue a few days ago, it was showing up as unknown before, so i just manually set to esp32s2 dev board and it worked fine. Now it resets to Deneyap Mini every time i reconnect it.

tomrodeheffer, 2023-03-04:
I am also having this issue with 2.0.7 on an ESP32-S3 FeatherS3.  I just upgraded from 2.0.6, which worked fine.  I've got version 1.18.4 of the Adafruit TinyUSB Library.  Here is my entire program:
```
#include <Adafruit_TinyUSB.h>
void setup() { }
void loop() { }
```
Attaching a TTL-to-USB serial cable onto serial port 0, using core debug level "verbose" gives:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[    84][D][esp32-hal-tinyusb.c:675] tinyusb_enable_interface(): Interface CDC e
nabled
[    85][D][esp32-hal-tinyusb.c:564] tinyusb_load_enabled_interfaces(): Load Don
e: if_num: 2, descr_len: 75, if_mask: 0x10

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == S
PINLOCK_FREE)

Backtrace: 0x40377042:0x3fcea8a0 0x40379ac9:0x3fcea8c0 0x4037f96d:0x3fcea8e0 0x4
037c96d:0x3fceaa10 0x4037a8fc:0x3fceaa50 0x42003fbd:0x3fceaa90 0x420043ee:0x3fce
aab0 0x42001e30:0x3fceaad0 0x420044f1:0x3fceaaf0 0x420049fc:0x3fceab10 0x42005da
9:0x3fceab60

ELF file SHA256: 1a5dc63c2413f6df

Rebooting...
```
Needless to say, if I remove the #include <Adafruit_TinyUSB.h> line, things work just fine (although not very interestingly).

hathach, 2023-03-06:
just switch to branch https://github.com/espressif/arduino-esp32/tree/release/v2.x to test, and it indeed has issue with TinyUSB lib, ESP32 repeatedly reset with both S2 and S3. Apparently there is difference between master and branch 2.x. I will try to llook more closely to find the root cause

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400262ed
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x2958
entry 0x4004c18c
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4002b1be  PS      : 0x00060c33  A0      : 0x8002a5ca  A1      : 0x3ffcc5d0  
A2      : 0x3ffc4550  A3      : 0x3ffcc774  A4      : 0x00060020  A5      : 0x00000000  
A6      : 0x400a0e0c  A7      : 0xffffffff  A8      : 0x3ffc4558  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00000002  A12     : 0xfffffc18  A13     : 0x3ffc59f8  
A14     : 0x00060423  A15     : 0x3ffcc754  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0xfffffc18  LEND    : 0x3ffc59f8  LCOUNT  : 0x40027480  

Backtrace: 0x4002b1bb:0x3ffcc5d0 0x4002a5c7:0x3ffcc5f0 0x40029976:0x3ffcc610 0x40083e85:0x3ffcc650 0x400842ae:0x3ffcc670 0x40081d04:0x3ffcc690 0x400843b1:0x3ffcc6b0 0x400848a8:0x3ffcc6d0 0x40085145:0x3ffcc720
```

me-no-dev, 2023-03-06:
@hathach the main difference is that tinyusb has been updated to a recent version/commit: https://github.com/espressif/arduino-esp32/blob/release/v2.x/tools/sdk/versions.txt#L10

hathach, 2023-03-06:
@me-no-dev yeah It seems to be out of sync between code in the arduino lib and the sdk/libarduino-tinyusb. I just bump up tinyusb core in the lib https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/256, they are synced up and works perfect now. I am preparing for the new release.

hathach, 2023-03-06:
Update: TinyUSB version 2.0.0 should fix this issue https://github.com/adafruit/Adafruit_TinyUSB_Arduino/releases/tag/2.0.0

Robinyoh, 2023-03-06:
Thx @hathach the 2.0.0 version fixed it! Big Happy over here :-)

VojtechBartoska, 2023-03-09:
Closing as solved.

closed 2023-03-09 by VojtechBartoska

oxygen545, 2023-04-07:
I just updated to 2.07 and am having the issue where I can only receive midi commands. Same example as the original poster. WeMos mini S2 v1