espressif/arduino-esp32 #7908: nvs_flash_deinit() breaks WiFi

beta-tester, 2023-02-28:
### Board

ESP32 Dev Module

### Device Description

n/a

### Hardware Configuration

n/a

### Version

v2.0.7

### IDE Name

Arduino IDE 1.8.19

### Operating System

ubuntu 22.04 / debian Bullseye

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

when nvs flash is deinitialized `nvs_flash_deinit();`, WiFi gives error
```
[E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 4353
[E][WiFiAP.cpp:154] softAP(): enable AP first!
[E][issue_wifi.ino:18] setup(): Soft AP creation failed.
```
even WiFi store is set to RAM instead of flash
`WiFi.persistent(false);` or `esp_wifi_set_storage(WIFI_STORAGE_RAM);`

### Sketch

```cpp
#include <WiFi.h>
#include <esp_wifi.h> // esp_wifi_set_storage(WIFI_STORAGE_RAM)
#include "nvs_flash.h" // nvs_flash_init(), ...

const char * ssid = "***********";
const char * password = "***********";

void setup() {
  Serial.begin(115200);

#if true
  nvs_flash_init();
  nvs_flash_erase();
  nvs_flash_deinit();
#endif

#if true
  WiFi.persistent(false);
#else
  esp_wifi_set_storage(WIFI_STORAGE_RAM);
#endif
  
  if (!WiFi.softAP(ssid, password))
    log_e("Soft AP creation failed.");
}

void loop() {
  yield;
}
```

### Debug Message

```plain
without nvs_flash_deinit();
20:58:12.606 -> [   105][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
20:58:12.606 -> [   657][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
20:58:12.606 -> [   659][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
20:58:12.606 -> [   665][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 11 - AP_STOP
20:58:12.639 -> [   665][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
20:58:12.639 -> [   677][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START

with nvs_flash_deinit() and WiFi.persistent(false);
21:19:38.367 -> [   252][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 4353
21:19:38.367 -> [   252][E][WiFiAP.cpp:154] softAP(): enable AP first!
21:19:38.367 -> [   252][E][issue_wifi.ino:18] setup(): Soft AP creation failed.

with nvs_flash_deinit() and esp_wifi_set_storage(WIFI_STORAGE_RAM);
21:21:14.377 -> [   245][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 4353
21:21:14.377 -> [   245][E][WiFiAP.cpp:154] softAP(): enable AP first!
21:21:14.377 -> [   245][E][issue_wifi.ino:24] setup(): Soft AP creation failed.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

beta-tester, 2023-02-28:
i DON'T want to use the nvs / flash to store WiFi credentials or WiFi RF calibration data.

mrengineer7777, 2023-03-01:
@beta-tester That error is throw from the IDF function esp_wifi_init().  The Arduino layer has no control over that.

My suggestion:
Leave NVS on but set WiFi.persistent(false);  That's what I did in my last project.  I do use NVS to store custom settings via Preferences.h.

mrengineer7777, 2023-03-01:
This may also be interesting: https://stackoverflow.com/questions/65873683/how-do-you-erase-esp32-wifi-smartconfig-credentials-from-flash

SuGlider, 2023-03-02:
> This may also be interesting: https://stackoverflow.com/questions/65873683/how-do-you-erase-esp32-wifi-smartconfig-credentials-from-flash

Just to add another information:
When using the Arduino IDE, there is an option to erase the flash before uploading the sketch.
IDE Menu -> Tools -> Board -> Erase All Flash Before Sketch Upload -> "Enabled" (or "Disabled"  as default).

beta-tester, 2023-03-02:
hmm, maybe i misunderstood the relationship between nvs and RF calibration.

in the ESP32-IDF documentation i only found that it is possible to prevent storing WiFi credentials in nvs flash.

but i could not find that it is possible to prevent storing RF calibration parameters to nvs flash.
RF calibration parameters can only be erased, but as soon RF is started, after a full RF calibration its calibration parameters will be stored to nvs flash again.

so WiFi must fail, when i deinit the flash, because nvs flash is mandatory for RF calibration...

are my thoughts right?

SuGlider, 2023-03-02:
>are my thoughts right?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/RF_calibration.html

SuGlider, 2023-03-02:
NVS is mandatory for WiFi. WiFi will always use NVS.

closed 2023-03-03 by beta-tester