espressif/arduino-esp32 #7914: Proper LED_BUILTIN for CORE-ESP32 (ESP32C3) board

fabriziop, 2023-03-01:
### Related area

Onboard led blinking

### Hardware specification

Onboard led

### Is your feature request related to a problem?

Arduino IDE works fine with CORE-ESP32 board (choosing "ESP32C3 Dev Module" from board manager) except for the LED_BUILTIN value that is set to 30, while CORE-ESP32 has a LED on pin 12 and a secondary LED on pin 13.

### Describe the solution you'd like

Provide the board manager with a board choice specific for the CORE-ESP32 board, returning the correct value of LED_BUILTIN: 12.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-03-02:
@fabriziop - 
`ESP32C3 Dev Module` has a [configuration](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html) with a NeoLED in GPIO 8.

We made a change to the code and pin definition accorfing to these links: 
https://github.com/espressif/arduino-esp32/issues/6783
https://github.com/espressif/arduino-esp32/issues/6968
https://github.com/espressif/arduino-esp32/pull/6979

https://github.com/espressif/arduino-esp32/blob/05ae83a051608e4d6c8821419ca40d1b35491561/variants/esp32c3/pins_arduino.h#L11-L15

Therefore, `LED_BUILTIN` may change in order to work with boards that use NeoLED instead of plain LED.

Maybe the `CORE-ESP32` board may need a new Board definition and Variant setup.

SuGlider, 2023-03-02:
>Provide the board manager with a board choice specific for the CORE-ESP32 board, returning the correct value of LED_BUILTIN: 12.

We rely on the community and board makers to add such new boards.
Please feel free to submit a PR adding it.

VojtechBartoska, 2023-08-18:
Closing this as it was already covered in mentioned PR above and there are no updates from author of this issue. If needed, you can reopen this issue.