espressif/arduino-esp32 #7915: Add an overload of WiFi.begin to support specifying a WPA2 EAP-TTLS phase 2 method

semicolonTransistor, 2023-03-02:
### Related area

WiFi

### Hardware specification

All WiFi capbale SoCs

### Is your feature request related to a problem?

The `WiFi.begin()` overload for WPA2-Enterprise added in #6398 seems to default to either MSCHAPV2 or MSCHAP as the phase 2 method for EAP-TTLS. There is no parameter to change the TTLS phase 2 method. However, my university network requires using PAP as the phase 2 method.  Currently, I am working around this by calling `esp_wifi_sta_wpa2_ent_set_ttls_phase2_method(ESP_EAP_TTLS_PHASE2_PAP);` before calling `WiFi.begin(...)`

I believe similar problems have been reported in #6427 and #5483.

### Describe the solution you'd like

It would be nice if an overload of WiFi.begin() can be added that accepts a `esp_eap_ttls_phase2_types` as a parameter so the user can select which TTLS phase 2 method they would like to use.

Alternatively, `WiFi.begin()` could try all available phase 2 options for TTLS. Which could make it easier to use and more "plug and play". Although I am not sure if that is a good idea since I have very limited knowledge of WPA2-Enterprise.

### Describe alternatives you've considered

I have used the following workaround to connect to my school network

```c++
  esp_wifi_sta_wpa2_ent_set_ttls_phase2_method(ESP_EAP_TTLS_PHASE2_PAP);
  WiFi.begin(ssid, WPA2_AUTH_TTLS, EAP_IDENTITY, EAP_USERNAME, EAP_PASSWORD);
```

This works although it took considerable research to find. I think the WiFi library would be more user-friendly if the option is available in `WiFi.begin()` and didn't require a separate call to an ESP-IDF function.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-03-02:
@semicolonTransistor 
It sounds to be a nice feature. 
Let's consider it for the next release 2.0.8 or 3.0.0.
Thanks for the contribution.

VojtechBartoska, 2024-04-12:
Hello, can you please validate this against 3.0.0-RC1 version? Thanks

semicolonTransistor, 2024-05-09:
I'm sorry for not getting back to you sooner. I tried the 3.0.0-RC1 version and it didn't appear to change anything. A separate function call is still required to set the TTLS phase 2 method. The connection sequence is shown in the snippet below. Without the first function call the connection fails to connect.
```
esp_eap_client_set_ttls_phase2_method(ESP_EAP_TTLS_PHASE2_PAP);
WiFi.begin(ssid, WPA2_AUTH_TTLS, EAP_IDENTITY, EAP_USERNAME, EAP_PASSWORD);
```
If there is another way I should be doing this please let me know, I'll be happy to test it out.

me-no-dev, 2024-05-09:
@semicolonTransistor so that order of calls works for you?

How do you imagine that option be added? Like extra argument to `WiFi.begin(ssid, WPA2_AUTH_TTLS, EAP_IDENTITY, EAP_USERNAME, EAP_PASSWORD);` ?

semicolonTransistor, 2024-05-09:
Yes, the 2 calls in the previous reply is working for connection to the network that requires PAP.

An extra argument to `WiFi.begin(ssid, WPA2_AUTH_TTLS, EAP_IDENTITY, EAP_USERNAME, EAP_PASSWORD)` is probably a good way to do this.

me-no-dev, 2024-05-14:
@VojtechBartoska will try to add it for RC3/final

JAndrassy, 2024-05-14:
https://www.arduino.cc/reference/en/libraries/wifinina/wifi.beginenterprise/

me-no-dev, 2024-05-15:
@semicolonTransistor how about
```cpp
WiFi.begin(ssid, WPA2_AUTH_TTLS, EAP_IDENTITY, EAP_USERNAME, EAP_PASSWORD, NULL, NULL, NULL, ESP_EAP_TTLS_PHASE2_PAP);
```

JAndrassy, 2024-05-15:
Arduino API:

WiFi.beginEnterprise()

Description

Initializes the WiFiNINA library’s network settings for a common WPA2 Enterprise network with username and password authentication (PEAP/MSCHAPv2).

Syntax

WiFi.beginEnterprise(ssid, username, password);
WiFi.beginEnterprise(ssid, username, password, identity);
WiFi.beginEnterprise(ssid, username, password, identity, ca);

Parameters

    ssid: the SSID (Service Set Identifier) is the name of the WiFi network you want to connect to.
    username: username part of WPA2 Enterprise (RADIUS) credentials
    password: password part of WPA2 Enterprise (RADIUS) credentials
    identity: WPA2 enterprise identity (optional)
    ca: root certificate (string) to validate against (optional)

Returns

    WL_CONNECTED when connected to a network
    WL_IDLE_STATUS when not connected to a network, but powered on

me-no-dev, 2024-05-15:
it's not the same thing.

closed 2024-05-15 by me-no-dev