espressif/arduino-esp32 #7917: implementing arduino client interface

mayankbytebeam, 2023-03-02:
### Board

ESP32

### Device Description

Development Board

### Hardware Configuration

None

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

40 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

can we make the arduino-esp32 core based on arduino client interface so that the core functionality can work on any standard arduino client interface.

for eg,
 
In HTTPUpdate Library we can do something like this,

t_httpUpdate_return update(Client& client, const String& url, const String& currentVersion = "");

Basically, I was doing HTTP OTA via GSM and then I get to know that it works only on WiFiClient interface, that does make sense because ESP32 inside is a WiFi Soc but what if we can make it out to generic so that any network interface will work.

any thoughts?

### Sketch

```cpp
nothing here
```

### Debug Message

```plain
nothing here
```

### Other Steps to Reproduce

nothing here

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-03-02:
This would be doable, if `Client` has all necessary methods. It is a good proposal and worth looking into.

mayankbytebeam, 2023-03-02:
I’m glad you liked it.

Seems like you added this to Arduino ESP32 Core Project Roadmap, but I couldn't find it.

me-no-dev, 2023-03-02:
In the project roadmap you need to select `3.0.0 Release` and it would be at the bottom of the `ToDo` column

VojtechBartoska, 2024-02-20:
Postponing this to 3.1.0 Milestone.

closed 2025-01-16 by Parsaabasi