espressif/arduino-esp32 #7918: ESP32-CAM module: how to start the WiFi Server passing a Port (e.g., 8080 or 8008) for a static www url?

dsyleixa, 2023-03-02:
### Related area

esp32-cam

### Hardware specification

esp32-cam

### Is your feature request related to a problem?

ESP32-CAM module: 
how to start the WiFi Server passing a Port (e.g., 8080 or 8008) for my static TCP url?
For my IoT Smart Home website I am using
WiFiServer   server(8080);
and I can call it from anywhere by cellphones via
http://myurl.sytes.net:8008/
(the port is activated in the internet router)

But how to do that with this ESP32-CAM website?

### Describe the solution you'd like

passing a TCP port to the WiFi/Web server

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-03-02:
After [this line](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp#L1208) you can add `config.server_port = 8080;`

closed 2023-03-02 by me-no-dev

dsyleixa, 2023-03-02:
great! I'll try that asap! where is this file in the normal installed Arduino directory tree?

dsyleixa, 2023-03-02:
BTW, that would be actually a real https://github.com/espressif/arduino-esp32/labels/Type%3A%20Feature%20request to be able to pass a port to 
void startCameraServer(int port);
or add that line 
config.server_port = 8080;
in the proprietary ino program file arbitrarily.

dsyleixa, 2023-03-03:
BTW, that would be actually a real feature request to be able to pass a port to 
void startCameraServer(int port);
or being able to add that line 
config.server_port = 8080;
directly in the proprietary ino program file arbitrarily (probably before calling startCameraServer() )

dsyleixa, 2023-03-03:
Sorry, but  I can't find thet source file to add the line... 
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp#L1208

dsyleixa, 2023-03-04:
update,
I found that file in a zip file
D:\Arduino\portable\packages\esp32\hardware\esp32\2.0.7\libraries\ESP32\examples\Camera\CameraWebServer
I saved it, moved it, uncompressed it, added the line 
config.server_port = 8008; // (!)
saved it, compressed it a .zip, copied back, recompiled my source file, but the result is the same:
accessable only locally by 192.168.2.202, but neither by 192.168.2.202:8008 nor by http://myurl.sytes.net:8008/ nor by http://myurl.sytes.net
 - other than my WifiServer programs running on my esp8266.
 
So as it not solved yet, please re-open this issue!
instructions to solve that welcome and highly appreciated!

dsyleixa, 2023-03-04:
update,
I found that file in a zip file
D:\Arduino\portable\packages\esp32\hardware\esp32\2.0.7\libraries\ESP32\examples\Camera\CameraWebServer
I saved it, moved it, uncompressed it, added the line 
config.server_port = 8008; // (!)
saved it, compressed it a .zip, copied back, recompiled my source file, but the result is the same:
accessable only locally by 192.168.2.202, but neither by 192.168.2.202:8008 nor by http://myurl.sytes.net:8008/ nor by http://myurl.sytes.net
 - other than my WifiServer programs running on my esp8266.
 

So apart from the fact that this patching is way to cumbersome to be proceeded each and every time there and back for either esp32-program:
As it not solved yet, please re-open this issue!
instructions to solve that welcome and highly appreciated!

dsyleixa, 2023-03-04:
update,
I found that file in a zip file
D:\Arduino\portable\staging\packages\esp32-2.0.7.zip\esp32-2.0.7\libraries\ESP32\examples\Camera\CameraWebServer\app_httpd.cpp
I saved it, moved it, uncompressed it, added the line 
config.server_port = 8008; // (!)
saved it, compressed it as .zip, copied back, recompiled my source file, but the result is the same:
accessable only locally by 192.168.2.202, but neither by 192.168.2.202:8008 nor by http://mysite.sytes.net:8008/ nor by http://mysite.sytes.net
 - other than my WifiServer programs running on my esp8266 and esp32.
 

So apart from the fact that this patching is way too cumbersome to be proceeded each and every time there and back for either esp32-program:
As this issue is not solved yet, please re-open this issue!
instructions to solve that are welcome and highly appreciated!

PS,
just to show that also another of my ESP32-WifiServer ino programs works fine via  port 8008 and static url:
[image]
[image]

dsyleixa, 2023-03-05:
update:
found that app_httpd.cpp surprisingly also in the local folder (before not observed it and not even found it by Windoes file search)...
But anyway, if I now start the url with port
192.168.2.202:8008
the website opens showing controls, nontheless there is no cam video started any more after clicking "Start Stream", and also the Flash-LED brightness control has no effect any more...

hmmh.
what can go wrong will go wrong.
Very strange though.

dsyleixa, 2023-03-05:
update:
found that app_httpd.cpp surprisingly also in the local folder (before not observed it and not even found it by Windows file search)...
But anyway, if I now start the url with port
192.168.2.202:8008
or alternatively by http://mysite.sytes.net:8008/
the website opens showing GUI controls, nontheless there is no cam video started any more after clicking "Start Stream", not even a blank white or black canvas, and also the Flash-LED brightness control has no effect any more...

hmmh.
what can go wrong will go wrong.
Very strange though.

dsyleixa, 2023-03-05:
please re-open because the issue still is not solved yet!

me-no-dev, 2023-03-06:
@dsyleixa it is possible that you would need to adjust something in the page HTML as well. The example as provided works fine, customizing and getting it to work by your requirements is another story that is outside of the scope of this issue tracker.

dsyleixa, 2023-03-06:
thank you, but my TOP question was a feature request to make it work by passing a http port.
Your proposal to add in app_httpd.cpp
config.server_port = 8008;
does not make it work correctly unfortunately, i.e.
a) the GUI shows up, albeit no video stream is processed, and
b) OTOH I can prove that the http port wors well when passing the port in proprietary esp32 and esp8266 ports via 
WiFiServer server(8008);

So the problem seems to be that the existing ESP32-CAM example still does not work correctly for http-ports yet even when config.server_port = 8008 has been added, and obviously additional things will have to be changed or extended or expanded in the given libs to make it work accordingly.
So please reopen this issue as it is not solved yet, in order to add the required features in the future.

dsyleixa, 2023-03-06:
thank you, but my TOP question was a feature request to make it work by passing a http port.
Your proposal to add in app_httpd.cpp
config.server_port = 8008;
does not make it work correctly unfortunately, i.e.
a) the GUI shows up, albeit no video stream is processed and shown up, and GUI camera controls don't work either, and
b) OTOH I can prove that the http port works well when passing the port in proprietary esp32 and esp8266 programs via
WiFiServer server(8008);

So the problem seems to be that the existing ESP32-CAM example still does not work correctly for http-ports yet even when config.server_port = 8008 has been added, and obviously additional things will have to be changed or fixed or extended or expanded in the given libs or helper/header files to make it work accordingly.
So please reopen this issue as it is not solved yet, in order to add the required features in the future.

dsyleixa, 2023-03-09:
update: module is defective, can be kept closed.

dsyleixa, 2023-03-13:
update: module has crashed meanwhile, topic can be kept closed albeit not been solved.