espressif/arduino-esp32 #7921: wifi roaming

chegewara, 2023-03-02:
### Related area

wifi

### Hardware specification

any

### Is your feature request related to a problem?

No

### Describe the solution you'd like

To enable roaming in menuconfig.

### Describe alternatives you've considered

Hi,
i have build custom arduino framework, which i can use with platformio, and it works.
After some testing i believe there is not much cons, like a significant increased flash or heap usage, and i can see pros, since wifi mesh routers are cheap and popular these days.

Thanks

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-03-06:
Added to 3.0.0 milestone. Need to investigate further :)

TilloNeko, 2023-11-16:
Also have a project that is depending on this. Bump

chegewara, 2023-11-17:
You can try this repo. It is compiled with roaming enabled:
https://github.com/Roobuck-RnD/arduino-esp32-roaming

VojtechBartoska, 2024-02-20:
Relates to https://github.com/espressif/arduino-esp32/pull/8760

closed 2024-04-03 by me-no-dev

victorprosa, 2026-08-28:
Although completed on April 3, the feature was disabled for most chips on April 17/29 by https://github.com/espressif/esp32-arduino-lib-builder/commit/4625630870e1b03deb3499e8efa5d0d50adc3e80 and https://github.com/espressif/esp32-arduino-lib-builder/commit/6307d26f45be343f42abb05b26dbe25a436ae399 remaining enabled only for the ESP32-C3, it seems.
@me-no-dev can you, please, provide some insight on what is broken with the functionality and whether there is anything the community can do to help or expect it to be re-added to the roadmap especially for chips that would benefit the most (i.e. the ESP32-C5)? The roaming app seems to have left experimental status recently (by https://github.com/espressif/esp-idf/commit/b585c0b364eddffe1b7d6fa5e926eb47a5657e9c).