espressif/arduino-esp32 #7923: Path to esptool not discoverable due to post-release tool dependency bump

per1234, 2023-03-03:
### Board

All

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v2.0.7

### IDE Name

Arduino IDE 1.8.19, 2.0.4, Arduino CLI 0.31.0

### Operating System

All

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

The [package index](https://arduino.github.io/arduino-cli/latest/package_index_json-specification/) provides support for installation of the platform via the Arduino Boards Manager. In addition to the platform itself, Boards Manager also installs the tool dependencies that are defined in the release entry.

The package index data is also used by the Arduino development tools to expand [the `runtime.tools.TOOLNAME.path` properties](https://arduino.github.io/arduino-cli/latest/platform-specification/#how-to-retrieve-tools-path-via-runtimetools-properties) used in the command patterns defined in `platform.txt` to the paths of the tool dependencies installed on the user's computer.

The [`packages[*].platforms[*].toolsDependencies[*].version`](https://arduino.github.io/arduino-cli/0.31/package_index_json-specification/#platforms-definitions:~:text=core%20archive%20file)-,toolsDependencies,-%3A%20the%20tools%20needed) field for the `esp32:esptool_py` tool dependency of the 2.0.7 platform release was changed from `4.5` to `4.5.1`:

https://github.com/espressif/arduino-esp32/commit/50269f6ec02bb341d878ff57d7ea5fc8c9ac82fe

This causes compilation or upload to fail under the following conditions:

- The user installed version 2.0.7 of the platform before [the change](https://github.com/espressif/arduino-esp32/commit/50269f6ec02bb341d878ff57d7ea5fc8c9ac82fe) was made to the package index.
- The [ESP32 package index URL](https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json) is in their "**Additional Boards Manager URLs**" IDE preference.
- The package index file has been updated on the user's computer (the Arduino development tools do this automatically).

#### Suggested Fix

Make a new release of the platform.

#### Workaround

The user can use Boards Manager to uninstall and then reinstall the platform:

##### Arduino IDE 1.x

1. Select **Tools > Board > Boards Manager** from the Arduino IDE menus.
   A "**Boards Manager**" dialog will appear.
1. Wait for the updates to finish, as shown by the messages printed at the bottom of the "**Boards Manager**" dialog.
1. Scroll down through the list of boards platforms until you find the "**esp32 by Espressif Systems**" entry. Click on it.
   Some buttons will appear on the entry.
1. Click the **<kbd>Remove</kbd>** button on the "**esp32 by Espressif Systems**" entry.
1. Wait for the removal to finish.
1. Click the **<kbd>Install</kbd>** button on the "**esp32 by Espressif Systems**" entry.
1. Wait for the installation to finish.
1. Click the **<kbd>Close</kbd>** button on the "**Boards Manager**" dialog.

##### Arduino IDE 2.x

1. Select **Tools > Board > Boards Manager** from the Arduino IDE menus to open the "**Boards Manager**" view in the left side panel.
1. Scroll down through the list of boards platforms until you find the "**esp32 by Espressif Systems**" entry.
1. Hover the mouse pointer over the "**INSTALLED**" label on the "**esp32 by Espressif Systems**" entry. You will now see it change to an "**UNINSTALL**" button.
1. Click the "**UNINSTALL**" button.
1. A confirmation dialog will appear. Click the "**YES**" button.
1. Wait for the uninstallation to finish.
1. Select "**2.0.7**" from the dropdown version menu at the bottom of the "**esp32 by Espressif Systems**" entry.
1. Click the "**INSTALL**" button at the bottom of the "**esp32 by Espressif Systems**" entry.
1. Wait for the installation to finish.

#### Additional reports

- https://github.com/arduino/Arduino/issues/11866
- https://forum.arduino.cc/t/t-display-s3-esp32s3-how-to-reset-to-factory-settings-after-burn-bootloader/1096636
- https://forum.arduino.cc/t/esptool-exe-is-not-recognized-as-an-internal-or-external-command-an-executable-program-or-batch-file/1096495
- https://forum.arduino.cc/t/esp32-s3-stuck-with-manual-uploading-resetting/1096945/3
- https://forum.arduino.cc/t/compilation-error-esptool-exe-not-recognized/1096550
- https://forum.arduino.cc/t/problem-latest-update-ide/1095897/5

### Sketch

```cpp
void setup() {}
void loop() {}
```

### Debug Message

```plain
**Arduino IDE 2.x** / **Arduino CLI**:

cmd /c IF EXIST "C:\\Users\\per\\Documents\\Arduino\\sketch_mar2a\\bootloader.bin" ( COPY /y "C:\\Users\\per\\Documents\\Arduino\\sketch_mar2a\\bootloader.bin" "C:\\Users\\per\\AppData\\Local\\Temp\\arduino-sketch-D4D9383D25724F4612E543EFB0213587\\sketch_mar2a.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\per\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.7\\variants\\esp32\\bootloader.bin" ( COPY "C:\\Users\\per\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.7\\variants\\esp32\\bootloader.bin" "C:\\Users\\per\\AppData\\Local\\Temp\\arduino-sketch-D4D9383D25724F4612E543EFB0213587\\sketch_mar2a.ino.bootloader.bin" ) ELSE ( /esptool.exe --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\per\\AppData\\Local\\Temp\\arduino-sketch-D4D9383D25724F4612E543EFB0213587\\sketch_mar2a.ino.bootloader.bin" "C:\\Users\\per\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.7\\tools\\sdk\\esp32\\bin\\bootloader_qio_80m.elf" ) )
'/esptool.exe' is not recognized as an internal or external command,
operable program or batch file.
        1 file(s) copied.

**Arduino IDE 1.x**:

```text
___REMOVE___/esptool.exe --chip esp32 --port COM10 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 C:\Users\per\AppData\Local\Temp\arduino_build_839818/foo_bar.ino.bootloader.bin 0x8000 C:\Users\per\AppData\Local\Temp\arduino_build_839818/foo_bar.ino.partitions.bin 0xe000 C:\Users\per\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/partitions/boot_app0.bin 0x10000 C:\Users\per\AppData\Local\Temp\arduino_build_839818/foo_bar.ino.bin 
java.io.IOException: Cannot run program "___REMOVE___/esptool.exe": CreateProcess error=2, The system cannot find the file specified
```
```

### Other Steps to Reproduce

1. Use Boards Manager to install version 2.0.7 of the platform if it is not already installed:
   https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide
1. If you have installed the platform after the time of the package index change, adequately simulate the conditions by renaming the folder at the following path to `4.5`:
   - **Windows**:
     ```text
     C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1
     ```
     (where `<username>` is your Windows username)
   - **Linux**:
     ```text
     ~/.arduino15/packages/esp32/tools/esptool_py/4.5.1
     ```
   - **macOS**:
     ```text
     ~/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1
     ```
1. Trigger an update of the package index:
   - **Arduino IDE 1.x**:
     1. Select **Tools > Board > Boards Manager** from the Arduino IDE menus.
        A "**Boards Manager**" dialog will appear.
     1. Wait for the updates to finish, as shown by the messages printed at the bottom of the "**Boards Manager**" dialog.
     1. Click the **<kbd>Close</kbd>** button on the "**Boards Manager**" dialog.
   - **Arduino IDE 2.x**:
     1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
     1. Select the "**Arduino: Update Package Index**" command from the menu.
     1. Wait for the "**Downloading index: ...**" notification to close.
1. Select any ESP32 board from Arduino IDE's **Tools > Board** menu.
1. Select a serial port from Arduino IDE's **Tools > Port** menu.
1. Select **Sketch > Upload** from the Arduino IDE menus.

🐛 The process fails due to an incomplete path to `esptool`.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-03-06:
new release is coming out very soon. Waiting on a couple of fixes in some of the components to be released.

Jason2866, 2023-04-21:
Fixed with release of 2.0.8

closed 2023-04-21 by Jason2866

877dev, 2023-05-07:
I don't understand the issue, but appreciate OP's effort and the subsequent fix!