espressif/arduino-esp32 #7925: Some Files seems to be missing in /bits

4Source, 2023-03-04:
### Board

XIAO_ESP32C3

### Device Description

Nothing.

### Hardware Configuration

Nothing

### Version

v2.0.7

### IDE Name

Arduino IDE 2.0.4

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

If I try to compile any project there are missing files. 

Didn't exist added the file manually.
"arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory" 

Then this file exist but can be included.
"arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\riscv32-esp-elf\rv32imc\ilp32\no-rtti\bits\gthr.h:151:10: fatal error: bits/gthr-default.h: No such file or directory"

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
In file included from c:\users\xxxxx.desktop-3pbgges\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\ext\atomicity.h:35,
                 from c:\users\xxxxx.desktop-3pbgges\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\basic_string.h:39,
                 from c:\users\xxxxx.desktop-3pbgges\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\string:52,
                 from c:\users\xxxxx.desktop-3pbgges\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\xxxxx.desktop-3pbgges\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\array:39,
                 from c:\users\xxxxx.desktop-3pbgges\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\xxxxx.desktop-3pbgges\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\functional:54,
                 from C:\Users\xxxxx.desktop-3pbgges\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\cores\esp32/HardwareSerial.h:49,
                 from C:\Users\xxxxx.desktop-3pbgges\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\cores\esp32/Arduino.h:184,
                 from C:\Users\xxxxx.desktop-3pbgges\AppData\Local\Temp\arduino\sketches\06982C1A17F19B9AF771A54228B9B2F6\sketch\TabakDryer.ino.cpp:1:
c:\users\xxxxx.desktop-3pbgges\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\riscv32-esp-elf\rv32imc\ilp32\no-rtti\bits\gthr.h:151:10: fatal error: bits/gthr-default.h: No such file or directory
 #include <bits/gthr-default.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

4Source, 2023-03-04:
Installed previous (2.0.4) seems to work fine for me.

Jason2866, 2023-03-04:
This file `<bits/gthr-default.h>` does not exists in arduino-es32

VojtechBartoska, 2023-03-09:
@me-no-dev Please help checking this? Thanks

4Source, 2023-03-09:
I saw while trying different versions there changed a lot with those paths. May you checkt the path for esp32 with the not the RISC-V architecture. (See path: riscv32-esp-elf-gcc) but for me they exists at Version > 2.0.6 not sure there exists at 2.0.4.

me-no-dev, 2023-03-13:
@VojtechBartoska those are inside the toolchain. Should be reported to `espressif/crosstool-NG`. Maybe there is a reason/workaround

closed 2025-01-16 by Parsaabasi