espressif/arduino-esp32 #7927: [documentation] ESP32-CAM Board: LED does not light up: wrong LED pin definition?

dsyleixa, 2023-03-04:
### Related area

esp32-cam

### Hardware specification

esp32-cam  CAMERA_MODEL_AI_THINKER // Has PSRAM

### Is your feature request related to a problem?

the auto-generated website has lot of control widgest, e.g. the LED Intensitiy slider widget.
But whe dragging the slider, nothing happens o my onboard-LED, it's always off.
I am using the 
#define CAMERA_MODEL_AI_THINKER // Has PSRAM

are there perhaps wrong LED pin definitions in a config file?
Which #defines as to the built-in LED have to be changed probably?
Where can I possibly find documentations about that?

### Describe the solution you'd like

correct on-board LED control

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Hieromon, 2023-03-05:
@dsyleixa, The LED pin is **GPIO33** of Ai-Thinker ESP32-CAM , which is coupled to the PWM channel. This is not necessarily a mistake.

https://github.com/espressif/arduino-esp32/blob/0d84018d969309addacbcc3e3782c1fadc95fbc8/libraries/ESP32/examples/Camera/CameraWebServer/camera_pins.h#L158

[image]

This red LED is placed on the opposite side of the ESP32-CAM from the camera. It is a 1608 LED for signals with less than 100mW output, so you might miss it. But it does change the brightness gracefully according to the `LED Intensity` in the Settings panel of the Camera Control WebUI.

In fact, my ESP32-CAM changes brightness depending on the LED Intensity value. PWM by ledc is working fine. However, it is far from being a **flash LED** as the function name suggests.

https://github.com/espressif/arduino-esp32/blob/0d84018d969309addacbcc3e3782c1fadc95fbc8/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp#L1388-L1391

On the other hand, the connection pin for the other LED equipped by ESP32-CAM (it has a white 1W-class output and is placed on the same side as the camera) is **GPIO4**.

https://docs.ai-thinker.com/_media/esp32/docs/esp32_cam_sch.pdf

Maybe just change the [pin definition](https://github.com/espressif/arduino-esp32/blob/0d84018d969309addacbcc3e3782c1fadc95fbc8/libraries/ESP32/examples/Camera/CameraWebServer/camera_pins.h#L158) to `4` to get the flash LED working as you expect.

```cpp
#define LED_GPIO_NUM      4
// #define LED_GPIO_NUM      33
```

But, I do not recommend changing it thoughtlessly. According to the schematic, this flash LED has no resistor (just a transistor for sinking current drive according to PWM). VF=3.3V is applied directly. It will probably **burn**.

[image]

In order for you to get this flash LED functionality, I think you should at least do the following:
1. Insert at least a 1K resistor between ANODE of the flash LED and 3V3.
2. Change the maximum PWM duty to 127. It defined with `CONFIG_LED_MAX_INTENSITY` macro in *app_httpd.cpp*.

And so, I thought it would be better to be at peace with a tiny red LED than to have ESP32-CAM boards burning up all over the world. Of course, this is my personal opinion and is separate from how the maintainer @P-R-O-C-H-Y judges.

dsyleixa, 2023-03-05:
thank you very much for your explanations!
In fact, the red on-board LED downside is always on, and it's intensitiy doesn't change by the slider.
Soldering is not possible unfortunately because of certain disablities.
Changing/patching app_httpd.cpp is no option either because that is fixed and compressed in a .zip file in an awkward nested Arduino IDE folder (CMIIW) - it should be possible to change those values arbitrarily in either proprietary user program.

I think I'll have to try just GPIO4, and if it burns, it burns, because it's of no use currently anyway.... :-/

Hieromon, 2023-03-05:
@dsyleixa, You are a challenger! Don't forget a glass of water.

dsyleixa, 2023-03-05:
just observed:
there is already a GPIO4 pin usage:
#define XCLK_GPIO_NUM     4

does that really not matter as to LED_GPIO_NUM    4?

Hieromon, 2023-03-05:
@dsyleixa, The [`XCLK_GPIO_NUM`](https://github.com/espressif/arduino-esp32/blob/0d84018d969309addacbcc3e3782c1fadc95fbc8/libraries/ESP32/examples/Camera/CameraWebServer/camera_pins.h#L142) of the Ai-Thinker ESP32-CAM is assigned to `Cam_Connector XCLK1`. This is GPIO0 and correct.

However, the circuit around the `LED_FLASH` in ESP32-CAM is inherently problematic. GPIO4 is shared with `HS2 DATA1` of Micro SD I/F as well as driving `LED_FLASH`. In other words, you cannot write to the Micro SD properly with the LED_FLASH applied.

Well, to be more details, there are various other dangers, but don't expect too much from this kind of board.

dsyleixa, 2023-03-05:
thank you very much!
Unfortunately I have not much information about all this stuff - especially no easily understandable for me as a hobby programmer. I do not really see for which purposes the SD would be needed actually

dsyleixa, 2023-03-09:
update: module is defective, can be closed.

NgocLe97, 2024-06-08:
hi, i have solved this problem,
in file "app_httpd.cpp"

#if CONFIG_LED_ILLUMINATOR_ENABLED

#define LED_LEDC_GPIO            4  //configure LED pin
#define CONFIG_LED_MAX_INTENSITY 255

[image]