espressif/arduino-esp32 #7928: Sanity check

FernandoGarcia, 2023-03-04:
### Related area

GPIO usage and chip features compatibily

### Hardware specification

All boards

### Is your feature request related to a problem?

We have discussed [here](https://github.com/espressif/arduino-esp32/issues/7893#issuecomment-1453806397) about a problem of incompatibility between GPIO usage and ESP32 PSRAM feature.

The example switch the RainMaker library uses GPIO 16 as outupt at same time this GPIO is used by PSRAM feature that is enabled by default for ESP32 chip.

So it has caused an error of memory allocation when using the core 2.0.7.

Moreover there's others problems like GPIO 1 and 3 being used as INPUT or OUTPUT together with serial monitor and so on.

Also, we have found a number issues like `E (355) gpio: gpio_set_level(226): GPIO output gpio_num error` . This error message doesn't explain the root of the problem and doesn't help so much.

### Describe the solution you'd like

To solve this problem I have proposed a implementation called sanity check similar to feature used in [Marlin Firmware](https://github.com/MarlinFirmware/Marlin/blob/579833e2a3d8ab8716fcab85dd26904a718fcec8/Marlin/src/inc/SanityCheck.h) where the features compatibilities are checked to each feature enabled on sketch.

In this case the compatibility between GPIO usage on sketch and CHIP features are checked. 
In case of incompatibility an compiling error will shown with clear description of the problem, so the user can fix it without ask in forums or here the meaning of such error.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-03-06:
This is not necessary. The problem comes from RainMaker and will be fixed. We should not have such issues in the future.

FernandoGarcia, 2023-03-06:
This is not a particular issue with RainMaker.

Looking for [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html#main-i-o-connector-jp1) documentation for example we can see that the misusage of GPIO 6 to 11 can result in unexpected behaviour.

[image]

igrr, 2023-03-06:
FWIW, in IDF v5.1 we are adding a feature to reserve the GPIOs; this will take care of Flash/PSRAM conflicts, among other things. So this might be addressed in arduino-esp32 v3.0 without extra work required.

Jason2866, 2023-03-07:
ESP32 variants (pico) do have PSRAM connected to other GPIOs. So general fixed sanity checks makes no sense at all. Provided examples should avoid GPIOs which are dangerous.

FernandoGarcia, 2023-03-07:
As I told before...

>To make it work to all boards you can use a bunch of #ifdef like [Marlin firmware](https://github.com/MarlinFirmware/Marlin/blob/579833e2a3d8ab8716fcab85dd26904a718fcec8/Marlin/src/inc/SanityCheck.h) does. :)

>Well, I'm talking about something like this pseudocode:

```
int gpio_get_level(gpio_num_t gpio_num);
{
  if(gpio_num == 16)
  {
    return true;
  }
  return false;
}

#if defined(ESP_32) && defined(USE_PSRAM)
  if(gpio_get_level(16) == true)
    {
      #error "GPIO 16 can't be used when PSRAM is enabled"
    }
#endif
```

Jason2866, 2023-03-07:
you forgot GPIO18 and GPIO23 which can be used for PSRAM too, and for ESP32-S3?
And this are just recommendations. You can connect PSRAM to nearly every GPIO.
Just a define in sdkconfig in IDF. A lot of ITEAD devices do NOT use GPIO 16/17 for PSRAM.

Jason2866, 2023-03-07:
It will be a fight against windmills. If the user don't know exactly what board (MCU Variant) used this will be a useless attempt. If knowing this is not needed at all, since avoiding because knowing restricted GPIOs

FernandoGarcia, 2023-03-07:
Ok, you gave your opinion now wait for others.

closed 2023-06-03 by FernandoGarcia