espressif/arduino-esp32 #7929: Old json installation link conflict - New core does not appear

rtek1000, 2023-03-04:
### Board

ESP32

### Device Description

Not applied

### Hardware Configuration

Not applied

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.8.19

### Operating System

Linux Xubuntu 22.04

### Flash frequency

Not applied

### PSRAM enabled

yes

### Upload speed

Not applied

### Description

https://dl.espressif.com/dl/package_esp32_index.json

[image]

### Sketch

```cpp
Not applied
```

### Debug Message

```plain
Not applied
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-03-06:
Hi :) What is the actual issue? Just switch to the new link.

rtek1000, 2023-03-06:
The problem occurs when you have this core version 1 and version 2 link, but only the core version 1 appears to install. If one has many core links, it can get a little confusing.

It's more of a tip than a problem to solve.

I noticed that due to the v2 core it generates duplicate library warnings as well. But they are ESP8266 core libraries.

rtek1000, 2023-03-14:
I noticed that it is not enough to remove the link, you have to uninstall the core version 1 for the core version 2 to appear in the manager.

VojtechBartoska, 2023-10-04:
There is nothing we can do about this, closing as won't fix.

closed 2023-10-04 by VojtechBartoska