espressif/arduino-esp32 #7930: I2S cannot work at low frequency (CPU freq<80MHz)

HelloDB, 2023-03-04:
### Board

esp32-c3-devkitm-1

### Device Description

esp32-c3-devkitm-1
MAX98357AETE+T

### Hardware Configuration

I2S.setSckPin(6);
I2S.setFsPin(7);
I2S.setDataPin(10);

### Version

latest master (checkout manually)

### IDE Name

platformio

### Operating System

WIN10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I2S cannot work at low frequency (CPU freq<80MHz)

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/I2S/examples/SimpleTone/SimpleTone.ino

Processing esp32c3 (board: esp32-c3-devkitm-1; framework: arduino; platform: espressif32)
--------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (6.0.1) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 40MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP8266Audio @ 1.9.7
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- I2S @ 1.0
|   |-- FS @ 2.0.0
|   |-- HTTPClient @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- SD @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SPI @ 2.0.0
|-- OneButton @ 2.0.3
|-- I2S @ 1.0
Building in release mode
Retrieving maximum program size .pio\build\esp32c3\firmware.elf
Checking size .pio\build\esp32c3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.8% (used 15832 bytes from 327680 bytes)
Flash: [===       ]  25.1% (used 262752 bytes from 1048576 bytes)
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5810,len:0x38c
load:0x403cc710,len:0x6ac
load:0x403ce710,len:0x24e4
entry 0x403cc710
I2S simple tone
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-07:
Issue confirmed. 
I2S Arduino doesn't user APLL as clock source.

Line 147: `.use_apll = false`
https://github.com/espressif/arduino-esp32/blob/ca77502ceb8284d306fe32eeac0a88377396539d/libraries/I2S/src/I2S.cpp#L138-L148

Maybe you could change it in your computer (locally)  to `.use_apll = true` and test it to check if it would work using CPU freq lower than 160MHz.

Let me know. Thanks!

HelloDB, 2023-03-08:
Thanks, but it doesn't work below 80MHz (tested 40MHz, 20MHz, 10MHz), I also tried to enable Audio PPL CLK by modifying `int ret = begin(mode, sampleRate, bitsPerSample, false);`, it still no sound

https://github.com/espressif/arduino-esp32/blob/0d84018d969309addacbcc3e3782c1fadc95fbc8/libraries/I2S/src/I2S.cpp#L226
https://github.com/espressif/arduino-esp32/blob/0d84018d969309addacbcc3e3782c1fadc95fbc8/libraries/I2S/src/I2S.cpp#L150-L153

closed 2025-01-16 by Parsaabasi