espressif/arduino-esp32 #7933: i2c read data error after 2.0.5

flybeyond, 2023-03-06:
### Board

ESP32S3 Dev Module

### Device Description

ESP32S3

### Hardware Configuration

GPIO8&9

### Version

v2.0.7

### IDE Name

Arduino

### Operating System

FreeRTOS

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Dear all,
After upgrading to v2.0.6 and v2.0.7, i2c read out value is not correct. When I read ICM42688 chip id, the expected value is 0x47 but read out value is 0x07. The same issue happens in other registers. Used function is Wire.requestfrom(). Any suggestions?

v2.0.6 and v2.0.7 address 0x00-0x7f
00 00 00 00 00 00 00 00 00 00 00 24 00 00 00 00 
00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 20 01 00 04 
04 04 01 09 03 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 C0 07 00 00 00 00 00 00 00 00 00 00

v2.0.5 and before address 0x00-0x7f
00 00 00 00 00 A2 00 00 00 00 00 B6 00 00 00 00 
00 00 00 05 00 00 00 00 10 00 10 00 00 80 00 80 
00 80 00 80 00 80 00 80 00 80 00 00 00 00 00 00 
FF 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 30 91 00 06 
06 16 11 0D 23 00 82 00 10 00 00 00 00 00 00 00 
00 00 10 00 10 10 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 E8 47 00 00 00 00 00 00 00 00 00 00 

### Sketch

```cpp
static bool imuI2cWrite(uint8_t regAddress, uint8_t* pdata, size_t len)
{
 Wire.beginTransmission(ICM42688_I2C_ADDR);
 Wire.write(regAddress);
 for(size_t i = 0; i < len; i++)
  Wire.write(*pdata++);
 if(Wire.endTransmission())
  return false;
 return true;
}

static bool imuI2cRead(uint8_t regAddress, uint8_t* pdata, size_t len)
{
 Wire.beginTransmission(ICM42688_I2C_ADDR);
 Wire.write(regAddress);
 if(Wire.endTransmission(0))
  return false;
 if(Wire.requestFrom(ICM42688_I2C_ADDR, len) == len)
 {
  for(size_t i = 0; i < len; i++)
   *pdata++ = Wire.read();
  return true;
 }
 return false;
}
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-07:
@me-no-dev - do you have any suggestion here?

mrengineer7777, 2023-03-08:
I2C library hasn't changed since 2.0.3.  Related changes: Heltec I2C pinout in 2.0.7 and wt32-eth01 pinout in 2.0.6.

flybeyond, 2023-03-10:
That's weird, but I didn't change any code, use 2.0.5 i2c works, but 2.0.6 and 2.0.7 don't work.

mrengineer7777, 2023-03-10:
What is the manufacturer of your dev module?  Are you using any third party libraries?  Have you tried creating a test project with just I2C?

flybeyond, 2023-03-16:
The access library is not third party. I will try a test project later. Anyway, the only change for me is upgrade 2.0.5 to 2.0.6 or 2.0.7 it will not work.

closed 2025-01-16 by Parsaabasi