espressif/arduino-esp32 #7936: Flash Read Failed - OTA Update

malyjak, 2023-03-07:
### Board

ESP 32 Dev Module

### Device Description

DevKitC on plain breadboard

### Hardware Configuration

None

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

After calling
```
Update.begin(UPDATE_SIZE_UNKNOWN, cmd)
```

from Update library, where `cmd` is:
```
cmd = (filename == "filesystem") ? U_SPIFFS : U_FLASH;
```

in case of `U_FLASH` an error is produced in serial monitor:

```
Flash Read Failed
```

and the update process is terminated.

In case of `U_SPIFFS `, everything goes well.

### Sketch

```cpp
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final)
{
  if (!index)
  {
    int cmd = (filename == "filesystem") ? U_SPIFFS : U_FLASH;
    if (!Update.begin(UPDATE_SIZE_UNKNOWN, cmd))
    {
      Update.printError(Serial);
      return request->send(400, "text/plain", "OTA could not begin");
    }
  }

  // Flash the new firmware.
  if (len)
  {
    if (Update.write(data, len) != len) {
      return request->send(400, "text/plain", "OTA could not write");
    }
  }

  if (final)
  {
    if (Update.end(true))
    {
      Serial.println("Update successful!");
      Serial.println("Rebooting...");
    }
    else
    {
      Update.printError(Serial);
    }
  }
} // End: handleUpload()
```

### Debug Message

```plain
Flash Read Failed
```

### Other Steps to Reproduce

I'm using https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json in Arduino IDE 1.8.19 to get this BSP. The reason for 1.8.19 is that I need https://github.com/me-no-dev/arduino-esp32fs-plugin to save web server data on the file system and 2.x seems not to support external tools.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-07:
It seems to be a partition error. 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html
https://blog.espressif.com/how-to-use-custom-partition-tables-on-esp32-69c0f3fa89c8

malyjak, 2023-03-08:
You might be right. However, I checked the first partition scheme from the Arduino IDE menu with `gen_esp32part` and it is actually quite similar to the “Factory app, two OTA definitions” configuration

[image]
Creates this:
[image]
Which is similar to:
[image]

The partition table starts at 0x8000 instead of 0x9000 for my esp32.

asushugo, 2023-03-08:
Board
ESP32-C3 

Version
v2.0.7

IDE Name
VS Code (PlatformIO plugin)

I have the same problem
[image]

And this is my partition
[image]

asushugo, 2023-03-09:
I replaced the v2.0.6, and it works
[image]

malyjak, 2023-03-09:
> I replaced the v2.0.6, and it works [image]

@asushugo can you please share your `platformio.ini`? I tried:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.6
monitor_speed = 115200
board_build.partitions = partitions.csv
lib_deps = 
	me-no-dev/ESP Async WebServer@^1.2.3
	arduino-libraries/LiquidCrystal@^1.0.7
	adafruit/Adafruit Unified Sensor@^1.1.9
	adafruit/DHT sensor library@^1.4.4
```

And I still haven't succeeded in making this error go away.

I set the `partitions.csv` to be identical with the one generated by the Arduino IDE:
```
# Name,   Type, SubType, Offset,   Size
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,         20K
otadata,  data, ota,     ,         8K
ota_0,    0,    ota_0,   ,         1280K
ota_1,    0,    ota_1,   ,         1280K
spiffs,   data, spiffs,  ,         1408K
coredump, data, coredump,,         64K
```

asushugo, 2023-03-09:
`# Name,   Type, SubType, Offset,  Size,   Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
ota_0,    app,  ota_0,         ,  1M    ,
ota_1,    app,  ota_1,         ,  1M    ,`

`[env:esp32-c3-16M-openluat]
platform = espressif32
board = esp32-c3-16M-openluat
framework = arduino
monitor_speed = 115200
board_build.partitions = partition.csv
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.6

upload_protocol = espota
upload_port = 192.168.8.201`

SuGlider, 2023-03-09:
What is the size in bytes of the firmware are both are trying to flash using OTA?
Does it fit in one OTA partition?

malyjak, 2023-03-10:
Yest, It's 1061K, so it fits well into 1280K. I've managed to solve the problem by basically wiping everything, just keeping the code and building the binary from scratch once again. We were both probably uploading corrupted/incompatible firmware because now the error disappeared and I have working OTA, regardless of the arduino-esp32 version. Sorry for calling to label this as a bug when it was an error on the user side :space_invader:

closed 2023-03-10 by malyjak