espressif/arduino-esp32 #7945: Serial Communication Possibly Broken after Update

rahmed31, 2023-03-09:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module using ESP32-D0WD-V3 chip.
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None, CH340 serial communication chip

### Hardware Configuration

No connected GPIOs, just the ESP32 board itself.  

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

MacOS Ventura 13.2.1

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

No longer able to connect to any ESP32 Dev Modules after updating to v2.0.7. I have tried three separate boards with the same specifications described above, all of which were in working condition prior to applying Espressif updates last weekend on 03/05/2023. Even after downgrading versions, the same error occurs. 

### Sketch

```cpp
No sketches can be uploaded to any ESP32 Dev Module that I own.
```

### Debug Message

```plain
Sketch uses 789557 bytes (60%) of program storage space. Maximum is 1310720 bytes.
Global variables use 45188 bytes (13%) of dynamic memory, leaving 282492 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbserial-14220
Connecting......
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e0:5a:1b:5f:33:30
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.

A fatal error occurred: Unable to verify flash chip connection (Packet content transfer stopped (received 6 bytes)).
Failed uploading: uploading error: exit status 2
```

### Other Steps to Reproduce

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-09:
I think that it is a problem with the Arduino installation.
Please remove the ESP32 boards using the Board Manager menu.
Then install the latest Arduino Core 2.0.7 again.

I can assure that it works correctly. I use it everyday.

SuGlider, 2023-03-09:
@me-no-dev - Could it be something related to MacOS and/or ESPTool 4.5.1?

rahmed31, 2023-03-09:
> @me-no-dev - Could it be something related to MacOS and/or ESPTool 4.5.1?

Hi, thank you for the response. I believe I figured out the issue. For some unknown reason, it seems like the default upload speed for ESP32 boards on Arduino IDE changed to 921600 as opposed to the typical speed of 115200. I also explicitly specify the Baudrate as 115200 in my code. I changed the upload speed back to 115200 and I am now able to upload my sketches. Do you know why 921600 was the default? I'm not sure if this might've been a result of updating my OS to Ventura 13.2.1, arduino-esp32 to v2.0.7, or just applying the usual Arduino IDE updates. Unfortunately, I did all three of these in that order.

closed 2023-03-09 by rahmed31