espressif/arduino-esp32 #7949: ESP32-S3 SD card formatter

lcv31913, 2023-03-10:
### Board

ESP32-S3

### Device Description

ESP32-S3-DEVKITC-1-N8R8

### Hardware Configuration

micro SD card model，6 PIN(VCC、GND、MISO、MOSI、SLK、CS)

### Version

latest master (checkout manually)

### IDE Name

arduino IDE

### Operating System

windows10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Vcc    >  3.3V
GND  > GND
MOSI > 13
MISO > 11
SLK    > 12
CS      > 10

I am using SPI to connect a micro SD card model, but I am getting a mount fail error. Do I need to format the SD card using Windows 10, or just run SD.begin(cs_pin)? Also, is there any library that can format the SD card to FAT32, or do I need to send SD commands to complete the process?

### Sketch

```cpp
/*
   Connect the SD card to the following pins:

   SD Card | ESP32
      D2       -
      D3       SS
      CMD      MOSI
      VSS      GND
      VDD      3.3V
      CLK      SCK
      VSS      GND
      D0       MISO
      D1       -
*/
#include "FS.h"
#include "SD.h"
#include "SPI.h"

void listDir(fs::FS &fs, const char * dirname, uint8_t levels) {
  Serial0.printf("Listing directory: %s\n", dirname);

  File root = fs.open(dirname);
  if (!root) {
    Serial0.println("Failed to open directory");
    return;
  }
  if (!root.isDirectory()) {
    Serial0.println("Not a directory");
    return;
  }

  File file = root.openNextFile();
  while (file) {
    if (file.isDirectory()) {
      Serial0.print("  DIR : ");
      Serial0.println(file.name());
      if (levels) {
        listDir(fs, file.path(), levels - 1);
      }
    } else {
      Serial0.print("  FILE: ");
      Serial0.print(file.name());
      Serial0.print("  SIZE: ");
      Serial0.println(file.size());
    }
    file = root.openNextFile();
  }
}

void createDir(fs::FS &fs, const char * path) {
  Serial0.printf("Creating Dir: %s\n", path);
  if (fs.mkdir(path)) {
    Serial0.println("Dir created");
  } else {
    Serial0.println("mkdir failed");
  }
}

void removeDir(fs::FS &fs, const char * path) {
  Serial0.printf("Removing Dir: %s\n", path);
  if (fs.rmdir(path)) {
    Serial0.println("Dir removed");
  } else {
    Serial0.println("rmdir failed");
  }
}

void readFile(fs::FS &fs, const char * path) {
  Serial0.printf("Reading file: %s\n", path);

  File file = fs.open(path);
  if (!file) {
    Serial0.println("Failed to open file for reading");
    return;
  }

  Serial0.print("Read from file: ");
  while (file.available()) {
    Serial0.write(file.read());
  }
  file.close();
}

void writeFile(fs::FS &fs, const char * path, const char * message) {
  Serial0.printf("Writing file: %s\n", path);

  File file = fs.open(path, FILE_WRITE);
  if (!file) {
    Serial0.println("Failed to open file for writing");
    return;
  }
  if (file.print(message)) {
    Serial0.println("File written");
  } else {
    Serial0.println("Write failed");
  }
  file.close();
}

void appendFile(fs::FS &fs, const char * path, const char * message) {
  Serial0.printf("Appending to file: %s\n", path);

  File file = fs.open(path, FILE_APPEND);
  if (!file) {
    Serial0.println("Failed to open file for appending");
    return;
  }
  if (file.print(message)) {
    Serial0.println("Message appended");
  } else {
    Serial0.println("Append failed");
  }
  file.close();
}

void renameFile(fs::FS &fs, const char * path1, const char * path2) {
  Serial0.printf("Renaming file %s to %s\n", path1, path2);
  if (fs.rename(path1, path2)) {
    Serial0.println("File renamed");
  } else {
    Serial0.println("Rename failed");
  }
}

void deleteFile(fs::FS &fs, const char * path) {
  Serial0.printf("Deleting file: %s\n", path);
  if (fs.remove(path)) {
    Serial0.println("File deleted");
  } else {
    Serial0.println("Delete failed");
  }
}

void testFileIO(fs::FS &fs, const char * path) {
  File file = fs.open(path);
  static uint8_t buf[512];
  size_t len = 0;
  uint32_t start = millis();
  uint32_t end = start;
  if (file) {
    len = file.size();
    size_t flen = len;
    start = millis();
    while (len) {
      size_t toRead = len;
      if (toRead > 512) {
        toRead = 512;
      }
      file.read(buf, toRead);
      len -= toRead;
    }
    end = millis() - start;
    Serial0.printf("%u bytes read for %u ms\n", flen, end);
    file.close();
  } else {
    Serial0.println("Failed to open file for reading");
  }

  file = fs.open(path, FILE_WRITE);
  if (!file) {
    Serial0.println("Failed to open file for writing");
    return;
  }

  size_t i;
  start = millis();
  for (i = 0; i < 2048; i++) {
    file.write(buf, 512);
  }
  end = millis() - start;
  Serial0.printf("%u bytes written for %u ms\n", 2048 * 512, end);
  file.close();
}

void setup() {
  pinMode(10, OUTPUT);
  
  Serial0.begin(115200);
  
  Serial0.println("--------------------");
  Serial0.println(SS);
  Serial0.println(SCK);
  Serial0.println(MOSI);
  Serial0.println(MISO);
  Serial0.println("--------------------");
  
  if (!SD.begin(10)) {
    Serial0.println("Card Mount Failed");
    return;
  }
  uint8_t cardType = SD.cardType();

  if (cardType == CARD_NONE) {
    Serial0.println("No SD card attached");
    return;
  }

  Serial0.print("SD Card Type: ");
  if (cardType == CARD_MMC) {
    Serial0.println("MMC");
  } else if (cardType == CARD_SD) {
    Serial0.println("SDSC");
  } else if (cardType == CARD_SDHC) {
    Serial0.println("SDHC");
  } else {
    Serial0.println("UNKNOWN");
  }

  uint64_t cardSize = SD.cardSize() / (1024 * 1024);
  Serial0.printf("SD Card Size: %lluMB\n", cardSize);

  listDir(SD, "/", 0);
  createDir(SD, "/mydir");
  listDir(SD, "/", 0);
  removeDir(SD, "/mydir");
  listDir(SD, "/", 2);
  writeFile(SD, "/hello.txt", "Hello ");
  appendFile(SD, "/hello.txt", "World!\n");
  readFile(SD, "/hello.txt");
  deleteFile(SD, "/foo.txt");
  renameFile(SD, "/hello.txt", "/foo.txt");
  readFile(SD, "/foo.txt");
  testFileIO(SD, "/test.txt");
  Serial0.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
  Serial0.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}

void loop() {
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
--------------------
10
12
11
13
--------------------
Card Mount Failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-03-13:
@P-R-O-C-H-Y can you please help with triage of this issue? Thanks

mrnsms, 2023-03-14:
i also cant mount any sd card with esp32 s3 n8r8

nickodimus, 2023-03-18:
I also am having this issue.  Followed the tutorial at:
https://microcontrollerslab.com/microsd-card-esp32-arduino-ide/

using the newest esp32 wroom 32E with the breakout board provided by xikentec.com HDO23 ver 1.1

21:12:25.047 -> ets Jul 29 2019 12:21:46
21:12:25.047 -> 
21:12:25.047 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:12:25.047 -> configsip: 0, SPIWP:0xee
21:12:25.047 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:12:25.047 -> mode:DIO, clock div:1
21:12:25.047 -> load:0x3fff0030,len:1344
21:12:25.047 -> load:0x40078000,len:13924
21:12:25.047 -> ho 0 tail 12 room 4
21:12:25.047 -> load:0x40080400,len:3600
21:12:25.047 -> entry 0x400805f0
21:12:25.810 -> Card Mount Failed

nickodimus, 2023-03-18:
I solved the problem on my end.  The SD card adapter as shown in the tutorial I followed needs to be powered by the 5V pin of the esp32 not the 3V3 pin.  I hope this helps some other people out.

lcv31913, 2023-03-26:
i solved this problem that my Dupont wires are broken. Sorry for this stupid problem.

closed 2023-03-26 by lcv31913