espressif/arduino-esp32 #7950: ESP32 SPIFFS x Timer Crashing

jpvieir, 2023-03-10:
### Board

ESP32

### Device Description

ESP32-WROOM-DA

### Hardware Configuration

Potentiometer connected to Analog PORT0 (just to test if the data is coming through). 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm trying to create a program that acquire audio through ESP32 and send it to a server hosted in my PC. Since audio occupies a lot of space and the RAM could not store all data I want, I'm trying to make recordings not bigger than 1 minute, and save the data inside ESP's flash memory through SPIFFS. I want a defined sampling frequency, so I'm using a Timer to acquire the data. 

The line that makes the program crash is "file.write(acquired_data,bytenum);", which call the SPIFFS. I've seen some posts telling to add "IRAM_ATTR" or "DRAM_ATTR" before the functions, but none of that worked for me. 
PS: I've provided a simpler code that I was using to debug/identify where is the error.

### Sketch

```cpp
// node-red server + mosquitto (to establish comunnication) + timer-driven acquistion + spiffs (to save data) 

extern "C" {
  #include "freertos/FreeRTOS.h"
  #include "freertos/timers.h"
}

#include <driver/adc.h>
#include <driver/adc_deprecated.h>

#include <SPIFFS.h>
#define _OPEN_SYS_ITOA_EXT

#define amostras 12000
byte  msg[amostras*2];
bool pub = false;

hw_timer_t * timer = NULL;
volatile SemaphoreHandle_t timerSemaphore;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

short temp;
int count=0;

short turn = 0;

unsigned long previousMillis = 0;   // Stores last time temperature was published
const long interval = 1;        // Interval at which to publish sensor readings
size_t bytenum = 2;
uint8_t * acquired_data;

File file;

void ARDUINO_ISR_ATTR onTimer(){
  
  portENTER_CRITICAL_ISR(&timerMux);  
  portEXIT_CRITICAL_ISR(&timerMux);
  xSemaphoreGiveFromISR(timerSemaphore, NULL);

  temp = adc1_get_raw((adc1_channel_t)ADC_CHANNEL_0);
  *acquired_data = temp >> 8; acquired_data++; 
  *acquired_data = temp & 0xFF; acquired_data++;
  acquired_data = acquired_data - 2; 
  
  file.write(acquired_data,bytenum);
  count=count+1;

  
  if (count >= amostras){pub = true;  timerAlarmDisable(timer); count = 0; }
  
}

void setup() {
 
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  bool success = SPIFFS.begin();
 
  if (!success) {
    Serial.println("Error mounting the file system");
    return;
  }
    
  file = SPIFFS.open("/file.bin", "w+");

  int bytesWritten;
  
  timerSemaphore = xSemaphoreCreateBinary();
  timer = timerBegin(0, 80, true);
  timerAttachInterrupt(timer, &onTimer, true);
  timerAlarmWrite(timer, 1000, true);
  //timerAlarmEnable(timer);
  adc1_config_width(ADC_WIDTH_BIT_12);
  adc1_config_channel_atten((adc1_channel_t)ADC_CHANNEL_0, ADC_ATTEN_DB_11);

  acquired_data = (uint8_t *) calloc(2 , sizeof(byte));

}

void loop() {
}
```

### Debug Message

```plain
I have none at the moment.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-03-10:
Writing to a file (slow) inside a timer interrupt that happens quickly sounds dangerous.  Try saving your bytes to a buffer.  When your buffer count gets to a certain size, call a function from your main loop() that appends the buffer to a file.  Consider using a Queue or even better a RingBuffer.  https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/FreeRTOS/Queue/Queue.ino.

jpvieir, 2023-03-10:
> Writing to a file (slow) inside a timer interrupt that happens quickly sounds dangerous. Try saving your bytes to a buffer. When your buffer count gets to a certain size, call a function from your [...]

Thank you! I'll check this Queue function. It makes sense to split acquisition and saving into memory. But one thing: i've measured the time it takes to run the file.write function and it takes approximately 10us to store 2 bytes of data, so I think time is not the reason it is crashing.

mrengineer7777, 2023-03-13:
> But one thing: i've measured the time it takes to run the file.write function and it takes approximately 10us to store 2 bytes of data, so I think time is not the reason it is crashing.

10us is way too fast for a flash write.  The file.write() function may be buffering internally or writing on a separate thread.

Things to watch out for:

1. Make sure to close your file.
2. Sometimes accessing variables across processor cores can cause issues.  Is the value for `file` handle the same inside onTimer() as setup()? What core does your timer run on? Typically user code runs on Core 1.
3. Is it safe to call File.Write() from a timer ISR?
4. When flash write occurs, it will likely disrupt your audio sampling.  https://github.com/micropython/micropython/issues/3782

jpvieir, 2023-03-13:
> 
So you mean there's no way of recording, for example, 1 minute of data, without having stalls in the middle of it?

mrengineer7777, 2023-03-13:
It depends. Is your audio polling rate faster than the time it takes to
save to a file? Do some testing. Keep in mind the file might not commit the
write until it's closed.

closed 2023-04-29 by mrengineer7777