espressif/arduino-esp32 #7957: mqtt publish lots of data

savagerex, 2023-03-13:
### Board

ESP32 S3

### Device Description

ADXL355 Sensor - SPI

### Hardware Configuration

NONE

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I want publish Lots of data to cloud.
but it fail.

the code as below

for(p = 0; p < (PACKAGE/OneData); p++)
               {
                   for (n = 0; n < OneData; n++)
                   {
                     TransferBuf[n] = ui8txLogBuf[o];
                     o+=1;

                     if (n == (OneData-1)) {
                         char *DataBuf = (char*)TransferBuf;
                         //client.publish(topic, DataBuf, PACKAGE, false, 0);
                         bool PubData = client.publish(topic, DataBuf, PACKAGE, false, 0);
                         //delay(2000);
                         Serial.println("Publish Data");
                         Serial.println(PubData);
                     }

                     if (p == ((PACKAGE/OneData)-1) && n == (OneData-1)) {
                         o = 0;
                     }
                     
                   }
                   delay(100);
               }

### Sketch

```cpp
[... 387 lines cut ...]
Publish Data
0
Publish Data
0
**************wifi Transfer Time*************260
test9
test5
test8
Publish Data
0
Publish Data
0
**************wifi Transfer Time*************260
test9
test5
test8
Publish Data
0
Publish Data
0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-13:
@savagerex 

I guess that you use this library: https://github.com/256dpi/arduino-mqtt
Its README/NOTES says:
```
The maximum size for packets being published and received is set by default to 128 bytes. To change the buffer 
sizes, you need to use MQTTClient client(256) or MQTTClient client(256, 512) instead of just MQTTClient client at 
the top of your sketch. A single value denotes both the read and write buffer size, two values specify them 
separately. 
Beginning with version 2.6, the message payload is sent directly during publishing. Therefore, the write buffer
 is only needed to encode the packet header and topic, for which the default 128 bytes should be enough. 
However, the receiving of messages is still fully constrained by the read buffer, which may be increased if necessary.
``` 

I see in the code that it tries to send 3,600 bytes:
`#define PACKAGE 3600   //792000 18x4000x11 //72000*3    //180, 1800`

At the same time it defines the maximum size of the packet as 2,000 bytes:
`MQTTClient client(2000);`

I think that it may be the problem.
But this is an external library... This is not an ESP32 Arduino Issue.

I'm not sure if there is any limit for the MQTT package size.
It may depende on a many factors, including the MQTT server, LWIP layer etc.

savagerex, 2023-03-13:
@SuGlider 

I use for loop. 
OneData is 1800 under  MQTTClient client(2000);

for(p = 0; p < (PACKAGE/OneData); p++)
               {
                   for (n = 0; n < OneData; n++)
                   {
                     TransferBuf[n] = ui8txLogBuf[o];
                     o+=1;

                     if (n == (OneData-1)) {
                         char *DataBuf = (char*)TransferBuf;
                         //client.publish(topic, DataBuf, PACKAGE, false, 0);
                         bool PubData = client.publish(topic, DataBuf, PACKAGE, false, 0);
                         //delay(2000);
                         Serial.println("Publish Data");
                         Serial.println(PubData);
                     }

                     if (p == ((PACKAGE/OneData)-1) && n == (OneData-1)) {
                         o = 0;
                     }
                     
                   }
                   delay(100);
               }

bertmelis, 2023-03-13:
I made a MQTT lib that can send as much as you like: [https://](https://github.com/bertmelis/espMqttClient)
Reasons are exactly the limitations other libs have/had.

It's obvious of course that you can not fill up the entire memory, but this is guarded in the lib.

mrdc, 2023-03-19:
> I made a MQTT lib that can send as much as you like: [espMqttClient](https://github.com/bertmelis/espMqttClient) Reasons are exactly the limitations other libs have/had.
> 
> It's obvious of course that you can not fill up the entire memory, but this is guarded in the lib.

Thanks, I'll give it a try. Looks like Async TLS  is a problem :(

bertmelis, 2023-03-19:
You don't need async tcp on ESP32, you have freertos.
The networking is offloaded to separate task so it doesn't interfere with any code you have. The API itself is non blocking so from a users' perspective there's no difference between async and non-async.

(Esp8266 is a different story)

Same goes for any library whether built-in or an external one. If the API is blocking and you don't want that, esp32 gives you the possibility to create a separate task with core and priority to choose.

closed 2023-03-20 by SuGlider