espressif/arduino-esp32 #7961: Rtos with Ring Buffer

savagerex, 2023-03-14:
### Board

ESP32 S3

### Device Description

ADXL355(Sensor) - SPI
W25Q256JV(External flash) - SPI

### Hardware Configuration

NONE

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I use ring buffer on rtos.
but it's fail.

### Sketch

```cpp
#include "SPI.h"
#include "SPIFFS.h"
#include "BoardConfig.h"
#include "Sensor.h"
#include <WiFi.h>
//#include <PubSubClient.h>
#include <MQTT.h>
#include <SerialFlash.h>
#include "freertos/ringbuf.h"

const char* ssid = "ASUS-38";     //SW1-WIFI6   ASUS-38   vivo Y21 
const char* pass = "12345678";   //wireless    12345678   1234567890

const char* mqtt_server = "192.168.50.128";   //broker.emqx.io  192.168.50.128
const char *topic = "DataTopic";
const char *mqtt_username = "emqx";
const char *mqtt_password = "public";

//WiFiClient espClient;
//PubSubClient client(espClient);

WiFiClient net;
MQTTClient client(2000);

#if CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3
[... 507 lines cut ...]
task2 test0
task2 test1
Failed to receive item

task2 test2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42002fb4  PS      : 0x00060630  A0      : 0x00000000  A1      : 0x3fcad5c0  
A2      : 0x3fc9b824  A3      : 0x3fc9b324  A4      : 0x3fc9b490  A5      : 0x3fc9b328  
A6      : 0x3fc9ab4c  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000008  A11     : 0x0000000b  A12     : 0x3fc9b3ec  A13     : 0x3c09f920  
A14     : 0x00000000  A15     : 0x80000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  

Backtrace: 0x42002fb1:0x3fcad5c0

ELF file SHA256: cf699ff301f044fa

Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-14:
@savagerex - 

If the message `Failed to receive item` was printed, `item` is `NULL`.

``` cpp
       if (item != NULL) {
            //Print item
//            for (int i = 0; i < item_size; i++) {
//                //printf("%c", item[i]);
//            }
            //printf("\n");
            //Return Item
            vRingbufferReturnItem(buf_handle, (void *)item);
        } else {
            //Failed to receive item
            Serial.println("Failed to receive item\n");    // <<<<<<<<<======== [item == NULL]
        }
```

Thus, the sketch can't use `item[]`  in the code:

``` cpp
        for (i = 0; i < 9; i++) 
        {
          ui8txLogBuf[j] = ascii_trans((item[i] >> 4));              // <<<<<===== CRASH!  item is NULL
          ui8txLogBuf[j + 1] = ascii_trans((item[i] & 0x0F));     // <<<<<===== CRASH!  item is NULL
          file.write(ui8txLogBuf[j]);
          file.write(ui8txLogBuf[j+1]);
          j += 2;
        }

```

SuGlider, 2023-03-14:
Please, next time, read this:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited

Hint: `EXCVADDR: 0x00000000`

SuGlider, 2023-03-14:
Other observation:

``` cpp
void taskTwo( void * parameter ) {
  disableCore0WDT();
  //uint8_t xQueueData[9] = {0};
  for (;;) {
    Serial.println("task2 test0");
    if (!SPIFFS.begin(true)) {
            Serial.println("An Error has occurred while mounting SPIFFS");
            return;   // <<<<<============= Task is not a function! It can't just return.
        }
```

Tasks shall never return! It can terminate itself by using `vTaskDelete(NULL);` ... 
but it may never start again if the code doesn't create the task again!

savagerex, 2023-03-14:
@SuGlider 

i already send data in taskOne.

why receive the data(item) fail???

SuGlider, 2023-03-14:
It seems that TaskOne didn't send any data to the RingBuffer.

Initial value:
DataReady = false;

SuGlider, 2023-03-14:
Other observation:

Make `DataReady` a volatile variable.
It is changed inside an ISR!

savagerex, 2023-03-14:
OK!! let me try!!

savagerex, 2023-03-14:
i have modified the code.

the code as below

void taskTwo( void * parameter ) {
  disableCore0WDT();
  //uint8_t xQueueData[9] = {0};
  for (;;) {
    Serial.println("task2 test0");
    if (!SPIFFS.begin(true)) {
            Serial.println("An Error has occurred while mounting SPIFFS");
            return;
        }
        
    SPIFFS.format();

    File file = SPIFFS.open("/test.txt", FILE_WRITE);
      
    if (!file) {
      Serial.println("There was an error opening the file for writing");
      return;
    }

    while(1)
    {
      //Serial.println("task2 test1");
      //if (xQueueReceive(xQueue, (void *)&xQueueData, portMAX_DELAY) == pdTRUE)

       size_t item_size;
       //char *item = (char *)xRingbufferReceive(buf_handle, &item_size, pdMS_TO_TICKS(1000));
       uint8_t *item = (uint8_t *)xRingbufferReceive(buf_handle, &item_size, pdMS_TO_TICKS(1000));

       if (item != NULL) {
            //Print item
//            for (int i = 0; i < item_size; i++) {
//                //printf("%c", item[i]);
//            }
            //printf("\n");
            //Return Item
            vRingbufferReturnItem(buf_handle, (void *)item);
        } else {
            //Failed to receive item
            Serial.println("Failed to receive item\n");
        }
           
       
        client.loop();
        //Serial.println("task2 test2");
        StartTime = millis();

        if (FileWriteFlag == 1) {
            FileWriteFlag = 0;  

            File file = SPIFFS.open("/test.txt", FILE_WRITE);
      
            if (!file) {
              Serial.println("There was an error opening the file for writing");
              return;
            }   
        }
        
        for (i = 0; i < 9; i++) 
        {
          ui8txLogBuf[j] = ascii_trans((item[i] >> 4));
          ui8txLogBuf[j + 1] = ascii_trans((item[i] & 0x0F));
          file.write(ui8txLogBuf[j]);
          file.write(ui8txLogBuf[j+1]);
          j += 2;
        }

        if(j == PACKAGE) {
               //Serial.println("test5");
//               ui8txLogBuf[j] = '\n';
//               ui8txLogBuf[j+1] = '\r';
//               file.write(ui8txLogBuf[j]);
//               file.write(ui8txLogBuf[j+1]);
               file.close();
               //Serial.println("test6.....test..time");
               //StartTime = millis();
               File file = SPIFFS.open("/test.txt", FILE_READ);
               //Serial.println("test7");
               while(file.available()) {
               //if(k == (PACKAGE+3))
               if(k == (PACKAGE))
               {
                  break;
               }
               else
               {
                 ui8txLogBuf[k] = (unsigned char)file.read();
               }
               k++; 
               }
               file.close();
               //Serial.println("test8");
               char *DataBuf = (char*)ui8txLogBuf;
               //client.publish(topic, DataBuf, PACKAGE+2, false, 0);
               
               bool PubData = client.publish(topic, DataBuf, PACKAGE, false, 0);
               Serial.println(PubData);

               //Serial.write(ui8txLogBuf, PACKAGE);
               
               //EndTime = millis();
               //Serial.print("**************wifi Transfer Time*************");
               //Serial.println(EndTime-StartTime);
               //Serial.println("test9");
               FileWriteFlag = 1;
               m += 1;
               j=0;
               k=0;
            }

            if (m==440){     //400 40
           EndTime = millis();
           Serial.print("**************wifi Transfer Time*************");
           Serial.println(EndTime-StartTime);
           client.publish(topic, "11 sec Data");
           m=0;
           while(1);
           }
  
      
    }
  }
  vTaskDelete( NULL );
}

=======================debug messsage=========================
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
Connecting...WiFi connected - IP address: 192.168.50.105

connecting...
connected!
DeviceId = AD
task2 test0
Failed to send item
Failed to send item
Failed to send item
Failed to send item
Failed to send item
Failed to send item
Failed to send item
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
E (29712) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (29712) task_wdt:  - IDLE (CPU 0)
E (29712) task_wdt: Tasks currently running:
E (29712) task_wdt: CPU 0: TaskTwo
E (29712) task_wdt: CPU 1: loopTask
E (29712) task_wdt: Aborting.

abort() was called at PC 0x420119f0 on core 0

Backtrace: 0x4037798a:0x3fc95ff0 0x4037cfc5:0x3fc96010 0x4038304d:0x3fc96030 0x420119f0:0x3fc960b0 0x40378dad:0x3fc960d0 0x400559dd:0x3fcae380 |<-CORRUPTED

ELF file SHA256: 364c4cc5d26f026b

Rebooting...

SuGlider, 2023-03-14:
In my opinion, the sketch should not use `DataReady` as the way to signal to the Tasks what to do.
This is a C variable. C compiler will optimize it by assuming that the ISR will never be called and this variable will never change its value, making the conditionals (`if() {}`) constants.

This is what I said that it shall be declared as `volatile`

`volatile bool DataReady = false;`

In order to solve the Watchdog Timer (WDT) issues, the sketch shall use a FreeRTOS resources and methods, instead of C variable flags.
It can use `EventGroupHandle_t` as RTOS flags to signal to the tasks that the Sensor has Data to be processed.

[Free RTOS Event Group](https://www.freertos.org/event-groups-API.html) has some FreeRTOS functions to be used in the Sketch:
[xEventGroupCreate()](https://www.freertos.org/xEventGroupCreate.html)
[xEventGroupSetBitsFromISR()](https://www.freertos.org/xEventGroupSetBitsFromISR.html)
[xEventGroupGetBits()](https://www.freertos.org/xEventGroupGetBits.html)
[xEventGroupSetBits()](https://www.freertos.org/xEventGroupSetBits.html)
[xEventGroupClearBits()](https://www.freertos.org/xEventGroupClearBits.html)
[xEventGroupWaitBits()](https://www.freertos.org/xEventGroupWaitBits.html)

You can get inspiration reading the code from ESP32 RMT. 
The RMT code uses Event Group, Ring Buffers, etc.
It uses Event Group to signal to a Task when the data has been read inside an ISR.
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rmt.c

Event Group related code here:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rmt.c#L90

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rmt.c#L181-L231
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rmt.c#L221-L224

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rmt.c#L483-L519
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rmt.c#L514-L517

SuGlider, 2023-03-14:
Finnaly, I'd suggest reading this documentation about WDT:

**https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html**

savagerex, 2023-03-17:
OK!!  THX

closed 2023-03-20 by SuGlider