espressif/arduino-esp32 #7970: Ring buffer problem

savagerex, 2023-03-17:
### Board

ESP32 S3

### Device Description

ADXL355(Sensor) - SPI

### Hardware Configuration

NONE

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I write the sample code about Ring Buffer.
but ESP32 -S3 always reset.

### Sketch

```cpp
#include "freertos/ringbuf.h"

static char tx_item[] = "test_item";

#define RingBuf 4500

RingbufHandle_t buf_handle;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  buf_handle = xRingbufferCreate(RingBuf, RINGBUF_TYPE_NOSPLIT);    //1028  4112
  if (buf_handle == NULL) {
      printf("Failed to create ring buffer\n");
  }

  UBaseType_t res =  xRingbufferSend(buf_handle, tx_item, sizeof(tx_item), pdMS_TO_TICKS(1000));
    if (res != pdTRUE) {
        printf("Failed to send item\n");
    }

  size_t item_size;
    char *item = (char *)xRingbufferReceiveUpTo(buf_handle, &item_size, pdMS_TO_TICKS(1000), sizeof(tx_item));

    //Check received data
    if (item != NULL) {
        //Print item
        for (int i = 0; i < item_size; i++) {
            //printf("%c", item[i]);
            Serial.println(item[i]);
        }
        Serial.println("\n");
        //Return Item
        vRingbufferReturnItem(buf_handle, (void *)item);
    } else {
        //Failed to receive item
        Serial.println("Failed to receive item\n");
    }

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
⸮ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42020e0a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4

assert failed: xRingbufferReceiveUpTo ringbuf.c:1247 (pxRingbuffer->uxRingbufferFlags & ( ( UBaseType_t ) 2 ))

Backtrace: 0x40377056:0x3fcebde0 0x4037a1b1:0x3fcebe00 0x4037fd95:0x3fcebe20 0x40379f46:0x3fcebf50 0x420015fb:0x3fcebf90 0x42001e36:0x3fcebfd0

ELF file SHA256: e837ff89094d334b
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Adrianotiger, 2023-03-17:
It has nothing to do with the Arduino library.
You can use the exception decoder to get more info.
Or you can just see the assert: the error is in the xRingbufferReceiverUpTo function (again, nothing with the ESP32 or Arduino project).

Read the documentation of the this function:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html#_CPPv422xRingbufferReceiveUpTo15RingbufHandle_tP6size_t10TickType_t6size_t

And you will see why it doesn't work.

savagerex, 2023-03-20:
OK!!!  THX

SuGlider, 2023-03-20:
@savagerex - I have tested your example and it fails.

I changed `xRingbufferReceiveUpTo` by `xRingbufferReceive` and then it worked.
`char *item = (char *)xRingbufferReceive(buf_handle, &item_size, pdMS_TO_TICKS(1000));`

SuGlider, 2023-03-20:
Anyway.... this issue should not happen. I'll run some other test cases and let you know.

SuGlider, 2023-03-20:
`xRingbufferReceiveUpTo()` function should only be called on byte buffers.
Change the type of the RingBuffer created and it will work correctly:

`buf_handle = xRingbufferCreate(RingBuf, RINGBUF_TYPE_BYTEBUF);`

Issue solved.

closed 2023-03-20 by SuGlider