espressif/arduino-esp32 #7976: error with client.connect(server, 3000)

sourav2342, 2023-03-19:
### Board

esp32

### Device Description

none

### Hardware Configuration

none

### Version

latest master (checkout manually)

### IDE Name

arduino

### Operating System

windows

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I was trying to connect to a nodejs server and send a https post request to the server . So, i have used this code to verify connection between esp32 and webserver but  client.connect(server, 3000) returns 0.

### Sketch

```cpp
#include <WiFiClientSecure.h>

const char* ssid = "hjmi";  // Enter SSID here
const char* password = "*******";//.. password

const char*  server = "https://mywebsite.com";  // Server URL

const char* test_root_ca= \........................................lsdkfsdsfs\\ 

WiFiClientSecure client;

void setup() {
  
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

 
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  //client.setInsecure();
  client.setCACert(test_root_ca);
  

  Serial.println("\nStarting connection to server...");
  
   
  int x = client.connect(server, 3000);
  Serial.println(x);
  if (!x)
    Serial.println("Connection failed!");
  else {
    Serial.println("Connected to server!");
    // Make a HTTP request:
    client.println("GET https://mywebsite.com/value");
    client.println("Host: https://mywebsite.com");
    client.println("Connection: close");
    client.println();

    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
 
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
  }
}

void loop() {
 /// empty
}
```

### Debug Message

```plain
0
connection failed!
```

### Other Steps to Reproduce

void loop() {
  WiFiClientSecure *client = new WiFiClientSecure;
  if(client) {
    // set secure client with certificate
    client->setCACert(rootCACertificate);
    //create an HTTPClient instance
    HTTPClient https;

    //Initializing an HTTPS communication using the secure client
    Serial.print("[HTTPS] begin...\n");
    if (https.begin(*client, "https://mywebsite/post_data")){  // HTTPS
      Serial.print("[HTTPS] GET...\n");
      
      https.addHeader("Content-Type", "text/plain");
      int data= 123;
      String str = String(data);
      int httpCode = https.POST(str);
    
      if (httpCode > 0) {
      
       Serial.printf("[HTTPS] GET... code: %d\n", httpCode);
 
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          // print server response payload
          String payload = https.getString();
          Serial.println(payload);
        }
      }
      else {
        Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
      }
      https.end();
    }
  }
  else {
    Serial.printf("[HTTPS] Unable to connect\n");
  }
  Serial.println();
  Serial.println("Waiting 2min before the next round...");
  delay(1200);
}

i have also tried this code but i am not able to communicate with the server . i am new to microcontroller programming can someone explain how to send https request to a server using esp32 and tag a resource or solution this issue.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

sourav2342, 2023-03-21:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/examples/WiFiClientSecure/WiFiClientSecure.ino

Actually this is the code i have modified from and it's printing 'Connection failed!' meaning  client.connect(server, 3000) 
returns 0; 
and just in case i wanted to mention i am using a nodejs server and webapp i have deployed which listens at port 3000 but i am not able to build connection between esp32 and server to post data. 
Sorry if this is not the response u are hoping for as i'm new to github.

sourav2342, 2023-03-22:
@mrengineer7777 thanku for the quick response i have tried it and it was connecting to the server and recieved header.
[image]

and then i have replaced the 
1) test_root_ca with that of my website.
2) changed the port to 3000.// 443
3)chaged the 
    client.println("GET https://mywebsite.com/value");  // from  https://www.howsmyssl.com//a/check HTTP/1.0. '/value' get route 
                                                                                      //returns a string.
    client.println("Host: https://mywebsite.com/"); 

these are the exact changed that i have made.  but it shows "connection failed!". I am thinking the prb is with the port  i have mentioned. but why only 443 is used?

sourav2342, 2023-03-30:
@mrengineer7777  yes, i have checked the certificate it is correct, and i am using wificlientsecure but i am thinking prd could be with these lines 
 client.println("GET  https://iot-temp-monitoring.onrender.com/value HTTP/1.0");
 client.println("Host: iot-temp-monitoring.onrender.com.cdn.cloudflare.net.");

// https://iot-temp-monitoring.onrender.com/value this is my url of my basic web app.

And i am also think about "HTTP/1.0" there are different versions in http.

sourav2342, 2023-03-30:
for testing the get request i am using this url   https://iot-temp-monitoring.onrender.com/value "/value" is the route i don't understand what is the error.

mrengineer7777, 2023-03-30:
Does your code get to `Serial.println("Connected to server!");`?  If so then maybe the issue is with HTTP/1.0.  If not then the issue is with the port or SSL cert.  Turn on verbose debug messages to confirm.

HTTP/1.0 was deprecated at least a year ago in all browsers and likely most servers.  See `HTTPClient::sendHeader(const char * type)` for an example of how HTTPClient generates headers.  Hint: it defaults to `1.1`.

I was recently bit by `HTTP\1.0` in a different project.  The client device only supported 1.0 but the server minimum was 1.1.  After a software update the server started chunking responses to the client, but the 1.0 client had no concept of chunking.  Fixed it by changing the server side to send the entire response in one packet.

sourav2342, 2023-03-31:
@mrengineer7777  No, it doesn't  print  "connected to server "  definitely the prb is with "client.connect(server, port)"  but, i have tried it many times made sure cert is correct but no clue where it is going wrong.

this is my website :  https://iot-temp-monitoring.onrender.com/

code used: 
#include <WiFiClientSecure.h>

const char* ssid = "Rjhgi";  // Enter SSID here
const char* password = "jh2";//.. passwordf wifi network)

const char*  server = "iot-temp-monitoring.onrender.com";  // Server URL https://iot-temp-monitoring.onrender.com/

const char* test_root_ca= \
      "-----BEGIN CERTIFICATE-----\n"
      "MIIDdzCCAl+gAwIBAgIEAgAAuTANBgkqhkiG9w0BAQUFADBaMQswCQYDVQQGEwJJ\n" 
      "RTESMBAGA1UEChMJQmFsdGltb3JlMRMwEQYDVQQLEwpDeWJlclRydXN0MSIwIAYD\n" 
      "VQQDExlCYWx0aW1vcmUgQ3liZXJUcnVzdCBSb290MB4XDTAwMDUxMjE4NDYwMFoX\n" 
      "DTI1MDUxMjIzNTkwMFowWjELMAkGA1UEBhMCSUUxEjAQBgNVBAoTCUJhbHRpbW9y\n" 
      "ZTETMBEGA1UECxMKQ3liZXJUcnVzdDEiMCAGA1UEAxMZQmFsdGltb3JlIEN5YmVy\n" 
      "VHJ1c3QgUm9vdDCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAKMEuyKr\n" 
      "mD1X6CZymrV51Cni4eiVgLGw41uOKymaZN+hXe2wCQVt2yguzmKiYv60iNoS6zjr\n" 
      "IZ3AQSsBUnuId9Mcj8e6uYi1agnnc+gRQKfRzMpijS3ljwumUNKoUMMo6vWrJYeK\n" 
      "mpYcqWe4PwzV9/lSEy/CG9VwcPCPwBLKBsua4dnKM3p31vjsufFoREJIE9LAwqSu\n" 
      "XmD+tqYF/LTdB1kC1FkYmGP1pWPgkAx9XbIGevOF6uvUA65ehD5f/xXtabz5OTZy\n" 
      "dc93Uk3zyZAsuT3lySNTPx8kmCFcB5kpvcY67Oduhjprl3RjM71oGDHweI12v/ye\n" 
      "jl0qhqdNkNwnGjkCAwEAAaNFMEMwHQYDVR0OBBYEFOWdWTCCR1jMrPoIVDaGezq1\n" 
      "BE3wMBIGA1UdEwEB/wQIMAYBAf8CAQMwDgYDVR0PAQH/BAQDAgEGMA0GCSqGSIb3\n" 
      "DQEBBQUAA4IBAQCFDF2O5G9RaEIFoN27TyclhAO992T9Ldcw46QQF+vaKSm2eT92\n" 
      "9hkTI7gQCvlYpNRhcL0EYWoSihfVCr3FvDB81ukMJY2GQE/szKN+OMY3EU/t3Wgx\n" 
      "jkzSswF07r51XgdIGn9w/xZchMB5hbgF/X++ZRGjD8ACtPhSNzkE1akxehi/oCr0\n" 
      "Epn3o0WC4zxe9Z2etciefC7IpJ5OCBRLbf1wbWsaY71k5h+3zvDyny67G7fyUIhz\n" 
      "ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS\n" 
      "R9I4LtD+gdwyah617jzV/OeBHRnDJELqYzmp\n" 
      "-----END CERTIFICATE-----\n";

// You can use x.509 client certificates if you want
//const char* test_client_key = "";   //to verify the client
//const char* test_client_cert = "";  //to verify the client

WiFiClientSecure client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(1000);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  
}

void loop() {
  // do nothing

  client.setCACert(test_root_ca);
  //client.setCertificate(test_client_cert); // for client verification
  //client.setPrivateKey(test_client_key);  // for client verification

  Serial.println("\nStarting connection to server...");
  if (!client.connect(server, 3000))
    Serial.println("Connection failed!");
  else {
    Serial.println("Connected to server!");
    // Make a HTTP request:
    client.println("GET  https://iot-temp-monitoring.onrender.com/value HTTP/1.1");
    client.println("Host: iot-temp-monitoring.onrender.com.cdn.cloudflare.net.");
    client.println("Connection: close");
    client.println();

    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
  }

  delay(2000);
}

output: prints connection failed!  

if u find anything wrong in the code please let me know.

mrengineer7777, 2023-03-31:
@sourav2342 I'm unable to verify your code because it lost the formatting.  https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks.

In your certificate variable `test_root_ca`, you may be missing the ` \ ` at the end of each line.  That is what tells the compiler that it should all be one string.

sourav2342, 2023-04-01:
Sorry for that @mrengineer7777 
yes even though i havn't mentioned ' \ ' in the code i have tried with it still the output is " connection failed!". what should i do?

is there any other way.

```

#include <WiFiClientSecure.h>

const char* ssid = "Rjhgi"; // Enter SSID here
const char* password = "jh2";//.. passwordf wifi network)

const char* server = "iot-temp-monitoring.onrender.com"; // Server URL https://iot-temp-monitoring.onrender.com/

const char* test_root_ca= \
      "-----BEGIN CERTIFICATE-----\n" \
      "MIIDdzCCAl+gAwIBAgIEAgAAuTANBgkqhkiG9w0BAQUFADBaMQswCQYDVQQGEwJJ\n" \ 
      "RTESMBAGA1UEChMJQmFsdGltb3JlMRMwEQYDVQQLEwpDeWJlclRydXN0MSIwIAYD\n" \
      "VQQDExlCYWx0aW1vcmUgQ3liZXJUcnVzdCBSb290MB4XDTAwMDUxMjE4NDYwMFoX\n" \
      "DTI1MDUxMjIzNTkwMFowWjELMAkGA1UEBhMCSUUxEjAQBgNVBAoTCUJhbHRpbW9y\n" \
      "ZTETMBEGA1UECxMKQ3liZXJUcnVzdDEiMCAGA1UEAxMZQmFsdGltb3JlIEN5YmVy\n" \
      "VHJ1c3QgUm9vdDCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAKMEuyKr\n" \
      "mD1X6CZymrV51Cni4eiVgLGw41uOKymaZN+hXe2wCQVt2yguzmKiYv60iNoS6zjr\n" \
      "IZ3AQSsBUnuId9Mcj8e6uYi1agnnc+gRQKfRzMpijS3ljwumUNKoUMMo6vWrJYeK\n" \
      "mpYcqWe4PwzV9/lSEy/CG9VwcPCPwBLKBsua4dnKM3p31vjsufFoREJIE9LAwqSu\n" \
      "XmD+tqYF/LTdB1kC1FkYmGP1pWPgkAx9XbIGevOF6uvUA65ehD5f/xXtabz5OTZy\n" \
      "dc93Uk3zyZAsuT3lySNTPx8kmCFcB5kpvcY67Oduhjprl3RjM71oGDHweI12v/ye\n" \
      "jl0qhqdNkNwnGjkCAwEAAaNFMEMwHQYDVR0OBBYEFOWdWTCCR1jMrPoIVDaGezq1\n" \
      "BE3wMBIGA1UdEwEB/wQIMAYBAf8CAQMwDgYDVR0PAQH/BAQDAgEGMA0GCSqGSIb3\n" \
      "DQEBBQUAA4IBAQCFDF2O5G9RaEIFoN27TyclhAO992T9Ldcw46QQF+vaKSm2eT92\n" \
      "9hkTI7gQCvlYpNRhcL0EYWoSihfVCr3FvDB81ukMJY2GQE/szKN+OMY3EU/t3Wgx\n" \
      "jkzSswF07r51XgdIGn9w/xZchMB5hbgF/X++ZRGjD8ACtPhSNzkE1akxehi/oCr0\n" \
      "Epn3o0WC4zxe9Z2etciefC7IpJ5OCBRLbf1wbWsaY71k5h+3zvDyny67G7fyUIhz\n" \
      "ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS\n" \
      "R9I4LtD+gdwyah617jzV/OeBHRnDJELqYzmp\n" \
      "-----END CERTIFICATE-----\n";

// You can use x.509 client certificates if you want
//const char* test_client_key = ""; //to verify the client
//const char* test_client_cert = ""; //to verify the client

WiFiClientSecure client;

void setup() {
//Initialize serial and wait for port to open:
Serial.begin(115200);
delay(1000);

Serial.print("Attempting to connect to SSID: ");
Serial.println(ssid);
WiFi.begin(ssid, password);

// attempt to connect to Wifi network:
while (WiFi.status() != WL_CONNECTED) {
Serial.print(".");
// wait 1 second for re-trying
delay(1000);
}

Serial.print("Connected to ");
Serial.println(ssid);

}

void loop() {
// do nothing

client.setCACert(test_root_ca);
//client.setCertificate(test_client_cert); // for client verification
//client.setPrivateKey(test_client_key); // for client verification

Serial.println("\nStarting connection to server...");
if (!client.connect(server, 3000))
Serial.println("Connection failed!");
else {
Serial.println("Connected to server!");
// Make a HTTP request:
client.println("GET https://iot-temp-monitoring.onrender.com/value HTTP/1.1");
client.println("Host: iot-temp-monitoring.onrender.com.cdn.cloudflare.net.");
client.println("Connection: close");
client.println();

while (client.connected()) {
  String line = client.readStringUntil('\n');
  if (line == "\r") {
    Serial.println("headers received");
    break;
  }
}
// if there are incoming bytes available
// from the server, read them and print them:
while (client.available()) {
  char c = client.read();
  Serial.write(c);
}

client.stop();
}

delay(2000);
}

```

mrengineer7777, 2023-04-02:
I tried "https://iot-temp-monitoring.onrender.com:3000" in Firefox browser. It takes a long time to load but works. "https://iot-temp-monitoring.onrender.com:3000" does not load.

I downloaded the website's .PEM file.  You are using the "Baltimore CyperTrust Root" cert, which is correct.  Your `test_root_ca` appears to be valid.

Your `server` variable should likely be `"https://iot-temp-monitoring.onrender.com"`.  Try port 443 instead of 3000.

sourav2342, 2023-04-02:
thank you @mrengineer7777  port 443 works! 

but, i have observed something i want to mention.
in most of the article i have read about making https connection using wificlientsecure library.

they have mentioned different ways like 
1) making insecure connection using  client.setInsecure();
2) or by using cert client.setCACert(test_root_ca);
3) by using SHA fingerprint.

the second one is what we have tried, eventhough we are trying to connect server using test_root_ca it is only connecting when i mention client.setInsecure() why is that. am i missing something here.

mrengineer7777, 2023-04-05:
>     1. making insecure connection using  client.setInsecure();
`setInsecure()` erases the cert and connects via HTTP.

>     2. or by using cert client.setCACert(test_root_ca);
I would NOT recommend using setCACert().
`setCACert()` sets a pointer to a buffer you supply.  If the buffer goes away you can crash.  In your example code the cert is a constant, so it should work.
This differs from `loadCACert()` which allocates internal storage and makes a copy of the cert.

`WiFiClientSecure.cpp` has some inconsistencies in pointer and buffer cleanup.  It could use an update.

closed 2023-04-12 by mrengineer7777