espressif/arduino-esp32 #7982: rtos queue

savagerex, 2023-03-22:
### Board

ESP32 S3

### Device Description

ADXL355(Sensor)-spi
W25q256jv(Flash)-spi

### Hardware Configuration

NONE

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I creat two task in free-rtos.

taskOne is capture the sensor(ADXL355) data.
taskTwo is read/write flash(W25q256jv).

i measure capture time in taskOne.  it need spend 1100 mill second , when taskTwo read/write flash

if taskTwo do nothing. the capture time in taskOne spend about 1000 mill second.

i hope when taskTwo read/write flash,  the capture time in taskOne spend about 1000 mill second.

So do you have any idea or suggestion  ????

### Sketch

```cpp
#include "SPI.h"
#include <SerialFlash.h>
#include "SPIFFS.h"
#include "BoardConfig.h"
#include "Sensor.h"
#include <WiFi.h>
//#include <PubSubClient.h>
#include <MQTT.h>
//#include "freertos/ringbuf.h"

const char* ssid = "ASUS-38";     //SW1-WIFI6   ASUS-38   vivo Y21 
const char* pass = "12345678";   //wireless    12345678   1234567890

const char* mqtt_server = "192.168.50.128";   //broker.emqx.io  192.168.50.128
const char *topic = "DataTopic";
[... 644 lines cut ...]
//}

/*
   Read multiple registries
*/
void readMultipleData(int *addresses, int dataSize, int *readedData) {
  digitalWrite(GPIO_OUTPUT_CS, LOW);
  for (int i = 0; i < dataSize; i = i + 1) {
    byte dataToSend = (addresses[i] << 1) | READ_BYTE;
    SPI1.transfer(dataToSend);
    readedData[i] = SPI1.transfer(0x00);
  }
  digitalWrite(GPIO_OUTPUT_CS, HIGH);
}
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-03-22:
This appears to be a request for coding help.  Your approach is reasonable.  I'm not sure what you are asking about timing.

Your example code is far too complex to analyze without running it.  Can you simplify it to just the timing question?

mrengineer7777, 2023-03-22:
I'm more confused now.  Can you explain the basic operation of your program?

savagerex, 2023-03-22:
In taskOne ,   MCU (ESP32 S3) read sensor data.
In taskTwo,   MCU (ESP32 S3) read write flash. 

i measure time
if (n == 4000) 
Serial.print("**************SPI Sensor Time*************");

the time is 1100 mill sec in taskOne.

If 
In taskOne ,   MCU (ESP32 S3) read sensor data.
In taskTwo,   MCU (ESP32 S3) do nothing.
the time is 1000 mill sec in taskOne

savagerex, 2023-03-22:
I hope 
In taskOne , MCU (ESP32 S3) read sensor data.
In taskTwo, MCU (ESP32 S3) read write flash.

the time is 1000 mill sec in taskOne.

savagerex, 2023-03-22:
Because look like, 
In taskTwo, MCU (ESP32 S3) read write flash.
this action, Increase about 100 miil second

closed 2023-04-12 by mrengineer7777