espressif/arduino-esp32 #7983: SD doesn't begins; There is no valid FAT volume

jnegrete2005, 2023-03-23:
### Board

ESP32 WROOM 32D

### Device Description

The chip has an SD module attached to it, connected via SPI, through the default SPI configuration.
It also has an MFRC522 connected to the HSPI.

### Hardware Configuration

Nothing else.

### Version

other

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

Default

### Description

I'm connecting the SD to the ESP32 via SPI with the default SPI pins, but the card won't start. The ESP32 recognizes that it has an SD connected, but when starting the device, it says:
```
[    43][E][sd_diskio.cpp:797] sdcard_mount(): f_mount failed: (13) There is no valid FAT volume
```
I know it recognizes the card because when I take the card out it says:
```
confi[   529][E][sd_diskio.cpp:797] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1030][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
```
I know the cables are not the problem because I have already measured continuity with a multimeter, and I have already checked that the microSD is using FAT32. So I really don't know what's the problem.
Thanks for the help in advance!

### Sketch

```cpp
I am using esp32_arduino_sqlite3_lib

#include <FS.h>
#include <SPI.h>
#include <sqlite3.h>

#include <string>

#include "SD.h"

using namespace std;

void setup() {
  Serial.begin(115200);

  sqlite3 *db;
  char *zErrMsg = 0;

  SPI.begin();
  SD.begin(5, SPI, 40000000, "/sd");

  sqlite3_initialize();
}
```

### Debug Message

```plain
confi[   548][E][sd_diskio.cpp:797] sdcard_mount(): f_mount failed: (13) There is no valid FAT volume
Card Mount Failed
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jnegrete2005, 2023-03-24:
I have it formatted with FAT32. 
I already tried using the builtin example, SD_test, from the ESP32 examples, and it doesn't work either. Here I just get "Card Mount Failed"

jnegrete2005, 2023-03-29:
I had it formatted in another thing than 32 kb, that was the problem. Make sure it is formatted in FAT32 with 32 kb.

closed 2023-03-29 by jnegrete2005

jonathonbrookshire, 2025-01-31:
This can also be caused by your SD card having the GPT partition scheme. I had the same issue until I converted my card to the MBR partition scheme.

me-no-dev, 2025-03-19:
@KenFellows this is not a bug. You card is not correctly formatted with FAT32.