espressif/arduino-esp32 #7989: Is there a way for the ESP32 (S3) to wait until a serial port/monitor has been started?

zztiger123, 2023-03-24:
### Board

ESP32/ESP32-S3

### Device Description

For example, ESP32-S3-WROOM-1-N16R8

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

N/A

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Is there a way for the ESP32 (S3) to wait until a serial port/monitor has been started? I’m using the Arduino IDE. On non-ESP32 boards, I’ve used something like this:

 

while (!Serial) {

   ; // wait for serial port to connect. Needed for native USB port only

  }

Serial.println("*****************************************************");

  Serial.println(" Com port is open");

 

This code doesn’t work with the ESP32. Any other suggestions? I need to print out a message to the serial port, but need to wait until the serial port and terminal/console are open/ready. Since the code currently isn’t able to wait, I lose the first few serial port prints as it takes a few seconds to manually open up the serial monitor program.

### Sketch

```cpp
while (!Serial) {

   ; // wait for serial port to connect. Needed for native USB port only

  }

Serial.println("*****************************************************");

  Serial.println(" Com port is open");
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-03-24:
Hi @zztiger123, just to make sure we are talking about the same, please answer these questions:

Which physical port of the S3 are you using?  The UART or the USB (there is a tag in the board in the top of the USB connector).
Are you using Windows, Linux or MacOS?
Are you using the Arduino IDE Serial Monitor to read the messages?

[image]

SuGlider, 2023-03-24:
I think that you are using an ESP32S3 board using its UART port, which is connected to some USB-UART chip, like the CP2102 or CH340.

In that case, there is no way to know that the USB or even the UART is connected.
This USB-UART chips don't send any signal to the ESP32 to let it know that the USB Host has requested enumeration.
All the USB Device processing is done internally into these chips and they just provide the UART RX/TX lines to the ESP32.

`while(!Serial)` just wait for the `Serial` driver to be installed, but it has nothing to do with the Terminal being open or closed. 

I have no further information about your configuration, therefore I guess you are using the Arduino IDE Serial Monitor. 
In this case, my suggestion is that you can open the Serial Monitor before uploading a new sketch, then the terminal will be suspended by the IDE, then just click on `Clear Output` and wait the firmware to be flashed. 
Right after the ESP32 is reset, you will see all the data sent to the UART. Nothing is lost.

mrengineer7777, 2023-03-24:
@zztiger123 I also used to miss the first few characters when debugging.  In my case it's because VSCode+PIO takes a few seconds to open the monitor terminal (FTDI USB to serial chip) after a flash.  Here's how I handle it:

```
//Power on: delays boot if debugging.
//Other reset: print reason code.
void PrintRebootReason(void) {
    RESET_REASON r = rtc_get_reset_reason(0);
    if(r == POWERON_RESET) {                                                        //When first powered on, or reset via RST line: delay to allow USB to connect
#ifdef APPDEBUG
        #warning "This is a debug build";
        delay(6500);
#endif
    }
    else {                                                                          //Otherwise print debug reason
        Serial.printf("Reset reason %i\r\n", r);
    }
}

setup() {
    Serial.begin(115200);                                                           //Open UART0
    Serial.println("Booting...");                                                   // USB won't be connected yet, so we will only see this on a reboot
    PrintRebootReason();
}
```

closed 2023-03-31 by mrengineer7777