espressif/arduino-esp32 #7995: API to make sure the device is connected to CLOUD

LEO0594, 2023-03-26:
### Related area

ESP32-WROOM-32

### Hardware specification

ESP32-WROOM-32

### Is your feature request related to a problem?

Firstly I am very new to Github. So please bare with me if i am doing something wrong.

When the device is connected to the Wifi and the ISP is up !! everything runs smoothly. Even if the wifi reboots it automatically connects back. But if the ISP is down, we cannot make sure whether the device is connected or not. 

### Describe the solution you'd like

So if there is a way to know the status (its knows as MQTT) i guess. 

Then we can have an LED turn ON if the connection is successful (Router - ON, ISP - UP).

### Describe alternatives you've considered

Currently I am pinging www.google.com to know whether the device has internet connection. But this doest mean the rainmaker cloud is UP or Down. 

By any chance if the device is kicked out of the cloud, we will never know. 

Can someone guide me on "HOW TO MAKE SURE THE DEVICE IS CONNECTED TO CLOUD" ?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2023-03-26:
Pinging Google will prove internet connectivity.  However you are correct.  It will not prove the service you want is available.  The only way to do that is to send a request to that service and make sure you get a response.  As a side effect this check-in will also prove that you have internet and WiFi connectivity.

LEO0594, 2023-03-26:
>The only way to do that is to send a request to that service and make sure you get a response. As a side effect this check-in will also prove that you have internet and WiFi connectivity.

Hello,
First of all thank you for the prompt reply. I really appreciate that.

can you please shed some more light on it? How to achieve it?

mrengineer7777, 2023-03-27:
I have not used MQTT specifically, but the concept is the same: your check-in can be any message that gets a reply.

LEO0594, 2023-03-27:
@mrengineer7777 even I am new to MQTT. And rainmaker as well. 

I am surprised why no one from Rainmaker team is replying to this topic. 

I did search github about this topic, but there was no definite answer. As a walk around i started pinging google. But this is not the solution. 

I am getting output on the serial monitor like - transporter base MQTT error when i disable only the internet(wifi ON).

if some how i can tap into it, we can know the connection status. I tried this before, but I couldn’t get that to happen. 

Can you tag someone from rainmaker team if you know anyone. 

Thanks

mrengineer7777, 2023-03-27:
Haven't used Rainmaker.  I would post an issue on their repository.  Looks like there are multiple:
https://github.com/orgs/espressif/repositories?q=rainmaker&type=all&language=&sort=
My best guess: https://github.com/espressif/esp-rainmaker/issues

LEO0594, 2023-03-27:
@mrengineer7777 

Can you please guide me on how to use this command 

esp_err_t esp_rmaker_mqtt_connect(void) in arduino ide

so that i can make something out of it.

sanketwadekar, 2023-03-28:
You can add event handlers to check whether device is connected to MQTT server.
```
#include <esp_rmaker_common_events.h>

static void mqtt_event_handler(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    if (event_base == RMAKER_COMMON_EVENT) {
        switch (event_id) {
            case RMAKER_MQTT_EVENT_CONNECTED:
                log_i("MQTT Connected.");
                break;
            case RMAKER_MQTT_EVENT_DISCONNECTED:
                log_i("MQTT Disconnected.");
                break;
            default:
                log_i("Unhandled RainMaker Event:");
        }
    }
}

void setup()
{
    // put the below line of code after RMaker.initNode();
    esp_event_handler_register(RMAKER_COMMON_EVENT, ESP_EVENT_ANY_ID, &mqtt_event_handler, NULL);
   
}
```

LEO0594, 2023-03-28:
@sanketwadekar I just tried the code you provided.

MAN !! You are just genius. You just solved my problem just like that !! Hats off !!

Please tell me one more thing (related to this topic)

Lets say due to bad code or something my device will flood the server and the MQTT budget will be 0. If this is the case, will this  `RMAKER_MQTT_EVENT_DISCONNECTED` function get triggered?

sanketwadekar, 2023-03-29:
>
> Lets say due to bad code or something my device will flood the server and the MQTT budget will be 0. If this is the case, will this `RMAKER_MQTT_EVENT_DISCONNECTED` function get triggered?
>
As of now, there is no event raised if the budget is exhausted. We will consider adding this in the future.
>
> It would be really helpful if you can guide me on how to use this function bool esp_rmaker_mqtt_is_budget_available(void)
>
You need to include `esp_rmaker_mqtt.h` in your sketch and call `esp_rmaker_mqtt_is_budget_available()` wherever you want to check if you are out of budget. This function returns true if budget is available, false otherwise.

LEO0594, 2023-03-30:
I am sorry for the late response. 
i will try the given suggestions and update soon. 

Thank you.

closed 2023-04-12 by mrengineer7777