espressif/arduino-esp32 #8000: arduino as component   failed

bihailantian655, 2023-03-28:
### Board

ESP32 SOLO

### Device Description

esp32 solo 

### Hardware Configuration

none

### Version

v2.0.7

### IDE Name

vscode

### Operating System

win11 

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

can not build the  project .

### Sketch

```cpp
vscode idf  start  a  new project 

and did not  Add ESP-IDF Component directory.

just choose template app .  

then   idf.py add-dependency "espressif/arduino-esp32^2.0.7"
 (https://components.espressif.com/components/espressif/arduino-esp32)

idf.py  menuconfig  enable  arduino   and  change the  tick rate to 1000hz

#include " Arduino.h"

void setup()
{

}
void loop()
{
	
}
```

### Debug Message

```plain
[{
[... 158 lines cut ...]
[87/89] Linking C executable bootloader.elf
[88/89] Generating binary image from built executable
esptool.py v3.2-dev
Merged 1 ELF section
Generated E:/COMPILE/solo-idf-arduino/build/bootloader/bootloader.bin    
[89/89] cmd.exe /C "cd /D E:\COMPILE\solo-idf-arduino\build\bootloader\esp-idf\esptool_py && C:\Users\home\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe E:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 E:/COMPILE/solo-idf-arduino/build/bootloader/bootloader.bin"
Bootloader binary size 0x61f0 bytes. 0xe10 bytes (13%) free.
[972/1093] Building C object esp-id...ir/cores/esp32/esp32-hal-uart.c.obj
FAILED: esp-idf/espressif__arduino-esp32/CMakeFiles/__idf_espressif__arduino-esp32.dir/cores/esp32/esp32-hal-uart.c.obj
ccache C:\Users\home\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../managed_components/espressif__arduino-esp32/variants/esp32 -I../managed_components/espressif__arduino-esp32/cores/esp32 -I../managed_components/espressif__arduino-esp32/libraries/ArduinoOTA/src -I../managed_components/espressif__arduino-esp32/libraries/AsyncUDP/src -I../managed_components/espressif__arduino-esp32/libraries/BLE/src -I../managed_components/espressif__arduino-esp32/libraries/BluetoothSerial/src -I../managed_components/espressif__arduino-esp32/libraries/DNSServer/src -I../managed_components/espressif__arduino-esp32/libraries/EEPROM/src -I../managed_components/espressif__arduino-esp32/libraries/ESP32/src -I../managed_components/espressif__arduino-esp32/libraries/ESPmDNS/src -I../managed_components/espressif__arduino-esp32/libraries/Ethernet/src -I../managed_components/espressif__arduino-esp32/libraries/FFat/src -I../managed_components/espressif__arduino-esp32/libraries/FS/src -I../managed_components/espressif__arduino-esp32/libraries/HTTPClient/src -I../managed_components/espressif__arduino-esp32/libraries/HTTPUpdate/src -I../managed_components/espressif__arduino-esp32/libraries/LittleFS/src -I../managed_components/espressif__arduino-esp32/libraries/Insights/src -I../managed_components/espressif__arduino-esp32/libraries/I2S/src -I../managed_components/espressif__arduino-esp32/libraries/NetBIOS/src -I../managed_components/espressif__arduino-esp32/libraries/Preferences/src -I../managed_components/espressif__arduino-esp32/libraries/RainMaker/src -I../managed_components/espressif__arduino-esp32/libraries/SD_MMC/src -I../managed_components/espressif__arduino-esp32/libraries/SD/src -I../managed_components/espressif__arduino-esp32/libraries/SimpleBLE/src -I../managed_components/espressif__arduino-esp32/libraries/SPIFFS/src -I../managed_components/espressif__arduino-esp32/libraries/SPI/src -I../managed_components/espressif__arduino-esp32/libraries/Ticker/src -I../managed_components/espressif__arduino-esp32/libraries/Update/src -I../managed_components/espressif__arduino-esp32/libraries/USB/src -I../managed_components/espressif__arduino-esp32/libraries/WebServer/src -I../managed_components/espressif__arduino-esp32/libraries/WiFiClientSecure/src -I../managed_components/espressif__arduino-esp32/libraries/WiFi/src -I../managed_components/espressif__arduino-esp32/libraries/WiFiProv/src -I../managed_components/espressif__arduino-esp32/libraries/Wire/src -I../managed_components/espressif__arduino-esp32/cores/esp32/libb64 -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/newlib/platform_include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/freertos/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/freertos/include/esp_additions/freertos -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/freertos/port/xtensa/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/freertos/include/esp_additions -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_hw_support/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_hw_support/include/soc -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_hw_support/include/soc/esp32 -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_hw_support/port/esp32/. -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/heap/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/log/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/include/apps -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/include/apps/sntp -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/lwip/src/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/port/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/port/esp32/include/arch -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/soc/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/soc/esp32/. -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/soc/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/hal/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/hal/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/hal/platform_port/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_rom/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_rom/include/esp32 -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_rom/esp32 -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_common/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_system/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_system/port/soc -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_system/port/public_compat -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/xtensa/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/xtensa/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/driver/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/driver/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_pm/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_ringbuf/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/efuse/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/efuse/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/vfs/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_wifi/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_event/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_netif/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_eth/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/tcpip_adapter/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_phy/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_phy/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_ipc/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/app_trace/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_timer/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/spi_flash/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/mbedtls/port/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/mbedtls/mbedtls/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/mbedtls/esp_crt_bundle/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/mdns/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/console -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_adc_cal/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/wifi_provisioning/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/protocomm/include/common -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/protocomm/include/security -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/protocomm/include/transports  [... cut ...]
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   uart_get_buffered_data_len
cc1.exe: some warnings being treated as errors
[985/1093] Building CXX object esp-...ir/cores/esp32/StreamString.cpp.obj
ninja: build stopped: subcommand failed.

 *  终端进程“C:\Windows\System32\cmd.exe /d /c ninja ”已终止，退出代码: 1。
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

bihailantian655, 2023-03-28:
already  enable idf component  manger in vscode  settings

bihailantian655, 2023-03-28:
[image]
![T UPRQB 2HY91GA@`$}F% M](https://user
[image]

[image]
-images.githubusercontent.com/35267721/228251710-24929be7-853c-40ca-bed4-eb0f224c0ad8.png)

Jason2866, 2023-03-28:
This is NOT an Arduino issue. The Platformio workflow you try to use is not correct.

bihailantian655, 2023-03-28:
I   did  not  use  platform  io . IT  IS IDF plugins in vscode

bihailantian655, 2023-03-28:
[image]
[image]

IS  IT  THE  GLOBAL COMPONENT  FOLDER?

bihailantian655, 2023-03-29:
[image]

Jason2866, 2023-03-29:
This question should be asked at https://github.com/espressif/vscode-esp-idf-extension
Since Arduino as Component compiles without any issues (seen here https://github.com/espressif/esp32-arduino-lib-builder) it is not related to Arduino or IDF. I use the Arduino Lib Builder to compile an Arduino Solo1 framework (https://github.com/jason2866/esp32-arduino-lib-builder/tree/solo1)

bihailantian655, 2023-03-29:
Ok, I see, thank you for your attention

SuGlider, 2023-03-30:
@bihailantian655 -

```
FAILED: esp-idf/espressif__arduino-esp32/CMakeFiles/__idf_espressif__arduino-esp32.dir/cores/esp32/esp32-hal-uart.c.obj
ccache C:\Users\home\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../managed_components/espressif__arduino-esp32/variants/esp32 -I../managed_components/espressif__arduino-esp32/cores/esp32 -I../managed_components/espressif__arduino-esp32/libraries/ArduinoOTA/src -I../managed_components/espressif__arduino-esp32/libraries/AsyncUDP/src -I../managed_components/espressif__arduino-esp32/libraries/BLE/src -I../managed_components/espressif__arduino-esp32/libraries/BluetoothSerial/src -I../managed_components/espressif__arduino-esp32/libraries/DNSServer/src -I../managed_components/espressif__arduino-esp32/libraries/EEPROM/src -I../managed_components/espressif__arduino-esp32/libraries/ESP32/src -I../managed_components/espressif__arduino-esp32/libraries/ESPmDNS/src -I../managed_components/espressif__arduino-esp32/libraries/Ethernet/src -I../managed_components/espressif__arduino-esp32/libraries/FFat/src -I../managed_components/espressif__arduino-esp32/libraries/FS/src -I../managed_components/espressif__arduino-esp32/libraries/HTTPClient/src -I../managed_components/espressif__arduino-esp32/libraries/HTTPUpdate/src -I../managed_components/espressif__arduino-esp32/libraries/LittleFS/src -I../managed_components/espressif__arduino-esp32/libraries/Insights/src -I../managed_components/espressif__arduino-esp32/libraries/I2S/src -I../managed_components/espressif__arduino-esp32/libraries/NetBIOS/src -I../managed_components/espressif__arduino-esp32/libraries/Preferences/src -I../managed_components/espressif__arduino-esp32/libraries/RainMaker/src -I../managed_components/espressif__arduino-esp32/libraries/SD_MMC/src -I../managed_components/espressif__arduino-esp32/libraries/SD/src -I../managed_components/espressif__arduino-esp32/libraries/SimpleBLE/src -I../managed_components/espressif__arduino-esp32/libraries/SPIFFS/src -I../managed_components/espressif__arduino-esp32/libraries/SPI/src -I../managed_components/espressif__arduino-esp32/libraries/Ticker/src -I../managed_components/espressif__arduino-esp32/libraries/Update/src -I../managed_components/espressif__arduino-esp32/libraries/USB/src -I../managed_components/espressif__arduino-esp32/libraries/WebServer/src -I../managed_components/espressif__arduino-esp32/libraries/WiFiClientSecure/src -I../managed_components/espressif__arduino-esp32/libraries/WiFi/src -I../managed_components/espressif__arduino-esp32/libraries/WiFiProv/src -I../managed_components/espressif__arduino-esp32/libraries/Wire/src -I../managed_components/espressif__arduino-esp32/cores/esp32/libb64 -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/newlib/platform_include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/freertos/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/freertos/include/esp_additions/freertos -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/freertos/port/xtensa/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/freertos/include/esp_additions -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_hw_support/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_hw_support/include/soc -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_hw_support/include/soc/esp32 -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_hw_support/port/esp32/. -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/heap/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/log/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/include/apps -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/include/apps/sntp -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/lwip/src/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/port/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/lwip/port/esp32/include/arch -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/soc/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/soc/esp32/. -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/soc/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/hal/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/hal/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/hal/platform_port/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_rom/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_rom/include/esp32 -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_rom/esp32 -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_common/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_system/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_system/port/soc -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_system/port/public_compat -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/xtensa/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/xtensa/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/driver/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/driver/esp32/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_pm/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/esp_ringbuf/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/components/efuse/include -IE:/COMPILE/Othert/IDF4.4/esp-idf-v4.4/ [... cut ...]
cc1.exe: some warnings being treated as errors
```

The error message with `uart_get_tx_buffer_free_size()` suggests that the IDF 4.4 version that your project is using is not updated.

The function `uart_get_tx_buffer_free_size()` was added in July of 2022 to IDF 4.4 and it was added to Arduino in the PR #6998
IDF PR that added it: https://github.com/espressif/esp-idf/commit/f4d33d27395e1ad9cc4a1979573cad3d9395c9a6

To solve it you must pull the lastest version of the IDF 4.4 branch to your local IDF folder.

SuGlider, 2023-04-13:
@bihailantian655 - Please let me know if the issue is solved.

bihailantian655, 2023-04-13:
the problem has been solved

closed 2023-04-18 by SuGlider