espressif/arduino-esp32 #8001: ESP32 Flash Write Freeze (SPIFFS and EEPROM)

savejeff, 2023-03-28:
### Board

all ESP32 based boards

### Device Description

bare board. no hardware attached

### Hardware Configuration

bare board. no hardware attached

### Version

v2.0.7

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

-

### PSRAM enabled

no

### Upload speed

921600

### Description

Okay this is the weirdest problem i have ever had. I'm at this for 3 straight days now

I have completely unrelated code that causes the Flash writes to freeze.
The resulting crash is caused by the WDT0 not the task WDT (TG0WDT_SYS_RESET) meaning its a low-level problem

I have reduced the demo code to the smallest possible version and weird things start to happen
- when i remove random string variables it works again
- if i move code from one .cpp file to another it works again
- if i rename the second source file it works again
- etc etc

I have already worked with @mrengineer7777 and he was able to reproduce the problem.

this looks to me as something wrong deep down in the espressif code (or how platformio creates the GCC commands)
Can somebody working on the core code look at this? it seems to me its an indication that something fundamental is wrong

### Sketch

```cpp
i have created a public repo this the demo code. it's just 3 files 

git@github.com:savejeff/ESP32_Flash_Freeze_Demo.git
```

### Debug Message

```plain
SPIFFS init
SPIFFS init done
>>>> WRITE 0                 <-- Freezes here in EEPROM.commit()
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-03-29:
@savejeff I found your crash.  It's caused by `jSPI_Hard SPI0(&SPI, 18, 19, 23);` in your library SERCOM_Module.cpp.

mrengineer7777, 2023-03-29:
I'm not sure what else to tell you.  You seem convinced the issue is not with your code.  Many developers are using the arduino-esp32 framework without issues, including TASMOTA.

valeros, 2023-03-30:
Hi @savejeff ! What happens if you add `lib_archive = no` to your environment configuration?

savejeff, 2023-03-31:
Hi @valeros good to see you!

the `lib_archive = no` seems to fix the issue. Removing it brings back the issue. Good first sign! 
Do you have an explanation for why this fixes the issue? what does `lib_archive = no` do ?
After all the weird ways if found to fix this by making seemingly unrelated changes (like for example renaming just a source file) I'm cautious to jump to a solution that has no explanation why it works. 

I'll also test this on the original big project if found this issue originally

valeros, 2023-03-31:
> what does lib_archive = no do?

The reason why it freezes requires a deeper dive, but my hypothesis is that some `weak` symbols (in either the framework or libraries) could be the culprit. By default, PlatformIO instructs the linker to link libraries as static archives, so the first encountered symbol is used even when it has the `weak` annotation (that's why the order of libraries is important). The `lib_archive` option instructs PlatformIO build system to link extra libraries as object files instead of static archives (similarly to the Arduino IDE) so that the linker will scan through all symbols and will use the proper `strong` symbol.

valeros, 2023-03-31:
> just a side question: does the lib_archive option influence weak and strong declarations when I define a function weak in a /lib/... library and declare a strong definition in /src/...

The situation is quite similar, but I guess it should work even without `lib_archive` mainly because the files from the `src` folder are always linked as objects, so a strong symbol from `src` should take precedence.

savejeff, 2023-03-31:
do you by chance know the correct syntax for weak declarations? as far as I know, there is no official standard before C++17. 
I tried every variant I found online but non worked.

valeros, 2023-03-31:
> do you by chance know the correct syntax for weak declarations? as far as I know, there is no official standard before C++17.

I think we're getting a little off topic here, I guess you'd better consult GCC docs for this functionality.

savejeff, 2023-03-31:
yes ok. just thought you might know it from the top of you head. 

About the original problem, feel free to contact me for testing possible solutions or further investigations

savejeff, 2023-04-02:
@valeros what advantages does `lib_archive = yes` have ? from what i understand it would be better to have this off by default

savejeff, 2023-04-04:
okay i understand. I also found that the RP2040 core using earlephilhower variant does not compile when lib_archive = no. some executable like ExecuteProcess.exe is not found.

Jason2866, 2023-04-19:
Closing since not related to Arduino ESP32

closed 2023-04-19 by Jason2866