espressif/arduino-esp32 #8005: esp32-s3 cannot find F_CPU for FastLed

coskunyildirimcosel, 2023-03-29:
### Board

ESP32-S3

### Device Description

 esp32-s3 cannot find F_CPU for FastLed

### Hardware Configuration

NO

### Version

v2.0.7

### IDE Name

ESP-IDF

### Operating System

Windows10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

 esp32-s3 cannot find F_CPU for FastLed

### Sketch

```cpp
#ifndef HAL_ESP32_HAL_H_
#define HAL_ESP32_HAL_H_

#include <stdint.h>
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <inttypes.h>
#include <string.h>
#include <math.h>
#include "sdkconfig.h"
#include "esp_system.h"
#include "esp_sleep.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "freertos/event_groups.h"

#ifdef __cplusplus
extern "C" {
#endif

#ifndef F_CPU
#if CONFIG_IDF_TARGET_ESP32 // ESP32/PICO-D4
#define F_CPU (CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ * 1000000U)
#elif CONFIG_IDF_TARGET_ESP32S2
#define F_CPU (CONFIG_ESP32S2_DEFAULT_CPU_FREQ_MHZ * 1000000U)

//not esp32-s3
```

### Debug Message

```plain
esp32-s3 cannot find F_CPU for FastLed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-03-29:
What is FastLed?  In this repository `F_CPU` should be defined in boards.txt.  However your code looks like IDF code.  Did you mean to post here: https://github.com/espressif/esp-idf/issues ?

Jason2866, 2023-03-29:
There is a open PR to fix this https://github.com/espressif/arduino-esp32/pull/7913

SuGlider, 2023-03-30:
Just modified the PR #7913 to work with and the ESP Arduino suported SoCs.

mrengineer7777, 2023-03-31:
@coskunyildirimcosel Fix was merged and will be available in next release.  Is it ok to close this issue?

coskunyildirimcosel, 2023-03-31:
Ok thank you

PR #7913 (2023-03-31): Fix the F_CPU frequency definition for the ESP32-S3 in esp32-hal.h
## Description of Change
Hello, I was using the FastLED library and it was complaining about F_CPU not being defined. So, I just noticed that it is not defined for the ESP32-S3 module. So I made this change in the header file and it compiled. Therefore I wanted to propose this change to the HAL library to improve compatibility. Thank you for your time.

## Tests scenarios
I have tested this with the ESP32-S3-WROOM-1-N16R2 module for the FastLED library. Haven't done any extensive testing since it is a trivial change (I hope).

 [... cut]
closed 2023-03-31 by coskunyildirimcosel