espressif/arduino-esp32 #8019: Why is there a large drop in performance between 2.0.5 and 2.0.6?

technoblogy, 2023-04-01:
### Board

ESP32

### Device Description

Adafruit ESP32 Feather

### Hardware Configuration

No.

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

macOS 10.13.6

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I've compared benchmarks for three application programs running on the ESP32 version of uLisp, my Lisp interpreter for microcontrollers, and there's a big drop in performance between arduino-esp32 version 2.0.5 and 2.0.6 (the figures are execution times, so shorter times are better):

| Version        | 2.0.5 | 2.0.6 | 2.0.7 |
| ----------- | ----- | ----- | ----- |
| Tak                | 8.1 s  | 14.2 s | 13.8 s|
| Q2                | 21.2 s | 38.1 s | 37.2 s|
| FFT              | 241 ms | 416 ms | 408 ms |

As you can see from the figures, 2.0.7 is a slight improvement, but it's still about 70% slower than 2.0.5.

Is there a reason for these differences, and is there anything I can do about it?

### Sketch

```cpp
For details of the benchmarks see [Benchmarks](http://www.ulisp.com/show?1EO1).
```

### Debug Message

```plain
There is no error.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-01:
That's interesting. Not sure what your benchmark is doing, but I'm guessing it's processor bound work. The differences are probably due to updates in the underlying IDF.

mrengineer7777, 2023-04-01:
It's also possible the compiler optimization flags have changed.

technoblogy, 2023-04-01:
Is there some way I can check that - I assume it's in the platform.txt file?

mrengineer7777, 2023-04-01:
The IDF version is noted in release notes. It's also a #define and there's
a function call to retrieve the string version. I believe compile flags are
set in arduino-lib-builder.

technoblogy, 2023-04-12:
Sorry, why is this closed? Is there no plan to investigate what's causing it?

technoblogy, 2023-04-12:
OK, I'll try and produce a simple code example.

technoblogy, 2023-04-12:
I've tried a pure C implementation of the first benchmark I used for the above comparison:
````text
int tak (int x, int y, int z) {
  if (!(y < x)) return z;
  else return tak(tak(x-1,y,z), tak(y-1, z,x), tak(z-1,x,y));
} 
````
The calculation of tak(24, 16, 8) took exactly the same time (170ms) on 2.0.5, 2.0.6, and 2.0.7.

I'll have to think about what my Lisp interpreter might be doing that's affected by the different versions of the ESP32 core.

mrengineer7777, 2023-04-12:
Sounds good.  FYI I'm not an Espressif employee.  I volunteered to help with managing their issues because they are way behind on issues and PRs.  I have even fixed some issues by submitting PRs.

dragoncoder047, 2023-04-12:
> I'll have to think about what my Lisp interpreter might be doing that's affected by the different versions of the ESP32 core.

My guess from a user standpoint is that the arduino core runs freeRTOS background tasks and the uLisp evaluator calls `yield()` somehow (although I can't figure out where) allowing other non-uLisp stuff to run in the middle of uLisp execution. The C example clearly doesn't call `yield()` so no tasks switch.

@technoblogy if you want to get rid of this issue, I would maybe add a flag `NOYIELD`, and turn it on when in `(time)`, and then only ever `yield()` if the `NOYIELD` flag is cleared -- similar to calling `testescape()` and the `NOESC` flag.

mrengineer7777, 2023-05-18:
@SuGlider How hard would it be to fix this issue?

closed 2025-01-16 by Parsaabasi