espressif/arduino-esp32 #8020: Arduino-esp32 usage with ESP-IDF version 5.0.0

5abina, 2023-04-02:
### Board

ESP32

### Device Description

ESP32-PIVO-V3

### Hardware Configuration

ESP32

### Version

v2.0.7

### IDE Name

VS Code

### Operating System

Windows 11

### Flash frequency

1843200

### PSRAM enabled

yes

### Upload speed

1843200

### Description

Can't open menucofing and use arduino ESP32.
Is there a way to port arduino ESP32 with ESP-IDF 5.0.0 now?

### Sketch

```cpp
idf.py menuconfig
```

### Debug Message

```plain
Arduino-esp32 can be used with ESP-IDF versions between 4.4.0 and 4.4.99, but a newer version is detected: 5.0.0.
```

### Other Steps to Reproduce

Follow the procedure on https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-04-03:
@thedeardanny - We are currently working on the Arduino Core 3.0.0 that will use IDF 5.1
It is still under development and it can be found in the `esp-idf-v5.1-libs` branch of this repository.
https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs

nilutpolkashyap, 2023-04-18:
Hi @SuGlider , I installed ESP-IDF VS code extension verson 1.6.1 which came with idf.py version 5.0.1 

> esp-idf-v5.1-libs

I tried using the arduino-esp32 package branch **esp-idf-v5.1-libs** as mentioned by you above but I still got the error message 

`Arduino-esp32 can be used with ESP-IDF versions between 5.1.0 and 5.1.99, but an older version is detected: 5.0.1.`

Can you please help in this?

SuGlider, 2023-04-18:
I'm not an expert with PIO.
I see that you shall replace the IDF5.0 by the 5.1.

@Jason2866 - do you have any good hint?

SuGlider, 2023-04-18:
I'm not an expert with PIO.
I see that you shall replace the IDF5.0 by the 5.1.

Maybe, if possible, use `git pull` within IDF folder?!

@Jason2866 - do you have any good hint?

wellknowndanny, 2023-05-13:
Hello @SuGlider Thank you for your reply. I have chanched the branch for esp-idf-v5.1-libs

but, there you have error while compilation

CMake Error at C:/Users/Daniel/esp/esp-idf/tools/cmake/build.cmake:265 (message):
  Failed to resolve component 'mdns'.

Jason2866, 2023-05-13:
You have to install IDF component `mdns` in your setup. It is NOT a part anymore in IDF.

wellknowndanny, 2023-05-14:
Hey @Jason2866 I'm not using Platformio. I'm working on ESP-IDF in VS Code.

Jason2866, 2023-05-14:
So just add mdns as component and you are ready to go.

wellknowndanny, 2023-05-14:
Ok. I've do that. I had to add also espressif/kconfgen module. I have no idea what is this but I have ERROR: Cannot find versions of "espressif/kconfgen" with version satisfying
  "*" for the current target "esp32"

wellknowndanny, 2023-05-15:
Ok. I'm still figthing with this. I have and extension template in my VS Code 
arduino-as-component

I have an updated esp-idf to release version 5.1 and have placed arduino in my project 
\components\arduino-esp32

currently I'm in esp-idf-v5.1-libs

I've had some issues but I found some tutorial on youtube and i have changed this 

if(NOT CONFIG_FREERTOS_HZ EQUAL 1000 AND NOT "$ENV{ARDUINO_SKIP_TICK_CHECK}")
    # See delay() in cores/esp32/esp32-hal-misc.c.
    message(FATAL_ERROR "esp32-arduino requires CONFIG_FREERTOS_HZ=1000 "
                        "(currently ${CONFIG_FREERTOS_HZ})")
endif()

to this:

if(NOT CONFIG_FREERTOS_HZ EQUAL 100 AND NOT "$ENV{ARDUINO_SKIP_TICK_CHECK}")
    # See delay() in cores/esp32/esp32-hal-misc.c.
    message(FATAL_ERROR "esp32-arduino requires CONFIG_FREERTOS_HZ=100 "
                        "(currently ${CONFIG_FREERTOS_HZ})")
endif()

in CMakeLists.txt

now I can open sdkconfig but i still have errors during compilation:

/components/arduino-esp32/cores/esp32/esp32-hal-cpu.c:57:8: error: unknown type name 'xSemaphoreHandle'
   57 | static xSemaphoreHandle apb_change_lock = NULL;
      |        ^~~~~~~~~~~~~~~~

wellknowndanny, 2023-05-15:
Ok. I'm answering to myself. 
There is an option CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY which must be set.

And that's it. It works ;)

roleroz, 2023-08-07:
Any updates on Arduino Core 3.0? Is this the right place to follow along on the updates?

VojtechBartoska, 2023-08-18:
Closing this as duplicate, the main issue for tracking IDF 5.1 updates is this one: https://github.com/espressif/arduino-esp32/issues/7852

closed 2023-08-18 by VojtechBartoska

my-dudhwala, 2023-09-13:
> ### Board
> ESP32
> 
> ### Device Description
> ESP32-PIVO-V3
> 
> ### Hardware Configuration
> ESP32
> 
> ### Version
> v2.0.7
> 
> ### IDE Name
> VS Code
> 
> ### Operating System
> Windows 11
> 
> ### Flash frequency
> 1843200
> 
> ### PSRAM enabled
> yes
> 
> ### Upload speed
> 1843200
> 
> ### Description
> Can't open menucofing and use arduino ESP32. Is there a way to port arduino ESP32 with ESP-IDF 5.0.0 now?
> 
> ### Sketch
> ```c++
> idf.py menuconfig
> ```
> 
> ### Debug Message
> ```
> Arduino-esp32 can be used with ESP-IDF versions between 4.4.0 and 4.4.99, but a newer version is detected: 5.0.0.
> ```
> 
> ### Other Steps to Reproduce
> Follow the procedure on https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html
> 
> ### I have checked existing issues, online documentation and the Troubleshooting Guide
> * [x]  I confirm I have checked existing issues, online documentation and Troubleshooting guide.

I installed IDF v4.4.5 and it worked

Jason2866, 2023-09-13:
No. Since there is no Arduino version based on IDF5.x