espressif/arduino-esp32 #8022: Multiple I2C nodes not working correctly

Noor-H, 2023-04-03:
### Board

ESP32 Dev Module, ATMEGA 328p

### Device Description

ATMEGA328p ICs

### Hardware Configuration

I am working on sometihing with 1 ESP32 as master and 15 ATMEGA 328p as nodes, on an I2C bus. The data transmission is two way. 
**I2C connection on Atmega:**
 SDA=A4 SCL=A5
**ESP32:**
 SDA=21 SCL=22
Pull up resistors of 3.3k ohm between SCL, SDA and 3.3 V of ESP32
Pull up resistors of 5.6k ohm between SCL, SDA and 5 V of ATMEGA328p
 A bidirectional logic level converter between ESP32 and ATMEGA 328p's SDA and SCL

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

**Project Description:**

The project I am working on is to monitor and control power in a certain premesis. It has 1 master as ESP32 and 15 Slaves as ATMEGA 328p.
Each slave has 2 relays that will control the appliances (like light, fan) based on commands recieved from master on I2C. This has been tested several times and works fine.
In addition to that, each slave also has 2 [PZEMs](https://www.amazon.com/PZEM-004T-Current-Voltage-Multimeter-Transformer/dp/B074QGWMKH) that measure voltage and current of the same electricity lines which are controlled by Relays and send the values back to the master on I2C.

Master takes these values from every slave after 1 minute (this interval can be increased) one by one from each slave, over I2C and publishes those to MQTT topics. The values include Voltage, Current, Frequency and Power factor. One slave sends these values at once (with a 10 ms delay in every Wire.write) after a single request from Master to that slave.

At the same time, if there is any msg recieved from MQTT topics (callbacks), master passes it to the particular slave over I2C and that particular relay is operated.

The slaves are close together and I2C wire length is not that long.

The data transmission is two way. So far I have tested with 4 nodes, everything was working fine, but suddenly Master stopped recognizing the nodes anymore, as there was something wrong with the I2C bus. Since then I have encountered multiple errors like:

can't init sda=0 scl=0 (sometimes it's different combinations sda=1, scl=0 and so on)
Whole system working fine for an hour or so and suddenly I2C communication stops, and is revived when I disconnect and then reconnect the power, basically turn it off and on again.
I have removed the whole code and gone back to the super basic scanner code but still same problem.
I have read multiple forums with same problems but none has any satisfactory / reliable answers.

**Things to know:**

ATMEGA 328p ICs are soldered with the crystals and capacitors just like arduino UNO and work fine otherwise
Each ATMEGA has 2 5V realys and 2 PZEMs attached to it. 
I have a bidirectional logic level converter between master and nodes for power, ground, SDA and SCL lines

There is a 3.3 k ohm resistor between ESP32's 3.3 V and SDA line and one between Vin and SCL line
Pull up resistors of 5.6k ohm between SCL, SDA and 5 V of ATMEGA328p
The power is being provided through a DC supply, from master to all the nodes

The code running in the master is I2C scanner code

The nodes have a simple code in which begin the I2C communication, assign address to the node

There is a 5 second delay in setup() before initializing I2C communication on both the master and nodes side so that all the devices are up before I2C bus begins and hence no device is pulling the line low

The master scans for the nodes, sometimes yields correct results, sometimes while giving correct results for a while switches to No devices found

Restarting everything helps sometimes but not everytime

Been stuck at it for a long time, Any sort of help would be extremely appreciated.
Thanks

### Sketch

```cpp
//==========================Node===================================:
void setup()
{
  Serial.begin(115200);
  delay(5000);

  pinMode(pz1, OUTPUT); //to control PZEM 1
  pinMode(pz2, OUTPUT); //to Control PZEM 2
  pinMode(pin7, OUTPUT); //to operate Relay 1
  pinMode(pin8, OUTPUT); //to operate Relay 2
  pinMode(pin6, OUTPUT); //led to indicate I2C transmit status
  Wire.begin(Node);             // Begins I2C communication with Node Address at pin (A4,A5)
  Wire.onRequest(requestEvent); // Function call when Master request value from Node
  Wire.onReceive(receiveEvent); // Function call when Node receives value from master
}

void loop()
{}

//===========================Master======================================:

#include <Wire.h>
void setup() {
  Wire.begin();
  Serial.begin(115200);
  delay(5000);
  Serial.println("\nI2C Scanner");
}
 void loop()
{
byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
      nDevices++;
    }
    else if (error==4) {
      Serial.print("Unknow error at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  }
  else {
    Serial.println("done\n");
  }
  delay(5000);
}
```

### Debug Message

```plain
Connected to MQTT as esp32dp_gw[E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
Scanning...
No I2C devices found

Total number of nodes:0
```

### Other Steps to Reproduce

[This ](https://megaeshop.pk/4-channel-i2c-logic-level-converter-module-5v-to-3-3v.html)is the logic level converter I am using: 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-04-04:
After `Bus Invalid State` it is normal not to have Wire working. Have you tried using pins other than 0 and 1? Since what version did you start having issues?

Noor-H, 2023-04-05:
> After `Bus Invalid State` it is normal not to have Wire working. Have you tried using pins other than 0 and 1? Since what version did you start having issues?

Yes, it's normal for it to not work. Once in a while it works after it automatically re initializes the bus but that's rare.
I am not using pins 0 and 1 for anything. The debugg msg says that sda is HIGH and scl is LOW hence bus can't be initialized.

VojtechBartoska, 2023-08-18:
Closing this as expired, if needed you can reopen it.

closed 2023-08-18 by VojtechBartoska