espressif/arduino-esp32 #8023: ESP32-S3 can not use Bluetooth Classic

limengdu, 2023-04-03:
### Board

ESP32S3-Dev-Module

### Device Description

ESP32S3-Dev-Module

### Hardware Configuration

The ESP32S3-Dev-Module connects directly to the computer and compiles the sample program via Arduino.

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

240MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am using the sample program that comes with the board: BletoothSerial -> bt_classic_device_discovery. but when I compile the program, I find that I get the error "#error Serial Bluetooth not available or not enabled. it is only It is only available for the ESP32 chip.", which seems to be caused by the esp32s3 not defining CONFIG_BT_SPP_ENABLED.

### Sketch

```cpp
#include <BluetoothSerial.h>

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif

BluetoothSerial SerialBT;

#define BT_DISCOVER_TIME	10000

static bool btScanAsync = true;
static bool btScanSync = true;

void btAdvertisedDeviceFound(BTAdvertisedDevice* pDevice) {
	Serial.printf("Found a device asynchronously: %s\n", pDevice->toString().c_str());
}

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");

  if (btScanAsync) {
    Serial.print("Starting discoverAsync...");
    if (SerialBT.discoverAsync(btAdvertisedDeviceFound)) {
      Serial.println("Findings will be reported in \"btAdvertisedDeviceFound\"");
      delay(10000);
      Serial.print("Stopping discoverAsync... ");
      SerialBT.discoverAsyncStop();
      Serial.println("stopped");
    } else {
      Serial.println("Error on discoverAsync f.e. not workin after a \"connect\"");
    }
  }
  
  if (btScanSync) {
    Serial.println("Starting discover...");
    BTScanResults *pResults = SerialBT.discover(BT_DISCOVER_TIME);
    if (pResults)
      pResults->dump(&Serial);
    else
      Serial.println("Error on BT Scan, no result!");
  }
}

void loop() {
  delay(100);
}
```

### Debug Message

```plain
bt_classic_device_discovery:8:2: error: #error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
 #error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
  ^~~~~
Using library BluetoothSerial at version 2.0.0 in folder: C:\Users\mengd\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\BluetoothSerial 
exit status 1
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
```

### Other Steps to Reproduce

It looks like all the sample programs in BletoothSerial are not available on the ESP32-S3 series.
I have tried commenting out the program:
//if !defined(CONFIG_BT_SPP_ENABLED)
//#It is only available for the ESP32 chip.
//#endif
But it brings more error messages with undefined files:
undefined reference to `esp_spp_disconnect'
undefined reference to `esp_spp_deinit'
undefined reference to `esp_spp_write'
...

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lyusupov, 2023-04-03:
S3 has Bluetooth LE only.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/index.html

closed 2023-04-04 by limengdu

ZakiyaMiller, 2023-11-13:
is there any module that has bluetooth classic and bluetooth 5.0 and usb jtag adapter

me-no-dev, 2023-11-13:
@ZakiyaMiller nope. Only the classic ESP32 has a BT Classic. And is also the chip without any USB support

terjeio, 2024-01-12:
The [API documentation for the S3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/bluetooth/index.html) is misleading:

"ESP-IDF currently supports two host stacks. The Bluedroid based stack (default) supports classic Bluetooth as well as Bluetooth Low Energy (Bluetooth LE). On the other hand, Apache NimBLE based stack is Bluetooth Low Energy only. For users to make a choice:"

mo-han, 2024-05-16:
@terjeio 

https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/bluetooth.html
> ESP-Bluedroid for ESP32-S3 supports Bluetooth LE only. Classic Bluetooth is not supported.