espressif/arduino-esp32 #8026: Maximum sketch size incorrect on "ESP32 Wrover Kit (all versions)"

ghmpi, 2023-04-04:
### Board

esp32-devkitc-ve

### Device Description

plain module

### Hardware Configuration

any

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 7

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Hello

I am having a problem with the sketch size. The sketch size maximum calculation for 8MB seems to be incorrect and will not allow me to have a sketch size about 1.3MB.

I am using an original devkitC board, I've selected "ESP32 Wrover Kit (all versions)"

When selecting 8MB with spiffs (3.0MB APP / 1.5MB SPIFFS)
I get the following output, which seesm to be 1.3MB app, NOT 3.0MB as selected
	Sketch uses 1199625 bytes (91%) of program storage space. Maximum is 1310720 bytes.

When selecting Minimal spiffs ((1.9MB APP / 190KB SPIFFS)
It seems correct..
	Sketch uses 1199625 bytes (61%) of program storage space. Maximum is 1966080 bytes.

Arduino version if 1.8.19

Is this a bug? Can anyone help?

Thank you

Regards,
-Moses

### Sketch

```cpp
any code will work
```

### Debug Message

```plain
Sketch uses 1199625 bytes (91%) of program storage space. Maximum is 1310720 bytes.
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2023-04-04:
Can you try editing the boards.txt file to add this line:
```
esp32wrover.menu.PartitionScheme.default_8MB.upload.maximum_size=3342336
```
just below:
```
esp32wrover.menu.PartitionScheme.default_8MB.build.partitions=default_8MB
```

If that fixes it we can get a PR to add the missing upload size property.

ghmpi, 2023-04-04:
Atanisoft,

Thank you for the quick reply.

I added the line you suggested to the boards.txt file, but replaced esp32wrover with esp32wroverkit and it worked! Now I get the correct maximum sketch size.

Sketch uses 1200777 bytes (35%) of program storage space. Maximum is 3342336 bytes.

Thank you!

Regards,
-Moses

VojtechBartoska, 2023-04-13:
Any chance you take a look on this @P-R-O-C-H-Y ?

P-R-O-C-H-Y, 2023-04-13:
@VojtechBartoska Will do tomorrow :) as CI stuff should be done.

PR #8078 (2023-04-17): Fixed missing upload.maximum_size for ESP32Wrover and ESP32WroverKit
## Description of Change
ESP32Wrover and ESP32WroverKit were missing `upload.maximum_size` for `default_8MB` partition scheme.
Also formatted ESP32WroverKit, so now its clear to read the code.

## Tests scenarios
Checked all boards in boards.txt which have `default_8MB` partition scheme.

## Related links
Closes #8026
closed 2023-04-17 by me-no-dev